raspberrypi/linux #3987: CMIO4&rpi-5.10.y - DSI1(2lanes) cannot display

mingzhangqun, 2020-12-03:
**Describe the bug**

1. Problem description:
CMIO4 runs the latest rpi-5.10.y. The initialization code can be sent to MIPI LCD(ST7701s 2lanes) through lane0, but the GUI cannot be displayed.

2. Raspberrypi kernel:
[https://github.com/raspberrypi/linux.git](https://github.com/raspberrypi/linux.git)
branch: rpi-5.10.y
commit: 4f86a599b4cd07e635f289451d6460b1fd6d78ed

3. Test code:
[https://github.com/Seeed-Studio/seeed-linux-dtoverlays.git](https://github.com/Seeed-Studio/seeed-linux-dtoverlays.git)
branch: mipi_dsi
commit: 668fdb0780c2e9685b9a6c96f15909d398398ada

4. Test Method:
> Config /boot/config.txt
dtparam=i2c_arm=on
dtparam=i2c_vc=on
ignore_lcd=1
dtoverlay=vc4-kms-v3d-pi4
dtoverlay=mipi_dsi-i2c10

> Compile the code:
cd seeed-linux-dtoverlays
make all_rpi
sudo make install_rpi
sudo reboot

5. Hardware:
I have a bridge board, an MCU on the board controls the RESET and PWM backlight of the LCD, I2C connected to I2C1 of CMIO4, and the LCD MIPI pin is directly connected between the RPI.

6. What I tried so far:
I can send test commands to the LCD, and the LCD can display the test screen, as shown in the picture below.
And I can see the correct display configuration through VNC.
[image]
[image]

**Logs**
[raspinfo.txt](https://github.com/raspberrypi/linux/files/5633098/raspinfo.txt)
[dts.txt](https://github.com/raspberrypi/linux/files/5633105/dts.txt)

6by9, 2020-12-03:
The ST7701 driver appears to be a hack of the mainline driver from https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/panel/panel-sitronix-st7701.c. Is there a reason for that? It looks like your additional bit is really just a regulator and backlight driver, so should be created as such.
https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/regulator/rpi-panel-attiny-regulator.c is a rework of the Raspberry Pi DSI panel Atmel driver as a regulator and backlight.

I really don't want to be trying to remotely debug a hacked together panel driver as anything could be wrong.

Note that DSI panels must be under the DSI host DT node, otherwise it doesn't get probed correctly. See https://github.com/raspberrypi/linux/pull/3985/commits/78577dcf00035e863b86443dbdaefe44aba25911 for an example.

When you say 
> I can send test commands to the LCD, and the LCD can display the test screen

is that you've sent DSI commands to enable the test mode, or is it over I2C?

mingzhangqun, 2020-12-03:
> The ST7701 driver appears to be a hack of the mainline driver from https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/panel/panel-sitronix-st7701.c. Is there a reason for that? It looks like your additional bit is really just a regulator and backlight driver, so should be created as such.
> https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/regulator/rpi-panel-attiny-regulator.c is a rework of the Raspberry Pi DSI panel Atmel driver as a regulator and backlight.
> 
> I really don't want to be trying to remotely debug a hacked together panel driver as anything could be wrong.
> 
> Note that DSI panels must be under the DSI host DT node, otherwise it doesn't get probed correctly. See [78577dc](https://github.com/raspberrypi/linux/commit/78577dcf00035e863b86443dbdaefe44aba25911) for an example.
> 
> When you say
> 
> > I can send test commands to the LCD, and the LCD can display the test screen
> 
> is that you've sent DSI commands to enable the test mode, or is it over I2C?

### Why I did this:
My hardware is similar to the official 7-inch screen, except that my LCD is MIPI interface, which does not need bridge chip, but needs reset signal to initialize LCD screen. Therefore, I added a MCU. CMIO4 controls MCU to send reset signal through I2C.

I created the DSI device and registered the 7701s panel by referring to this driver:
https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/panel/panel-raspberrypi-touchscreen.c

> I can send test commands to the LCD, and the LCD can display the test screen

Yes, I used the DSI command to enable the test mode, which proved that my MIPI driver is correct, and I can see the waveforms on the lane0 and lane1 data lines with the oscilloscope.

### pi@raspberrypi:~ $ fbset
My LCD screen is configured with MIPI_DSI_FMT_RGB888, but the information I get with fbset is 16bit, as shown below:
mode "480x854"
    geometry 480 854 480 854 16
    timings 0 0 0 0 0 0 0
    accel true
    rgba 5/11,6/5,5/0,0/0
endmode`

### dmesg, I can see that the process is correct, but I cannot see the correct GUI.

> [    6.052258] [DSI]i2c_md_init:
[    6.052526] [DSI]i2c_md_probe:
[    6.053369] [DSI]i2c_md_probe:READ_ID=0xc3
[    6.079943] [DSI]i2c_mipi_dsi_device:endpoint[0xef44fbd0]endpoint,endpoint
[    6.080021] [DSI]i2c_mipi_dsi_device:host_node[0xef44f2e8]dsi,dsi@7e700000
[    6.080040] [DSI]i2c_mipi_dsi_device:host[0xc3213244](null)
[    6.080066] [DSI]i2c_mipi_dsi_device:info_node[0xef44f5f4]port,port
[    6.080335] [DSI]st7701_dsi_probe:
[    6.080467] [DSI]i2c_mipi_dsi_device:dsi[0xc229e600]i2c_mipi_dsi
[    6.088753] vc4-drm gpu: bound fe700000.dsi (ops vc4_dsi_ops [vc4])
[    6.089039] vc4-drm gpu: bound fe400000.hvs (ops vc4_hvs_ops [vc4])
[    6.089363] vc4-drm gpu: bound fe004000.txp (ops vc4_txp_ops [vc4])
[    6.089664] vc4-drm gpu: bound fe206000.pixelvalve (ops vc4_crtc_ops [vc4])
[    6.089962] vc4-drm gpu: bound fe207000.pixelvalve (ops vc4_crtc_ops [vc4])
[    6.090267] vc4-drm gpu: bound fe20a000.pixelvalve (ops vc4_crtc_ops [vc4])
[    6.090510] vc4-drm gpu: bound fe216000.pixelvalve (ops vc4_crtc_ops [vc4])
[    6.090787] vc4-drm gpu: bound fec12000.pixelvalve (ops vc4_crtc_ops [vc4])
[    6.091596] checking generic (3e8c7000 300000) vs hw (0 ffffffffffffffff)
[    6.091614] fb0: switching to vc4drmfb from simple
[    6.095094] Console: switching to colour dummy device 80x30
[    6.142755] [drm] Initialized vc4 0.0.0 20140616 for gpu on minor 1
[    6.145442] Console: switching to colour frame buffer device 60x53
[    6.145493] vc4-drm gpu: [drm] fb0: vc4drmfb frame buffer device
[    6.667602] [DSI]st7701_prepare:
[    7.564944] [DSI]st7701_enable:

ghollingworth, 2020-12-03:
Are you sending the DSI commands to the display in high speed or low power mode?  I don't know what the options are in the driver, but it's possible you could have low power working fine (since there are almost no parameters to get wrong in low power mode), but high speed mode has all the signals wrong.

6by9, 2020-12-03:
I do note that you've added ```MIPI_DSI_MODE_LPM``` at https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/mipi_dsi/modules/mipi_dsi/panel-st7701s.c#L385 when the original driver doesn't. 

AFAIK Commands also only make use of data lane 0, so it's possible there is an issue on data lane 1.

6by9, 2020-12-03:
(Forgot to hit comment on this earlier)

vc4-kms-dsi-7inch-overlay.dts is about to get rewritten to use the TC358762 and ATTiny regulator drivers I'd linked to.
At the moment the touchscreen controller driver probes when the display is powered down which then fails.
I'll have a go at that now.

mingzhangqun, 2020-12-04:
> Are you sending the DSI commands to the display in high speed or low power mode? I don't know what the options are in the driver, but it's possible you could have low power working fine (since there [...]

I send the DSI commands to the LCD in low power mode by enable MIPI_DSI_MODE_LPM. I speculate that there may be some problems in high-speed mode, but there is no way to prove it.

mingzhangqun, 2020-12-04:
> I do note that you've added `MIPI_DSI_MODE_LPM` at https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/mipi_dsi/modules/mipi_dsi/panel-st7701s.c#L385 when the original driver doesn't.
> 
> AFAIK Commands also only make use of data lane 0, so it's possible there is an issue on data lane 1.

Yes, I have tried it. MIPI DSI MODE LPM must be enabled before the LCD can receive commands.
Observed by an oscilloscope, lane1 has data, but I can't determine whether it is correct.

ghollingworth, 2020-12-04:
OK, since you're writing the commands in low power mode then it's clear you've just not got the high-speed timing quite right at the moment.  I would suggest you scope D0P and D0N and check the timing for the LP -> HS transition is correct in comparison to what your display is expecting.

When you get the high-speed register writes working, then go back to actually trying to get a picture on the display.

mingzhangqun, 2020-12-07:
> OK, since you're writing the commands in low power mode then it's clear you've just not got the high-speed timing quite right at the moment. I would suggest you scope D0P and D0N and check the [...]

The picture below is the waveform of DSI1_DN0, it's the timing for the LP -> HS transition,but it’s hard to distinguish.
I found that the oscilloscope cannot see the waveform of DS1_CN/P, even on the official 7-inch LCD. Why?
[image]

mingzhangqun, 2020-12-07:
> I do note that you've added `MIPI_DSI_MODE_LPM` at https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/mipi_dsi/modules/mipi_dsi/panel-st7701s.c#L385 when the original driver doesn't.
> 
> AFAIK Commands also only make use of data lane 0, so it's possible there is an issue on data lane 1.

Is there a way to solve the issue on data lane1?

6by9, 2020-12-07:
The official display only uses a single data lane.

With the official display the clock lane should be bouncing along at around 333MHz (666Mbit/s but DDR).

I have had a display working with 2 data lanes on a Pi4, so fundamentally it is working. I will transplant that build onto a CM4 with CM4IO to confirm that it works there too, but I expect it to.

mingzhangqun, 2020-12-07:
> I have had a display working with 2 data lanes on a Pi4, so fundamentally it is working. I will transplant that build onto a CM4 with CM4IO to confirm that it works there too, but I expect it to.

How can I get your 2lane working for reference.

6by9, 2020-12-07:
My test case was an ILI9881 panel with config for two lanes.
Effectively #3985, but with an alternate init command set, and amending https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c#L675 to request 2 lanes instead of 4.

6by9, 2020-12-07:
2 lane ILI9881 panel verified on a CM4IO on DISP1.
DISP0 is being a little more troublesome.

mingzhangqun, 2020-12-08:
> My test case was an ILI9881 panel with config for two lanes.
> Effectively #3985, but with an alternate init command set, and amending https://github.com/raspberrypi/linux/blob/rpi-5.10.y/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c#L675 to request 2 lanes instead of 4.

In this code, I noticed that the reset pin is always low, which will cause the panel to always be in the reset state.
https://github.com/raspberrypi/linux/blob/f77383ec0ed313e8ab19067065b423fb3e17ab82/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c#L512

pelwell, 2020-12-08:
Yes and no - the reset signal is "not asserted", but the Device Tree declaration of that GPIO should mark it as being active-low, meaning that setting the signal to 0 actually drives the pin high. This is the result of changes in the kernel between 4.19 and 5.4, and although it took us a while to catch on I think most of the overlays should have been fixed by now.

If you point me at the overlay you are trying to use I'll check it for you.

mingzhangqun, 2020-12-08:
> 2 lane ILI9881 panel verified on a CM4IO on DISP1.
> DISP0 is being a little more troublesome.

 Hi, @6by9.
I have tried to modify the code based on ILI9881, and rewrite dts, panel under the DSI host DT node. But it still doesn't work, and I can't find the reason.
My panel must be initialized in MIPI_DSI MODE LPM mode, I guess the problem is switching from MIPI_DSI_MODE_LPM to MIPI_DSI_MODE_VIDEO.
And now I have removed the bridge board, and no need I2C. RASPI controls the reset pin of the panel through GPIO17.
I have attached dmesg below:
[dmesg.txt](https://github.com/raspberrypi/linux/files/5658557/dmesg.txt)

https://github.com/Seeed-Studio/seeed-linux-dtoverlays.git
branch: mipi_dsi_st7701s
commit:1e9151bb743ab058d3b4e99c72fcf3f0c76ed5ab

mingzhangqun, 2020-12-08:
> Yes and no - the reset signal is "not asserted", but the Device Tree declaration of that GPIO should mark it as being active-low, meaning that setting the signal to 0 actually drives the pin high. [...]

Okay, I understand now, thanks.

mingzhangqun, 2020-12-08:
I also noticed that the information obtained by fbset is incorrect. Why?
[image]

6by9, 2020-12-08:
If you look at the full PR #3985, there is a patch to make the reset line optional as it isn't wired up in our case. pelwell's explanation is correct anyway - DT should deal with any active low/active high symantics.
The PR also includes a patch to set the ```MIPI_DSI_MODE_VIDEO``` flag as ```MIPI_DSI_MODE_VIDEO_SYNC_PULSE``` is a modifier to using video mode instead of command mode.

Why are you now basing your driver for a ST7701 panel on ILI9881 when there is an ST7701 panel driver in the kernel?
https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/panel/panel-sitronix-st7701.c

fbset reports the geometry of the emulated framebuffer that the DRM core framework creates. Use [modetest](https://gitlab.freedesktop.org/mesa/drm/tree/master/tests/modetest) or ```xrandr --verbose``` if you want to see the timings of the underlying display devices.

mingzhangqun, 2020-12-08:
> Why are you now basing your driver for a ST7701 panel on ILI9881 when there is an ST7701 panel driver in the kernel?
> https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/panel/panel-sitronix-st7701.c

I have tested panel-sitronix-st7701.c before.
I use ILI9881c to test, the purpose is to make the code closer to yours.

6by9, 2020-12-08:
Generally the quality of the drivers in the Linux kernel is good, and the fact that it has been merged means someone has had it working with some hardware.
If it's not working for you then generally it's better to do a little debugging than to throw it away and try rolling your own.

Is this a panel that can be easily bought from somewhere? We can make up an interconnecting board as long as the documentation gives the appropriate pinouts.

mingzhangqun, 2020-12-08:
The st7701s panel can work normally on other platforms (such as Android), so I tried to analyze whether there are some problems with vc4_dsi.
This panel was purchased in China, and I don’t know other purchase channels.

mingzhangqun, 2020-12-08:
> Is this a panel that can be easily bought from somewhere? We can make up an interconnecting board as long as the documentation gives the appropriate pinouts.

If convenient, you can provide the address and I will mail you a set of st7701s panel.

Pillar1989, 2020-12-08:
@6by9  Thank you for your support, I work with @mingzhangqun  and I just sent you an email. I know there is some private information here, and we'll discuss it in an email.

igrek-xavier, 2022-01-18:
@6by9 @Pillar1989 @mingzhangqun Hi there, thanks for the very interesting thread. I am trying to connect a st7701s (DSI 2-lanes) LCD screen from DLC (DLC0200CCP04DF) to a Rpi3 using the MIPI-DSI connector (J4) but so far no success to get the display to start.

I was wondering if you guys had succeeded for a DSI 2-lanes support for st7701s with Rpi linux ?

- hardware : D0_N/P, D1_N/P, C_N/P, RST on GPIO17 of Rpi, VCC=3V3 from Rpi 3V3, IOVCC=3V3 from Rpi 3V3
- software : following https://github.com/harlab/CM4_LCD_LT070ME05000 and finally loading st7701 kernel module.
- I can see vc4 loading but dsi last message is 
vc4-drm soc:gpu: bound 3f700000.dsi (ops vc4_dsi_ops [vc4])
- device info : cat /proc/device-tree/soc/dsi@7e700000/compatible
brcm,bcm2835-dsi1