raspberrypi/linux #3989: Buster udev rules for symlinks to serial-to-USB devices broken

Fuchks, 2020-12-07:
udev rules for symlinks to serial-to-USB devices will symlink to gpiochip on freshly installed and updated Buster.
This issue arises on a Pi2B Rev 1.1 HW 1a01041 with 2x FTDI FT232 adapters.

used udev rule (example)
`SUBSYSTEMS=="usb", ATTRS{serial}=="12345678", SYMLINK+="MyUSBDevice"`

Discussion for that issue:
https://www.raspberrypi.org/forums/viewtopic.php?t=283585

Disabling stock /etc/udev/rules.d/99-com.rules solves the issue partly (not everytime) - or excluding the following lines:

```
#SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"
#SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
#       chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
#       chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
#       chown -R root:gpio /sys$devpath && chmod -R 770 /sys$devpath\
#'"
```

Babbion, 2020-12-09:
I have the same issue on a Pi3B+ with 4xFTDI Ft232 adapters. 
The udev rule create "bad" symlinks in /dev/ and the only way to get them works is send: 
udevadm test $(udevadm info -q path -n /dev/ttyUSBx) 
for every device connected

Fuchks, 2020-12-09:
My workaround as i do not need this gpio stuff:

Exclude the following lines from the stock **/etc/udev/rules.d/99-com.rules**:

```
#SUBSYSTEM=="gpio", GROUP="gpio", MODE="0660"
#SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
#       chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
#       chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
#       chown -R root:gpio /sys$devpath && chmod -R 770 /sys$devpath\
#'"
```

Add the line to **/etc/rc.local**: `sudo udevadm control --reload-rules && sudo udevadm trigger`

Let me know if this works reliable on your system too...

XECDesign, 2021-01-05:
Does it work if you change `/sys$devpath` to `/sys/$devpath`?

XECDesign, 2021-01-05:
To answer my own question, no it doesn't seem to. I was hoping this was a variant of https://github.com/raspberrypi/linux/issues/3779

XECDesign, 2021-01-06:
Using `SUBSYSTEM=="tty"` (note the lack of `S`) makes your rule work for me. It looks like your rule also applies to gpiochip2, which belongs to the USB device and there's a race between whether ttyUSB0 or gpiochip2 is linked last. Specifying `SUBSYSTEM=="tty"` ensures that the rule does not apply to gpiochip2.

It looks like the rules in `99-com.rules` doesn't directly affect it. If I remove it, I still see the rule applying to gpiochip2. I suspect it just introduces enough of a delay to change the likely outcome of the race described above.

jsjames, 2021-01-06:
Hi XECDesign - That suggestion seems to work for me.  Thanks for the help here.

vdavy, 2021-01-08:
Go the same strange stuff : type lsusb and got all usb devices listed. Plug an FTDI USB RS232 devices, and types lsusb and got no devices.
With the FTDI plugged, all devices need root access to work. 
So I fixed it by adding this content to file : /etc/udev/rules.d/60-rpi.gpio-common.rules to override the one provided :
SUBSYSTEM=="bcm2835-gpiomem", KERNEL=="gpiomem", GROUP="dialout", MODE="0660"
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys/class/gpio/export /sys/class/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; chmod 660 /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value'"

REboot needed to have changes applied.

The bugged original one has the last line with added slash before each %p pattern : /sys/%p/active_low instead of the correct form /sys%p/active_low
The buggy apt package is rpi.gpio-common

XECDesign, 2021-01-08:
@vdavy That's unrelated to this issue. Please see https://github.com/raspberrypi/linux/issues/3779 where somebody actually suggested adding the slash as a fix.

vdavy, 2021-01-08:
@XECDesign thanks for notifying me.
So I deleted my comment on this issue and migrated it to the issue you mentionned me.

Fuchks, 2021-01-18:
> Using `SUBSYSTEM=="tty"` (note the lack of `S`) makes your rule work for me. It looks like your rule also applies to gpiochip2, which belongs to the USB device and there's a race between whether [...]

Not thorougly tested, but seems to work.
IMO this is only a workaround, as most suggested to create symlinks to usb-devices with serialnr. is:
`
SUBSYSTEMS=="usb", ATTRS{serial}=="12345678", SYMLINK+="MyUSBDevice"`

XECDesign, 2021-01-18:
I don't think it's just a workaround, since udev is doing what it's being asked to do. That gpiochip device happens to match the udev rule, so the udev rule needs to be more specific.

jsjames, 2021-01-18:
> I don't think it's just a workaround, since udev is doing what it's being asked to do. That gpiochip device happens to match the udev rule, so the udev rule needs to be more specific.

I don't believe the gpiochip should have matched the original udev rule because the serial number was listed as part of the udev rule.  Here were my original rules where I included the serial number in each line.  All of these in buster were mapped to gpiochip - which can't be correct operation.

SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="A60335LE", SYMLINK+="ttyInsteonPLM"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="A5003uIM", SYMLINK+="ttyAD2USB"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="AK05ERIN", SYMLINK+="ttyRS422"

XECDesign, 2021-01-18:
The serial number matches the parent USB device gpiochip is on. Why do you believe this to be incorrect behaviour?

jsjames, 2021-01-18:
Thanks XECDesign. I had thought the serial number was for the device, not the parent.  In looking into this some more, I see what you are saying and why the previous rules would apply to both.

XECDesign, 2021-01-18:
It seems to inherit parent attributes. If you run `udevadm info --attribute-walk --name=/dev/gpiochip2` (or `ttyUSB0`), I think you'll see that the serial number is only present on the parent usb device and not any of the children. Since the tty and gpiochip devices are both children of the usb device, both match. I was a bit confused when I saw that as well because that's not how I assumed it worked.

Fuchks, 2021-02-27:
> 
> 
> I don't think it's just a workaround, since udev is doing what it's being asked to do. That gpiochip device happens to match the udev rule, so the udev rule needs to be more specific.

You are right and I withdraw my statement 
> IMO this is only a workaround, as most suggested to create symlinks to usb-devices with serialnr. is:
`SUBSYSTEMS=="usb", ATTRS{serial}=="12345678", SYMLINK+="MyUSBDevice"`

Fuchks, 2021-02-27:
Solution from XECDesign: https://github.com/raspberrypi/linux/issues/3989#issuecomment-755231352

Udev rule has to look like:
`SUBSYSTEM=="tty", ATTRS{serial}=="12345678", SYMLINK+="MyUSBDevice"`

closed 2021-02-27 by Fuchks