raspberrypi/linux #3990: Non-FUSE exfat drivers

josephernest, 2020-12-07:
It seems that Linux Kernel has 2 non-FUSE exfat drivers:

* exfat in kernel 5.4
* exfat in kernel 5.7 (improved by samsung, made available for previous kernel versions here: https://github.com/arter97/exfat-linux)

I've tried with RaspiOS Lite (2020-12-02-raspios-buster-armhf-lite.zip), but `modprobe exfat` or `modprobe exfat-linux` or `modprobe exfat-nonfuse` don't work.

Questions: 

* Is exFAT (non fuse) included in this RPi Linux kernel?
* In which version of RasperryPiOS Lite is it included? If not, how to `apt install` it?

popcornmix, 2020-12-07:
It is available in 5.9 (and newer kernels). See https://github.com/raspberrypi/linux/issues/3901
You can update early by following this [thread](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=288234) are wait for apt kernel to be updated to 5.10 (no definite timescale but perhaps in next couple of months).
There's no current play to enable the less capable version in 5.4.

josephernest, 2020-12-07:
Thanks @popcornmix. Is there a way to load this new driver without having to change the whole kernel?

popcornmix, 2020-12-07:
No. The code isn't present in your current kernel.

josephernest, 2020-12-07:
In the meantime I tried https://github.com/arter97/exfat-linux and it seems to work! (with manual compile and install)

Abu-Abdullah, 2021-02-13:
i have upgrade to  5.10.14 but it seems it is still missing the support for exfat out of the box
```
pi@raspberrypi:~ $ cat /proc/filesystems
nodev   sysfs
nodev   tmpfs
nodev   bdev
nodev   proc
nodev   cgroup
nodev   cgroup2
nodev   cpuset
nodev   devtmpfs
nodev   configfs
nodev   debugfs
nodev   tracefs
nodev   securityfs
nodev   sockfs
nodev   bpf
nodev   pipefs
nodev   ramfs
nodev   rpc_pipefs
nodev   devpts
        ext3
        ext2
        ext4
        vfat
        msdos
nodev   nfs
nodev   nfs4
nodev   autofs
        f2fs
nodev   mqueue

```

6by9, 2021-02-13:
https://github.com/raspberrypi/firmware/tree/master/modules/5.10.14-v7l%2B/kernel/fs/exfat
It's a module not a builtin driver. Have you done ```sudo modprobe exfat```?

Abu-Abdullah, 2021-02-13:
thank you @6by9 
exfat is enabled when loading the module using `sudo modprobe exfat`
i modified `/etc/modules-load.d/modules.conf` to load it at boot time, and install exfatprogs manually for its tools. it seems this tool is recent and not in buster (i found it in Debian Unstable). hope this tool will find its way to raspi at one point of time

closed 2021-02-13 by pelwell

XECDesign, 2021-02-15:
I've just uploaded a backported (but untested) exfatprogs package to the repo. It should be public within an hour.