raspberrypi/linux #3991: Output of /proc/cpuinfo is missing the 'model name' field on RPi4 64-bit

graysky2, 2020-12-07:
**Describe the bug**
The field `model name` is missing from 64-bit kernel I build from [rpi-5.10.y](https://github.com/raspberrypi/linux/tree/rpi-5.10.y) when I  `cat /proc/cpuinfo`

**To reproduce**
Boot into a 64-bit kernel and `grep 'model name' /proc/cpuinfo`

**Expected behaviour**
Expectation is to see something similar to the output under the 32-bit version:
```
% grep 'model name' /proc/cpuinfo
model name	: ARMv7 Processor rev 3 (v7l)
model name	: ARMv7 Processor rev 3 (v7l)
model name	: ARMv7 Processor rev 3 (v7l)
model name	: ARMv7 Processor rev 3 (v7l)
```

**Actual behaviour**
On a 64-bit build:
```
% grep 'model name' /proc/cpuinfo
% echo $?
1
```

**System**
* Which model of Raspberry Pi? RPi4 B
* Which OS and version Arch ARM aarch64 running rpi-5.10.y kernel NOT mainline
* Which firmware version (`Nov 25 2020 16:55:59 
Copyright (c) 2012 Broadcom
version 2e6292a5304b95ecfd700fc5ab2b793402056332 (clean) (release) (start)`)?
* Which kernel version (`Linux submyth 5.10.0-rc6-2-ARCH #1 SMP PREEMPT Sun Dec 6 08:16:01 EST 2020 aarch64 GNU/Linux`)?

**Logs**
Link to dmesg [output](https://gist.github.com/graysky2/272f36b4bac37f47f3d2df65f4fca21f).

**Additional context**
`cat /proc/cpuinfo` on a kernel build from the same commit of rpi-5.10.y for 32-bit is [here](https://gist.github.com/graysky2/4d2d6522c5fbbf0a146c4bc93be61475).
`cat /proc/cpuinfo` on a kernel build from the same commit of rpi-5.10.y for 64-bit is [here](https://gist.github.com/graysky2/d58aa3b2b28d168c71a26452d6ccf936).

Is this a genuine bug in the code or due to differences in the kernel config I am building or something else on my end?

popcornmix, 2020-12-07:
I believe this is just expected behaviour. Upstream arm64 kernel code reports less information in /cpu/cpuinfo.

6by9, 2020-12-07:
Just differences between https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/kernel/setup.c#L1243 and https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm64/kernel/cpuinfo.c#L137

The base upstream cpuinfo for arm64 doesn't include revision, serial, or model

graysky2, 2020-12-07:
Thanks guys.  I am not learned with c at all... how difficult would be be to force the behavior for consistency between arm and arm64?

pelwell, 2020-12-07:
You'd be better off picking the values out of /proc/device-tree - then it will work with an upstream kernel as well:
```
$ cat /proc/device-tree/model
$ cat /proc/device-tree/serial-number
```

graysky2, 2020-12-07:
Thanks for the pointers, all.  I will close this since it is not a bug.

graysky2, 2020-12-07:
Reopening... I found it was a trivial as the attached.  Would you consider incorporating this change into rpi-5.10.y?

```
--- a/arch/arm64/kernel/cpuinfo.c	2020-12-07 08:21:53.000000000 -0500
+++ b/arch/arm64/kernel/cpuinfo.c	2020-12-07 12:03:34.004597471 -0500
@@ -153,9 +153,8 @@ static int c_show(struct seq_file *m, vo
 		 * "processor".  Give glibc what it expects.
 		 */
 		seq_printf(m, "processor\t: %d\n", i);
-		if (compat)
-			seq_printf(m, "model name\t: ARMv8 Processor rev %d (%s)\n",
-				   MIDR_REVISION(midr), COMPAT_ELF_PLATFORM);
+		seq_printf(m, "model name\t: ARMv8 Processor rev %d (%s)\n",
+			   MIDR_REVISION(midr), COMPAT_ELF_PLATFORM);
 
 		seq_printf(m, "BogoMIPS\t: %lu.%02lu\n",
 			   loops_per_jiffy / (500000UL/HZ),
```

Sample on 5.10.0-rc7
```
% grep 'model name' /proc/cpuinfo
model name	: ARMv8 Processor rev 3 (v8l)
model name	: ARMv8 Processor rev 3 (v8l)
model name	: ARMv8 Processor rev 3 (v8l)
model name	: ARMv8 Processor rev 3 (v8l)
```

graysky2, 2020-12-07:
Reopening... I found it was trivial as dropping the if and leaving the existing code...  I am no expert as I said above.  Would you consider incorporating this change into rpi-5.10.y?  I can create a formal PR if so.  Please let me know either way.

```
--- a/arch/arm64/kernel/cpuinfo.c	2020-12-07 08:21:53.000000000 -0500
+++ b/arch/arm64/kernel/cpuinfo.c	2020-12-07 12:03:34.004597471 -0500
@@ -153,9 +153,8 @@ static int c_show(struct seq_file *m, vo
 		 * "processor".  Give glibc what it expects.
 		 */
 		seq_printf(m, "processor\t: %d\n", i);
-		if (compat)
-			seq_printf(m, "model name\t: ARMv8 Processor rev %d (%s)\n",
-				   MIDR_REVISION(midr), COMPAT_ELF_PLATFORM);
+		seq_printf(m, "model name\t: ARMv8 Processor rev %d (%s)\n",
+			   MIDR_REVISION(midr), COMPAT_ELF_PLATFORM);
 
 		seq_printf(m, "BogoMIPS\t: %lu.%02lu\n",
 			   loops_per_jiffy / (500000UL/HZ),
```

Sample on 5.10.0-rc7
```
% grep 'model name' /proc/cpuinfo
model name	: ARMv8 Processor rev 3 (v8l)
model name	: ARMv8 Processor rev 3 (v8l)
model name	: ARMv8 Processor rev 3 (v8l)
model name	: ARMv8 Processor rev 3 (v8l)
```

lategoodbye, 2020-12-07:
@graysky2 Could you please describe the actual use case or requirement?

graysky2, 2020-12-07:
@lategoodbye - It's purely cosmetic.  I understand if you don't want to incorporate the change.

pelwell, 2020-12-16:
The `lscpu` tool gives better information about the processor cores without requiring kernel modifications. On 32-bit:
```
pi@raspberrypi:~$ lscpu
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A72
Stepping:            r0p3
CPU max MHz:         1500.0000
CPU min MHz:         600.0000
BogoMIPS:            108.00
Flags:               half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
```
And on 64-bit:
```
pi@raspberrypi:~$ lscpu
Architecture:        aarch64
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A72
Stepping:            r0p3
CPU max MHz:         1500.0000
CPU min MHz:         600.0000
BogoMIPS:            108.00
Flags:               fp asimd evtstrm crc32 cpuid
```

closed 2022-02-18 by graysky2

ijiki16, 2024-02-02:
`lscpu` is good but some tools like `neofetch` use `/proc/cpuinfo`.

lategoodbye, 2024-02-02:
Since procfs is user-space API it's not a good idea to play around in a vendor tree. The userspace application should be fixed instead.