raspberrypi/linux #3992: g_mass_storage not reflecting any filesystem change on host or device

PTS93, 2020-12-07:
**Describe the bug**
When using g_mass_storage as described below by mounting a file formatted as a fat32 filesystem, it is not possible to see changes done by the host on the device (pi) or changed done on the device on the host.
I tried `sync` on both systems, that made no difference.

**To reproduce**
```
# Plug Pi4 with USB otg enabled in config.txt into computer
sudo dd if=/dev/zero of=/piusb.bin bs=512 count=2880
sudo mkdosfs /piusb.bin
sudo modprobe g_mass_storage file=/piusb.bin stall=0 removable=0
#add to fstab:
# /piusb.bin /home/pi/pidisk vfat users,umask=000 0 2
sudo mount -a
```
Write file to /home/pi/pidisk, changes should not reflect back to USB MSD mounted on host computer. The other way around, write a file to MSD on host. Changes will not reflect on Pi.

Changes will only reflect if either the Pi unmounts and mounts the file again or if the MSD device is removed via modprobe -r and added again to re-initialize with the host.

**Expected behaviour**
Any change on either system should be reflected on both systems.

**Actual behaviour**
No changes are reflected on either system until the connection/mount is cycled.

**System**
https://pastebin.com/59e0h3mS

**Logs**
see above pastebin for dmesg

**Additional context**
Using a Pi4 with a USB-C to USB-C cable attached to a Lenovo T480s with Manjaro Linux running kernel 5.8.18-1-MANJARO

RICCIARDI-Adrien, 2020-12-16:
Hi,
I already tried that and it failed too. I think this can't work because FAT32 is not a "shared" file system, so it can't be mounted by more than a single machine at a time.
A lot of FAT32 informations are cached by the system that mounts it, so an update by another system does not change the cached information and data are randomly overwritten when syncing.

timonsku, 2020-12-16:
I also arrived at that in the meantime. I found a hacky solution by remounting on file change. This leaves open the possibility of reading in files while another write is happening but is an ok compromise in my case and can be recovered from.
But I guess there is nothing really to fix here but would require a new driver.

closed 2020-12-16 by timonsku