raspberrypi/linux #3994: Can‘t compile new module.ko file based on latest firmware 5.4.79-v7+ kernel version

UCTRONICS, 2020-12-09:
Hello,
After the new firmware based on 5.4.79 kernel version released, I can't compile my own nodule.ko file? Is anyone know where is the kernel source code?

6by9, 2020-12-09:
https://github.com/raspberrypi/linux/releases/tag/raspberrypi-kernel_1.20201201-1
Tag to match the Debian package version of 1.20201201-1

UCTRONICS, 2020-12-10:
Hello @6by9 
Thanks for your response.
I have tried this code you recommend for me but the same result.  Do you know why?
I find what I get is 5.4.79-v7 instead of 5.4.79-v7+. Thanks for any help.
[image]
[image]
[image]

yoyojacky, 2020-12-10:
about 5.4.79-v7 instead of 5.4.79-v7+ 
The plus sign at the end of your version string is there as an indicator that the kernel was built from modified sources (that is, there were non-committed changes). This is also indicated by the comments in scripts/setlocalversion 
try this 
```
sed -i "s|CONFIG_LOCALVERSION_AUTO=.*|CONFIG_LOCALVERSION_AUTO=n|" .config 
```

UCTRONICS, 2020-12-11:
Hello @yoyojacky
Thanks for your response. I try to add  sed -i "s|CONFIG_LOCALVERSION_AUTO=.*|CONFIG_LOCALVERSION_AUTO=n|" .config  Then recompiled the source code. But the same result. Any other suggestions?

Luu-Dung, 2021-01-02:
Hi @UCTRONICS
Let try to change "CONFIG_LOCALVERSION" from "-v7" to "-v7+" in .config (after "make ... bcm2709_defconfig").