raspberrypi/linux #3996: raspberrypi-kernel-headers does not contain 5.4.72-v7l+ headers

bojomojo, 2020-12-09:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
I have 5.4.72-v7l+ running, but when I install raspberrypi-kernel-headers other versions are downloaded, but never 5.4.72-v7l+

**To reproduce**
Install the official raspian homebridge sd image,
then 
sudo apt install raspberrypi-kernel-headers

**Expected behaviour**
5.4.72-v7l+ header files shall be downloaded

**Actual behaviour**
Other versions (newer) are downloaded, but not 5.4.72-v7l+

**System**
 pi@homebridge:~/raspbee2-rtc-master/raspbee2-rtc-master $ raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference Raspbian
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, d921b3fc42da0e2d02ffcddbbf958b406d97d038, stage3_homebridge

Linux homebridge 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux
Revision	: b03114
Serial		: 100000009e05f877
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Oct 22 2020 13:59:27
Copyright (c) 2012 Broadcom
version 74e754ff8947c58d2773253f77f6f68a303188f8 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       29772492 2636384  25890956  10% /
devtmpfs          823892       0    823892   0% /dev
tmpfs             955988       0    955988   0% /dev/shm
tmpfs             955988   17120    938868   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             955988       0    955988   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   55986    202110  22% /boot
tmpfs             191196       0    191196   0% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	0	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: (none)
raspberrypi-sys-mods:
  Installed: 20201026
[... 912 lines cut ...]
[   38.487043] brcmfmac: F1 signature read @0x18000000=0x15264345
[   38.495669] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   38.496076] usbcore: registered new interface driver brcmfmac
[   38.951276] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[   39.177996] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   39.263287] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea
[   51.869714] uart-pl011 fe201000.serial: no DMA platform data
[   53.022409] random: crng init done
[   53.022428] random: 7 urandom warning(s) missed due to ratelimiting
[   71.883732] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   93.487897] bcmgenet: Skipping UMAC reset
[   93.490139] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[   93.490484] bcmgenet fd580000.ethernet eth0: Link is Down
[   93.541091] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   94.068660] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   98.590974] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   98.713829] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   98.713862] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  103.917647] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[34970.800858] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

6by9, 2020-12-09:
The kernel being shipped via apt as raspberrypi-kernel is now 5.4.79, therefore raspberrypi-kernel-headers will give you 5.4.79 headers.

bojomojo, 2020-12-09:
i did rasoberrypi-kernel reinstall, it still gave me .72

XECDesign, 2020-12-09:
Did you run `sudo apt update` first? If so, what's the output of `apt policy raspberrypi-kernel`?

bojomojo, 2020-12-09:
I did another update/upgrade and reboot.. now Kernel is updated to .79 and everything matches.. thanks

closed 2020-12-09 by bojomojo