raspberrypi/linux #4000: kernel BUG: unsupported FP instruction in kernel mode

scyld, 2020-12-11:
**Describe the bug**
I have recently starterd using rtorrent on RPi4. After some hours of running,
like 4 or five the kernel BUG appears and makes rtorrent z zombie process.
Sending SIGQUIT actually ends the process.

Dmesg shows only this:
```
[ 2865.782350] BUG: unsupported FP instruction in kernel mode
[ 2865.783346] Internal error: Oops - undefined instruction: 0 [#1] SMP ARM
[ 2865.784312] Modules linked in: btrfs xor xor_neon zstd_decompress zstd_compress xxhash raid6_pq zlib_deflate dm_crypt aes_arm_bs crypto_simd cryptd algif_skcipher af_alg dm_mod cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc sg brcmfmac vc4 brcmutil cec sha256_generic libsha256 drm_kms_helper cfg80211 v3d gpu_sched rfkill drm bcm2835_codec(C) snd_soc_core v4l2_mem2mem bcm2835_isp(C) bcm2835_v4l2(C) snd_compress bcm2835_mmal_vchiq(C) videobuf2_dma_contig snd_pcm_dmaengine snd_bcm2835(C) videobuf2_vmalloc videobuf2_memops snd_pcm videobuf2_v4l2 syscopyarea sysfillrect videobuf2_common snd_timer drm_panel_orientation_quirks videodev sysimgblt fb_sys_fops snd mc raspberrypi_hwmon vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio nfsd ip_tables x_tables ipv6
[ 2865.788903] CPU: 0 PID: 1029 Comm: rtorrent main Tainted: G         C        5.4.79-v7l+ #1373
[ 2865.790094] Hardware name: BCM2711
[ 2865.791277] PC is at vfp_reload_hw+0x30/0x44
[ 2865.792452] LR is at __und_usr_fault_32+0x0/0x8
[ 2865.793617] pc : [<c0204a48>]    lr : [<c0201f40>]    psr: a0000013
[ 2865.794791] sp : d5755fb0  ip : d5755f30  fp : 00000000
[ 2865.795963] r10: d57541f8  r9 : c0201fa4  r8 : c0a95438
[ 2865.797137] r7 : 00000001  r6 : c0251df4  r5 : d5754208  r4 : 00000000
[ 2865.798309] r3 : c12b9068  r2 : b6ef6b60  r1 : d5754254  r0 : ed2d8b04
[ 2865.799491] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 2865.800677] Control: 30c5383d  Table: 161dc100  DAC: fffffffd
[ 2865.801867] Process rtorrent main (pid: 1029, stack limit = 0xfbfd47e7)
[ 2865.803073] Stack: (0xd5755fb0 to 0xd5756000)
[ 2865.804264] 5fa0:                                     024195dc 02419618 00000004 024195dc
[ 2865.805480] 5fc0: 00000004 024195d4 00133e00 025c7068 00010102 00000004 0000000d 00134908
[ 2865.806699] 5fe0: b6dc0394 bece2580 b6d81c28 b6ef6b60 20000010 ffffffff 00000000 00000000
[ 2865.807919] Backtrace: no frame pointer
[ 2865.809136] Code: 128aa080 e89a0162 e3110102 0a000003 (eee96a10) 
[ 2865.810376] ---[ end trace 0dcf18c9b807b19c ]---
```

**To reproduce**
Try running rtorrent with the following configuration:

```
scgi_port = 127.0.0.1:5001
bind = 0.0.0.0
encoding_list = UTF-8
port_range = 45000-46000
port_random = no
check_hash = no
directory = /home/pi/seedbox/torrents
session = /home/pi/seedbox/.session
encryption = allow_incoming, try_outgoing, enable_retry
schedule = watch_directory,1,1,"load.start=/home/pi/seedbox/watch/*.torrent"
schedule = untied_directory,5,5,"stop_untied=/home/pi/seedbox/watch/*.torrent"
trackers.use_udp.set = yes
dht = off
protocol.pex.set = no
min_peers = 40
max_peers = 100
min_peers_seed = 10
max_peers_seed = 50
max_uploads = 15
execute = {sh,-c,/usr/bin/php /media/php/initplugins.php seedbox &}
schedule = espace_disque_insuffisant,1,30,close_low_diskspace=500M
```

on latest and fully upgraded Raspbian GNU/Linux 10 (buster) 32 bit.

**Expected behaviour**
Not having rtorrent or any other processes to get into ZOMBIE state.
Currently only rtorrent identified.

**System**
Output of raspinfo:
https://pastebin.com/d4gAVdLc

pelwell, 2020-12-11:
Although this is being reported as a kernel error, the kernel is running in user thread context when it occurs. I'm not well versed in the ARM's FP (and FP emulation) support, but it looks like the threads floating point state doesn't match the hardware's capabilities - it claims to have saved state for the fpinst register when in fact that register doesn't exist.

The quickest way forward for you might be to try running in 64-bit mode - put `arm_64bit=1` in config.txt and reboot. This might resolve the problem, or it might just report it in a different way, but either way we learn something.

scyld, 2020-12-11:
Hello,

well, I'm not an expert but it seems that running in 64-bit broken rtorrent completely:

```
Caught SIGBUS, dumping stack:
rtorrent() [0x1ed84]
/lib/arm-linux-gnueabihf/libc.so.6(__default_rt_sa_restorer+0) [0xf71c8130]
/lib/arm-linux-gnueabihf/libtorrent.so.20(+0xb9f9c) [0xf7660f9c]
/lib/arm-linux-gnueabihf/libtorrent.so.20(_ZN7torrent9PollEPoll7performEv+0xdc) [0xf75d64e8]
/lib/arm-linux-gnueabihf/libtorrent.so.20(_ZN7torrent9PollEPoll7do_pollExi+0xa8) [0xf75d669c]
/lib/arm-linux-gnueabihf/libtorrent.so.20(_ZN7torrent11thread_base10event_loopEPS0_+0x17c) [0xf760fb7c]
rtorrent() [0x1d8f4]
/lib/arm-linux-gnueabihf/libc.so.6(__libc_start_main+0x10c) [0xf71b2718]

Error: Success
Signal code '1': Invalid address alignment.
Fault address: 0x12a3ba1
The fault address is not part of any chunk.
Interupted
```
Funny I do not see such errors for other processes. Only rtorrent.
Shall I return to 32-bit. I thing I should…

pelwell, 2020-12-11:
It's bad practise to do unaligned accesses to data, and the aarch64 kernel removed the ability for the kernel to patch them up.

If you must run rtorrent then you might find you get can avoid the floating point exceptions if you build it yourself. Unless you already did...

longdadouzi, 2021-01-25:
May I ask whether the landlord solves this kind of problem, and seeks solutions

pelwell, 2021-01-25:
If you are asking are we trying to fix rtorrent the answer is no.

yutayu, 2021-03-24:
Mar 25 02:19:51 raspberrypi kernel: BUG: unsupported FP instruction in kernel mode
Mar 25 02:19:51 raspberrypi kernel: FPEXC == 0x40000700
Mar 25 02:19:51 raspberrypi kernel: Internal error: Oops - undefined instruction: 0 [#2] S
Mar 25 02:19:51 raspberrypi kernel: Modules linked in: fuse hci_uart btbcm bluetooth ecdh_
Mar 25 02:19:51 raspberrypi kernel: CPU: 0 PID: 1640 Comm: iotop Tainted: G      D  C     
Mar 25 02:19:51 raspberrypi kernel: Hardware name: BCM2835
Mar 25 02:19:51 raspberrypi kernel: PC is at vfp_reload_hw+0x30/0x44
Mar 25 02:19:51 raspberrypi kernel: LR is at __und_usr_fault_32+0x0/0x8
Mar 25 02:19:51 raspberrypi kernel: pc : [<80103834>]    lr : [<80100fc0>]    psr: a000001
Mar 25 02:19:51 raspberrypi kernel: sp : 82ad7fb0  ip : 82ad7f30  fp : 00000000
Mar 25 02:19:51 raspberrypi kernel: r10: 82ad61f8  r9 : 80101024  r8 : 80ee5ea8
Mar 25 02:19:51 raspberrypi kernel: r7 : 00000001  r6 : 00000001  r5 : 80651af8  r4 : 9893
Mar 25 02:19:51 raspberrypi kernel: r3 : 80f4807c  r2 : 76f02100  r1 : 8066bd24  r0 : eee0
Mar 25 02:19:51 raspberrypi kernel: Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  S
Mar 25 02:19:51 raspberrypi kernel: Control: 10c5383d  Table: 029f806a  DAC: 00000055
Mar 25 02:19:51 raspberrypi kernel: Process iotop (pid: 1640, stack limit = 0xbddfb8ce)
Mar 25 02:19:51 raspberrypi kernel: Stack: (0x82ad7fb0 to 0x82ad8000)
Mar 25 02:19:51 raspberrypi kernel: 7fa0:                                     76411fb4 000
Mar 25 02:19:51 raspberrypi kernel: 7fc0: 76411f98 0019f670 003d06d4 00000000 76439c80 763
Mar 25 02:19:51 raspberrypi kernel: 7fe0: 003b617c 7e9fc8f8 00196700 76f02100 20000010 fff
Mar 25 02:19:51 raspberrypi kernel: Backtrace: no frame pointer
Mar 25 02:19:51 raspberrypi kernel: Code: 128aa080 e89a0162 e3110102 0a000003 (eee96a10) 
Mar 25 02:19:51 raspberrypi kernel: ---[ end trace 100b7f3c50ac5a20 ]---

This is pulseaudio problem.
similer?

pelwell, 2021-03-24:
It looks more like an iotop problem, but you won't get taken seriously until you start saying what platform (hardware and software) you are running on etc. It's better to create a new issue.

yutayu, 2021-03-24:
@pelwell 
Thanks.I will continue to notice it.

yutayu, 2021-03-24:
@scyld 
I changed zram algo lzo-rle from zstd and this problem don't reproduce , yet.

longdadouzi, 2021-03-31:
> @scyld
> I changed zram algo lzo-rle from zstd and this problem don't reproduce , yet.

Do you have a way to solve it？

yutayu, 2021-03-31:
@longdadouzi 
Change algo zstd. I think this is solution.

longdadouzi, 2021-04-19:
> @longdadouzi
> Change zstd algo lzo-rle algo . I think this is solution.
I don't seem to understand your intention

yutayu, 2021-04-19:
@longdadouzi 
zstd with raspbian causes error. I confirmed.
so I suggests you not to use zstd .