raspberrypi/linux #4001: zram: Decompression failed

yutayu, 2020-12-11:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.
Chromium stole often and sometimes pi freezes.

**To reproduce**
List the steps required to reproduce the issue.

Use much memory with zram device on pi .

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

No errors.

**Actual behaviour**
Add a clear and concise description of what actually happened.

Error happens often when program like chromium wastes zram memory 

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

https://paste.debian.net/1176483/

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
[ 7002.412278] zram: Decompression failed! err=-22, page=47388
[ 7465.315151] zram: Decompression failed! err=-22, page=137336
[ 8109.158347] zram: Decompression failed! err=-22, page=5338
[ 8109.280596] zram: Decompression failed! err=-22, page=9302
[ 8384.470910] zram: Decompression failed! err=-22, page=21931
[ 8416.053896] zram: Decompression failed! err=-22, page=172042
[ 8509.027964] zram: Decompression failed! err=-22, page=30609
[ 8509.120017] zram: Decompression failed! err=-22, page=10450
[ 8509.130924] zram: Decompression failed! err=-22, page=138703
[ 8749.015625] zram: Decompression failed! err=-22, page=111561
[ 8749.015668] zram: Decompression failed! err=-22, page=111561
[ 8749.015681] Read-error on swap-device (254:0:892488)
[ 9635.264807] zram: Decompression failed! err=-22, page=18348
[10106.536377] zram: Decompression failed! err=-22, page=42178
**Additional context**
Add any other relevant context for the problem.

This seems relevant.
https://github.com/armbian/build/pull/2377/commits/e90a488f24d4ba7f0947930f485c4e69a09e9374
https://github.com/armbian/build/pull/2377

pelwell, 2020-12-11:
Duplicate (or partial duplicate) of https://github.com/raspberrypi/linux/issues/3839?

yutayu, 2020-12-11:
@pelwell
Duplicate. 
But , I thought I should post zram problem as cleary.
Excuse me.

yutayu, 2021-02-22:
https://github.com/raspberrypi/linux/commit/e91d8d78237de8d7120c320b3645b7100848f24d

closed 2021-02-22 by yutayu