raspberrypi/linux #4003: alsamixer settings of HifiBerry DAC+ ADC Pro do not persist through a reboot

jwoillez, 2020-12-14:
I have an HifiBerry DAC+ ADC Pro sound card on two raspberry pi running Raspberry Pi OS and derivatives. I am able to adjust the parameters and levels of the DAC and ADC with `alsamixer`, but these settings do not persist through a reboot.

Inspecting `journalctl`'s logs I found the following, confirming an issue with the restore process:
> Dec 14 12:37:14 zynthian systemd-udevd[192]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore 3' failed with exit code 22.

There is also the following, more descriptive messages:
> Dec 14 12:37:14 zynthian alsactl[434]: alsactl 1.1.8 daemon started
> Dec 14 12:37:14 zynthian alsactl[434]: /usr/sbin/alsactl: load_state:1674snd_config_load error: Invalid argument
> Dec 14 12:37:14 zynthian alsactl[434]: ALSA lib conf.c:1887:(_snd_config_load_with_include) _toplevel_:353:30:Unexpected char
> Dec 14 12:37:14 zynthian alsactl[435]: ALSA lib conf.c:1887:(_snd_config_load_with_include) _toplevel_:353:30:Unexpected char
> Dec 14 12:37:14 zynthian alsactl[435]: /usr/sbin/alsactl: load_state:1674: snd_config_load error: Invalid argument`

The unexpected char reported by `alsactl` can be traced to the following entry in `/var/lib/alsa/asound.state`
> value VINL1[SE]

Going quickly through the `alsactl` code, it seems that brackets (`[` and `]`) have a special meaning.

@j-schambacher, I'm guessing that changing the names of the enumerations for these problematic controls, in the hifiberry_dacplusadcpro driver, would solve the issue. With instructions, I could probably test this hypothesis.

See: https://github.com/raspberrypi/linux/blob/76c49e60e742d0bebd798be972d67dd3fd007691/sound/soc/bcm/hifiberry_dacplusadcpro.c#L65

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 4.

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

* Which firmware version (`vcgencmd version`)?
Aug 19 2020 17:38:16 
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux zynthian 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

hifiberry, 2020-12-15:
The brackets don't have a special meaning, this is simply a bug in alsactl that has been fixed some time ago. You need to upgrade alsa-utils. 
latest version 1.2.4 is working fine.

jwoillez, 2021-03-21:
@hifiberry Sorry for getting back to you only now. The issue is indeed fixed with 1.2.4. However Debian Buster is still running 1.1.8. Do you know the process to ask for a back port of this fix to Buster?