raspberrypi/linux #4008: Rpi4 kernel crash when using more than 34 USB serial ports

r-schmidt, 2020-12-17:
**Describe the bug**
The system prints kernel messages about an unhandled fault when accessing more than 34 serial ports (ttyUSB).

**To reproduce**
- connect more than 34 USB serial ports (see https://freetserv.github.io/ for such a system)
- open "screen" sessions to 35 of those ports ('for i in `seq 0 35`; do screen -d -m -S port${i} /dev/ttyUSB${i} 115200; done')

**Expected behaviour**
- the system should simply create the sessions and not crash

**Actual behaviour**
- crash with kernel messages

**System**
Raspberry Pi 4B

$ cat /etc/rpi-issue
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

$ vcgencmd version
Nov 30 2020 22:12:08 
Copyright (c) 2012 Broadcom
version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start)

$ uname -a
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

**Logs**
On the console is printed the following output:

[ 103.715803] 8<--- cut here ---
[ 103.720089] Unhandled fault: asynchronous external abort (0x1211) at 0x00000000
[ 103.728619] pgd = d30b99fe
[ 103.732532] [00000000] *pgd=17787003, *pmd=00000000
[ 103.738634] Internal error: : 1211 [#1] SMP ARM
[ 103.744388] Modules linked in: sha256_generic libsha256 cfg80211 rfkill 8021q garp stp llc ftdi_sio usbserial vc4 cec rpivid_mem v3d drm_kms_helper gpu_sched drm bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common drm_panel_orientation_quirks raspberrypi_hwmon snd_soc_core videodev mc vc_sm_cma(C) snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio ip_tables x_tables ipv6
[ 103.797884] CPU: 0 PID: 2336 Comm: sudo Tainted: G C 5.4.79-v7l+ #1373
[ 103.807232] Hardware name: BCM2711
[ 103.812161] PC is at lockref_put_return+0x50/0x9c
[ 103.818379] LR is at dput.part.8+0x58/0x3b8
[ 103.824052] pc : [] lr : [] psr: 60000013
[ 103.831839] sp : d6dbbe50 ip : 00000000 fp : d6dbbe6c
[ 103.838602] r10: d71f8c70 r9 : 00000001 r8 : 00080060
[ 103.845375] r7 : 00000001 r6 : 00030003 r5 : 00000000 r4 : 00030003
[ 103.853475] r3 : 00000001 r2 : 00030003 r1 : 00000064 r0 : d91b1fb8
[ 103.861570] Flags: nZCv IRQs on FIQs on Mode SVC_32 ISA ARM Segment user
[ 103.870286] Control: 30c5383d Table: 177ea000 DAC: fffffffd
[ 103.877630] Process sudo (pid: 2336, stack limit = 0xb7a13c6b)
[ 103.885085] Stack: (0xd6dbbe50 to 0xd6dbc000)
[ 103.891075] be40: d91b1f68 d91b1fb8 c0de3628 00080040
[ 103.900914] be60: d6dbbea4 d6dbbe70 c04287c0 c06a2f54 00000000 c040e3bc 00000000 d6c58180
[ 103.910762] be80: d71f8c70 080a801d ef32e850 d91b1f68 00000010 d71f8c70 d6dbbeb4 d6dbbea8
[ 103.920635] bea0: c0428b38 c0428774 d6dbbef4 d6dbbeb8 c040e42c c0428b2c 00000000 00000000
[ 103.930523] bec0: c020ce54 d6c58188 d6d0eff0 d6c58180 00000000 d6d0eac0 d6d0f01c c12b9d54
[ 103.940423] bee0: d6d0eff0 c1204fc8 d6dbbf04 d6dbbef8 c040e59c c040e34c d6dbbf2c d6dbbf08
[ 103.950351] bf00: c02433f8 c040e590 ffffe000 d6dbbfb0 c02011c4 00000004 fffffe30 5ac3c35a
[ 103.960296] bf20: d6dbbfac d6dbbf30 c020ce54 c0243350 d6dbbf4c d6dbbf40 5ac3c35a c1204fcc
[ 103.970269] bf40: d6dbbf6c d6dbbf50 c0407de8 c040eb18 00000004 d6d60600 d7a73c00 00000000
[ 103.980258] bf60: d6dbbf94 d6dbbf70 c042fcc0 c0407d84 02008150 b6e63b1c 00000000 8dd7b910
[ 103.990263] bf80: c02011c4 02008150 b6e63b1c 00000000 00000006 c02011c4 d6dba000 00000006
[ 104.000294] bfa0: 00000000 d6dbbfb0 c0201034 c020c8cc 00000000 00000444 b6e636d8 b6d83864
[ 104.010348] bfc0: 02008150 b6e63b1c 00000000 00000006 b66ff29c b66ff298 020082b8 00000000
[ 104.020416] bfe0: 00000000 bef0ef68 b6d85088 b6de8f74 20000010 00000004 00000000 00000000
[ 104.030494] Backtrace:
[ 104.034839] [] (lockref_put_return) from [] (dput.part.8+0x58/0x3b8)
[ 104.044880] r7:00080040 r6:c0de3628 r5:d91b1fb8 r4:d91b1f68
[ 104.052509] [] (dput.part.8) from [] (dput+0x18/0x1c)
[ 104.061293] r10:d71f8c70 r9:00000010 r8:d91b1f68 r7:ef32e850 r6:080a801d r5:d71f8c70
[ 104.071136] r4:d6c58180
[... 290 lines cut ...]
[ 273.426617] rcu: rcu_sched kthread starved for 13610 jiffies! g3593 f0x2 RCU_GP_DOING_FQS(6) ->state=0x0 ->cpu=0
[ 273.438413] rcu: RCU grace-period kthread stack dump:
[ 273.445088] rcu_sched R running task 0 10 2 0x00000000
[ 273.453755] Backtrace:
[ 273.457812] [] (__schedule) from [] (preempt_schedule_common+0x1c/0x34)
[ 273.467789] r10:c1203d00 r9:ef910000 r8:00000000 r7:c120508c r6:00000000 r5:00000000
[ 273.477251] r4:ffffe000
[ 273.481396] [] (preempt_schedule_common) from [] (_cond_resched+0x50/0x58)
[ 273.491646] r5:00000000 r4:00000001
[ 273.496841] [] (_cond_resched) from [] (rcu_gp_kthread+0x548/0xa40)
[ 273.506480] r5:00000000 r4:c12110c0
[ 273.511672] [] (rcu_gp_kthread) from [] (kthread+0x170/0x174)
[ 273.520777] r7:ef910000
[ 273.524921] [] (kthread) from [] (ret_from_fork+0x14/0x28)
[ 273.533763] Exception stack(0xef911fb0 to 0xef911ff8)
[ 273.540424] 1fa0: 00000000 00000000 00000000 00000000
[ 273.550232] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 273.560026] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 273.568242] r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0245e60
[ 273.577668] r4:ef86fcc0

**Additional context**
When connecting the USB serial ports to a different computer (my notebook) this works fine.
The original bug report is in the freetserv project:
https://github.com/freetserv/freetserv/issues/48

P33M, 2020-12-17:
Please post a full dmesg since boot (with the devices connected). Also post the output of lsusb -t.

I would expect the controller to reject more than 31 devices, because it has 32 device slots (minus 1 for the onboard hub).

r-schmidt, 2020-12-17:
$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.79-v7l+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:27:40 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 308224 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49804 r8192 d23924 u81920
[    0.000000] pcpu-alloc: s49804 r8192 d23924 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 502528
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:3F:EA:97 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=53c6fbe5-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x19400000-0x1d400000] (64MB)
[    0.000000] Memory: 1647796K/2019328K available (10240K kernel code, 739K rwdata, 2816K rodata, 2048K init, 854K bss, 109388K reserved, 262144K cma-reserved, 1232896K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31111 entries in 61 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x344/0x518 with crng_init=0
[    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000029] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000107] bcm2835: system timer (irq = 17)
[    0.000756] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000775] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000795] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000811] Switching to timer-based delay loop, resolution 18ns
[    0.001053] Console: colour dummy device 80x30
[    0.001549] printk: console [tty1] enabled
[    0.001616] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001659] pid_max: default: 32768 minimum: 301
[    0.001821] LSM: Security Framework initializing
[    0.002028] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002070] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003432] Disabling memory control group subsystem
[    0.003569] CPU: Testing write buffer coherency: ok
[    0.004086] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005010] Setting up static identity map for 0x200000 - 0x20003c
[    0.005233] rcu: Hierarchical SRCU implementation.
[    0.005933] smp: Bringing up secondary CPUs ...
[    0.007133] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008458] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009720] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009877] smp: Brought up 1 node, 4 CPUs
[    0.009948] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.009975] CPU: All CPU(s) started in HYP mode.
[    0.009999] CPU: Virtualization extensions available.
[    0.010832] devtmpfs: initialized
[    0.024409] VFP support v0.3: implementor 41 architecture 3 part 40 variant 8 rev 0
[    0.024687] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.024738] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.031552] pinctrl core: initialized pinctrl subsystem
[    0.032569] NET: Registered protocol family 16
[    0.036411] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.036996] audit: initializing netlink subsys (disabled)
[    0.037259] audit: type=2000 audit(0.030:1): state=initialized audit_enabled=0 res=1
[    0.038328] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.038364] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 493 lines cut ...]
                    |__ Port 3: Dev 14, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 3: Dev 14, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
[... cut ...]
                        |__ Port 2: Dev 17, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 2: Dev 17, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M

r-schmidt, 2020-12-17:
Now with correct indentation:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 12M
            |__ Port 1: Dev 4, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 4, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 4, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 4, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 5, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 5, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 5, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 5, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 4: Dev 7, If 0, Class=Hub, Driver=hub/4p, 12M
                |__ Port 1: Dev 8, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 1: Dev 8, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 1: Dev 8, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 1: Dev 8, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 9, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 9, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 9, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 9, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 10, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 10, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 10, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 10, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 4: Dev 11, If 0, Class=Hub, Driver=hub/4p, 12M
                    |__ Port 3: Dev 14, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 3: Dev 14, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 3: Dev 14, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 3: Dev 14, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 1: Dev 12, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 1: Dev 12, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 1: Dev 12, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 1: Dev 12, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 4: Dev 15, If 0, Class=Hub, Driver=hub/4p, 12M
                        |__ Port 3: Dev 18, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 3: Dev 18, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 3: Dev 18, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 3: Dev 18, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 1: Dev 16, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 1: Dev 16, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 1: Dev 16, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 1: Dev 16, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 2: Dev 17, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 2: Dev 17, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 2: Dev 17, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                        |__ Port 2: Dev 17, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                    |__ Port 2: Dev 13, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
```

P33M, 2020-12-18:
The first device in this huge chain is a TUSB2046 hub - which is running at full speed. All downstream devices have messages similar to `usb 1-1.4.4.4.4.2: not running at top speed; connect to a high speed hub`. Why is the first external hub in the chain at full-speed? This limits the entire bus segment to 12mbit/s, and in practice the useable bandwidth will be even lower because of the vast numbers of endpoints that will be active (i.e. polled regularly).

r-schmidt, 2020-12-18:
The TUSB2046B only supports full speed. So it is to be expected that it only runs at 12 MBit/s.
The 48 serial ports running at 115200 kBit/s would in total need 5529600 kBit/s.
That is less than half of the gross USB speed so there is still more than 50% left for the USB overhead.
And the crash happens even when there is nothing actually transmitting or receiving on those ports.

Also as I wrote in the bug report, the crash doesn't happen when the serial port chain is connected to my notebook.
So it is something specific to the Raspi.

P33M, 2020-12-18:
Weird hardware design decisions aside, the controller is crashing - not the kernel. The clue is 
```
[ 115.025783] pcieport 0000:00:00.0: AER: PCIe Bus Error: severity=Uncorrected (Non-Fatal), type=Transaction Layer, (Requester ID)
[ 115.046856] pcieport 0000:00:00.0: AER: device [14e4:2711] error status/mask=00004000/00000000
[ 115.058368] pcieport 0000:00:00.0: AER: [14] CmpltTO (First)
```
Which is the root complex timing out on a read transaction. 

It'll probably be easier/cheaper to buy a boatload of FT4232H adapters to replicate the issue rather than try and source this specialist hardware.

r-schmidt, 2020-12-18:
So the VL805 connected to the PCIe bridge crashes or enters a state that isn't handled?
I'm actually surprised to learn that this USB controller has a firmware. As a firmware developer myself I know how often limits in there are not properly checked because "that can never happen"...

You can't simply buy this HW, you would have to build it yourself, it is open hardware.
And one of the reasons to use a full speed hub chip (aside from the simpler board layout) is probably that the chances of getting high speed USB reliably across pin connectors are low.

So yes, buying 9 FT4232H adapters (e.g. Digitus DA-70159) and a few switches is the only other way to replicate it.
Still quite the investment, though. But definitely faster than getting the PCBs produced and assembled.
I'm happy to do whatever is needed to debug this. Though that has to wait until after New Years.

P33M, 2020-12-18:
The fact that a read returned a completion timeout indicates an access to one of the xhci controller registers (nominally hardware) failed.

It's not clear what the interaction is between the firmware and controller hardware in this case. We know it has supervisory functions and can e.g. write to power state control registers inside the PCIe endpoint based on controller activity, but any clarification would have to come from VIA.

r-schmidt, 2021-01-22:
To eliminate the HUB daisy chain as the cause for the issue I split the 48 ports into 4x12 ports and connected 3 of those so the USB tree looked like this:

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 7, If 0, Class=Hub, Driver=hub/4p, 12M
            |__ Port 2: Dev 9, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 9, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 9, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 9, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 10, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 10, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 10, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 10, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 8, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 8, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 8, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 8, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 3: Dev 11, If 0, Class=Hub, Driver=hub/4p, 12M
            |__ Port 1: Dev 12, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 12, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 12, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 12, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 13, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 13, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 13, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 13, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 14, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 14, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 14, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 14, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 4: Dev 3, If 0, Class=Hub, Driver=hub/4p, 12M
            |__ Port 2: Dev 5, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 5, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 5, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 5, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 3: Dev 6, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 4, If 2, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 4, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 4, If 3, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 1: Dev 4, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
```

The result was the same, with the 35th serial connection opened, the kernel driver prints errors and the system is unresponsive.

I then tested with a Rpi 1 (with some rather old Linux on it) and there it was working with the full 48 port chain, pretty slow but at least no hangs or crash.

Thinking that the Rpi 3B+ had a similar USB layout to the Rpi 1 (i.e. no PCIe device) I tried that one and the result was even worse than with the Rpi 4. The Rpi 3B+ silently hangs after about 10-20 seconds after connecting the chain.
The only indication of a problem was in dmesg before the system got unresponsive:

```
[ 2029.999363] usb 1-1.2.4.4.3: Detected FT4232H
[ 2030.010073] usb 1-1.2.4.4.3: FTDI USB Serial Device converter now attached to ttyUSB35
[ 2030.109891] usb 1-1.2.4.4.4: new full-speed USB device number 44 using dwc_otg
[ 2030.241726] usb 1-1.2.4.4.4: New USB device found, idVendor=0451, idProduct=2046, bcdDevice= 1.25
[ 2030.241742] usb 1-1.2.4.4.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 2030.242544] hub 1-1.2.4.4.4:1.0: USB hub found
[ 2030.243091] hub 1-1.2.4.4.4:1.0: 4 ports detected
[ 2030.559810] usb 1-1.2.4.4.4.1: new full-speed USB device number 45 using dwc_otg
[ 2030.690575] usb 1-1.2.4.4.4.1: not running at top speed; connect to a high speed hub
[ 2030.693822] usb 1-1.2.4.4.4.1: New USB device found, idVendor=0403, idProduct=6011, bcdDevice= 8.00
[ 2030.693831] usb 1-1.2.4.4.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2030.693838] usb 1-1.2.4.4.4.1: Product: Quad RS232-HS
[ 2030.693844] usb 1-1.2.4.4.4.1: Manufacturer: FTDI
[ 2030.694943] ftdi_sio 1-1.2.4.4.4.1:1.0: FTDI USB Serial Device converter detected
[ 2030.695052] usb 1-1.2.4.4.4.1: Detected FT4232H

[... cut ...]
[ 2050.880318] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 5
[ 2050.880642] ftdi_sio 1-1.2.4.4.4.1:1.1: FTDI USB Serial Device converter detected
[ 2050.880847] usb 1-1.2.4.4.4.1: Detected FT4232H
```

Reducing the chain to 36 ports everything was fine on the Rpi3B+. Trying to attach the last 12 ports separately again triggered the memory allocation error. After some searching the solution was simple: Adding _coherent_pool=2M_ to cmdline.txt allows the use of all 48 ports on the Rpi3B+.