raspberrypi/linux #4009: 5.10 breaks using cachefilesd & prevents login on raspi-os 64-bit.

satmandu, 2020-12-18:
**Describe the bug**
Having cachefilesd enabled gives me this error with the 5.10 kernel. (This wasn't happening with the earlier 5.4 kernels.):
```
[172295.694427] CacheFiles: 
[172295.694440] CacheFiles: Assertion failed
[172295.694476] ------------[ cut here ]------------
[172295.694482] kernel BUG at fs/cachefiles/rdwr.c:716!
[172295.694554] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[172295.694607] Modules linked in: rpcsec_gss_krb5 aes_neon_blk crypto_simd cryptd rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc lz4 lz4_compress 8021q garp stp llc zram zsmalloc xfs spidev vc4 cec drm_kms_helper drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops brcmfmac backlight raspberrypi_hwmon brcmutil sha256_generic i2c_bcm2835 cfg80211 rfkill bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) snd_bcm2835(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig snd_pcm videobuf2_memops videobuf2_v4l2 snd_timer videobuf2_common spi_bcm2835 snd videodev vc_sm_cma(C) mc uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[172295.695310] CPU: 1 PID: 23640 Comm: bash Tainted: G         C        5.10.0-v8+ #1382
[172295.696521] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
[172295.697721] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[172295.698928] pc : cachefiles_read_or_alloc_pages+0xbe0/0xc20
[172295.700127] lr : cachefiles_read_or_alloc_pages+0xbe0/0xc20
[172295.701307] sp : ffffffc011c13740
[172295.702472] x29: ffffffc011c13740 x28: ffffff800ff7c028 
[172295.703656] x27: ffffff800ff7c000 x26: ffffff8014e1b60c 
[172295.704833] x25: ffffff800ff7c000 x24: ffffffc011c1395c 
[172295.705998] x23: ffffffc0114d4000 x22: ffffffc011c13b08 
[172295.707144] x21: ffffff8011fbc8b0 x20: ffffff800f1afd00 
[172295.708270] x19: 0000000000000000 x18: 0000000000000000 
[172295.709389] x17: 0000000000000000 x16: 0000000000000000 
[172295.710483] x15: 0000000000000000 x14: 0000000000000000 
[172295.711558] x13: 0000000000000000 x12: ffffffc0112b5318 
[172295.712624] x11: 0000000000000003 x10: ffffffc01129d2d8 
[172295.713676] x9 : ffffffc0100e5380 x8 : 0000000000017fe8 
[172295.714736] x7 : c0000000ffffefff x6 : ffffffc011c133d0 
[172295.715805] x5 : ffffff803b1a18e0 x4 : 0000000000000000 
[172295.716855] x3 : 0000000000000001 x2 : ee2f1aea39596e00 
[172295.717889] x1 : 0000000000000000 x0 : 000000000000001c 
[172295.718892] Call trace:
[172295.719846]  cachefiles_read_or_alloc_pages+0xbe0/0xc20
[172295.720799]  __fscache_read_or_alloc_pages+0x374/0x5f8
[172295.721721]  __nfs_readpages_from_fscache+0x64/0x1b8
[172295.722615]  nfs_readpages+0xe0/0x228
[172295.723474]  read_pages+0x184/0x278
[172295.724291]  page_cache_ra_unbounded+0x148/0x220
[172295.725086]  do_page_cache_ra+0x48/0x58
[172295.725845]  ondemand_readahead+0x130/0x2d8
[172295.726582]  page_cache_sync_ra+0xec/0xf0
[172295.727292]  generic_file_buffered_read+0x50c/0x9d0
[172295.728000]  generic_file_read_iter+0x108/0x1a8
[172295.728691]  nfs_file_read+0x80/0x108
[172295.729393]  new_sync_read+0x10c/0x188
[172295.730077]  vfs_read+0x150/0x1d8
[172295.730758]  ksys_read+0x70/0xf8
[172295.731433]  __arm64_sys_read+0x24/0x30
[172295.732127]  el0_svc_common.constprop.2+0xac/0x1d8
[172295.732838]  do_el0_svc+0x2c/0x98
[172295.733550]  el0_svc+0x20/0x30
[172295.734250]  el0_sync_handler+0x90/0xb8
[172295.734966]  el0_sync+0x174/0x180
[172295.735689] Code: 9416d4d4 d00044c0 913a2000 9416d4d1 (d4210000) 
[172295.736451] ---[ end trace 2ff3c8a307cb6bc7 ]---
```
From another machine:
```
[   98.714190] CacheFiles: 
[   98.714214] CacheFiles: Assertion failed
[   98.714253] ------------[ cut here ]------------
[   98.714260] kernel BUG at fs/cachefiles/rdwr.c:716!
[   98.714268] Internal error: Oops - BUG: 0 [#6] PREEMPT SMP
[   98.714276] Modules linked in: uinput rfcomm fuse rpcsec_gss_krb5 aes_neon_blk crypto_simd cryptd bnep hci_uart btbcm bluetooth ecdh_generic ecc lz4 lz4_compress 8021q garp stp llc zram zsmalloc joydev brcmfmac brcmutil sha256_generic cfg80211 rfkill vc4 cec snd_soc_core snd_compress snd_pcm_dmaengine raspberrypi_hwmon bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) snd_bcm2835(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_pcm snd_timer videodev snd mc vc_sm_cma(C) uio_pdrv_genirq uio ads7846 drm_kms_helper drm drm_panel_orientation_quirks backlight fb_ili9486(C) fbtft(C) syscopyarea sysfillrect sysimgblt fb_sys_fops spi_bcm2835 i2c_dev ip_tables x_tables ipv6
[   98.714481] CPU: 2 PID: 1382 Comm: video_decoder Tainted: G      D WC        5.10.0-v8+ #1382
[   98.714487] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
[   98.714495] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[   98.714513] pc : cachefiles_read_or_alloc_pages+0xbe0/0xc20
[   98.714521] lr : cachefiles_read_or_alloc_pages+0xbe0/0xc20
[   98.714526] sp : ffffffc012ae3810
[   98.714532] x29: ffffffc012ae3810 x28: ffffff800cfe61a8 
[... 704 lines cut ...]
[172649.125205]  page_cache_ra_unbounded+0x148/0x220
[172649.126013]  do_page_cache_ra+0x48/0x58
[172649.126785]  ondemand_readahead+0x130/0x2d8
[172649.127533]  page_cache_sync_ra+0xec/0xf0
[172649.128258]  generic_file_buffered_read+0x50c/0x9d0
[172649.128981]  generic_file_read_iter+0x108/0x1a8
[172649.129682]  nfs_file_read+0x80/0x108
[172649.130396]  new_sync_read+0x10c/0x188
[172649.131091]  vfs_read+0x150/0x1d8
[172649.131783]  ksys_read+0x70/0xf8
[172649.132472]  __arm64_sys_read+0x24/0x30
[172649.133177]  el0_svc_common.constprop.2+0xac/0x1d8
[172649.133902]  do_el0_svc+0x2c/0x98
[172649.134626]  el0_svc+0x20/0x30
[172649.135335]  el0_sync_handler+0x90/0xb8
[172649.136062]  el0_sync+0x174/0x180
[172649.136783] Code: 9416d4d4 d00044c0 913a2000 9416d4d1 (d4210000) 
[172649.137546] ---[ end trace 2ff3c8a307cb6bc9 ]---
[172649.138323] note: bash[24072] exited with preempt_count 1
```
**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2020-12-18:
I think your "enable in /etc/default/cachefilesd" is a larger task than that simple phrase suggests. I'm not a cachefilesd user, and only an occasional nfs user. If you want us to dive into a problem with a non-standard configuration that is likely to be a general upstream problem then you'll need to give clear step-by-step instructions.

The immedate cause of failure appears to be the use of an inode from a filesystem that has no readpages method (or perhaps corruption of the datastructures). If I had a system that was failing like that I'd replace the ASSERT with some logging to print out the value of `inode->i_mapping->a_ops`, then search the output of `nm vmlinux` for that value.

satmandu, 2020-12-18:
That's fair. 

Just noticed this change in 5.10-rc2:
```
Matthew Wilcox (Oracle) (1):
      cachefiles: Handle readpage error correctly
```
so this does seem like an upstream problem and out of scope for a RPI change.  Thanks for helping me elucidate though! I'll take this upstream.

satmandu, 2020-12-18:
This appears to be abandoned upstream, so it's probably best to just drop usage of this package. I'm not having any issues otherwise with 5.10. :)

closed 2020-12-18 by satmandu

pelwell, 2020-12-18:
No problem - thanks for keeping us on our toes.

Cebtenzzre, 2021-02-06:
This is fixed in Linux 5.10.11 by torvalds/linux@db58465, which just removes the assert since it was never actually necessary.