raspberrypi/linux #4011: python3-pip extra-index-url

cwerther-truce, 2020-12-18:
This looks similar to an issue raised for pip (https://github.com/pypa/pip/issues/7486).  On my Raspberry Pi 3 running the latest version of Buster and pip3 version 18.1, when I try to install a package from my private index, it fails saying it can't find it on the public index.  If I uninstall the version installed via apt and install the latest version of pip (https://pip.pypa.io/en/stable/installing/) using the get-pip.py script, it installs pip version 20.3.3 which successfully installs the package.

XECDesign, 2020-12-18:
AFAIK, pip moves fairly fast, so the distro-packaged version will always be out of date. One of the first things pip users tend to do is run the pip command to update pip.

cwerther-truce, 2020-12-18:
Fair enough.  That being said, 18.1 is a 2018 release and its almost 2021 now.  In addition, pip officially recommends not to manually install pip and instead use the one provided by your package manager, so I posted this so that people using the official best practice (from pip at least) to have this work.

XECDesign, 2020-12-18:
Yeah, there tends to be a conflict between distros like Debian and fast moving projects like node.js, pip and so on.

This should be something that Debian fixes and if there's a patch that works, I can apply it in Raspberry Pi OS. But I can't pull in a whole new pip release, since I don't know what new issues it will introduce.

scottamain, 2021-01-21:
This is affecting me as well. According to [bug #6799](https://github.com/pypa/pip/issues/6799) this might have been fixed in [pr #6373](https://github.com/pypa/pip/pull/6373). And supposedly that was released with pip 19.1 and included with Debian Buster, so I don't understand why RPI OS still has pip 18.1. 

Please consider updating this package.