raspberrypi/linux #4014: libcamera does not work on kernel 5.10.1-v7l+

kamomil, 2020-12-20:
I followed the instructions to install libcamera on rpi. (I use rpi 4 model B and my sensor is imx219)
The `rpi-update` command leads to kernel `5.10.1-v7l+`
Then, when running the command:
`cam -c 1 -C`
I get the error:
```
ERROR V4L2 v4l2_videodevice.cpp:931 /dev/video13[out]: Unable to set format: Invalid argument
```
The dmesg says:
```
[87627.802495] bcm2835_mmal_vchiq: error -1 queuing message
[87627.808096] bcm2835-isp bcm2835-isp: populate_qdata_fmt: Failed vchiq_mmal_port_set_format on port, ret -1
```

6by9, 2020-12-21:
@naushir @davidplowman Are you now both on 5.10? If not then please do update.

naushir, 2020-12-21:
Hmmm, I've been running on 5.10 for a week or more, and I don't seem to have any problem.  Will do an update and try again.

naushir, 2020-12-21:
I do not have any issues with my current kernel build running with imx477.

Kernel built at commit 967d45b29ca2902f031b867809d72e3b3d623e7a
```
$ uname a
Linux pi4-naush 5.10.1-v7l+ #3 SMP Mon Dec 21 09:51:29 GMT 2020 armv7l GNU/Linux
```

libcamera tree built at commit 50ec902aa8b6b345b20529e338c2f09d72b99184

@kamomil, what sensor are you using here? And can you conform you are running with the top of tree commit in libcamera?