raspberrypi/linux #4015: CM3+ vc4 4.19.97 7" Powertip DSI Panel (ili9881C)

wudo94, 2020-12-21:
We want to connect the Powertip DSI Panel on the DSI1 port of a custom CM3+ baseboard.
The original RPI Display works fine, but on the Powertip Display we have some issues (see image below).

[image]

We added an overlay and modified the existing ili9881 panel driver to include the correct init commands. Please see the following branch for more details: https://github.com/wudo94/linux/tree/rpi-4.19.97_powertip_dsi.

If anyone can provide some help or information of similar setups this would be very much appreciated.

Thanks!

@fschrempf

6by9, 2020-12-21:
Can I strongly suggest you update to the rpi-5.10.y branch. The DRM/KMS drivers are being actively worked on there, and particularly for DSI. Backporting individual patches may or may not work, but we won't invest effort in debugging old kernel versions.

The rpi-5.10.y branch already has the driver and [overlay](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/vc4-kms-dsi-7inch-overlay.dts) merged for the standard Pi 7" DSI panel.

#3985 is a PR which added an example DSI panel to the system, and is actually using the ILI9881 driver. I ought to rebase it and get it merged.

How certain are you of your init sequence? Beyond 2/4 DSI data lanes, and 720x1280 vs 800x1280, I haven't analysed what is actually being changed in these tables for the different displays. Those which are merged should be valid, so it would be interesting to test with those rather than a table of potentially unknown content. I have known display manufacturers give out totally bogus timing and inits.

fschrempf, 2020-12-22:
> Can I strongly suggest you update to the rpi-5.10.y branch. The DRM/KMS drivers are being actively worked on there, and particularly for DSI. Backporting individual patches may or may not work, but we won't invest effort in debugging old kernel versions.

Ok, thanks we will try it with the rpi-5.10.y branch and see how it goes. It might take until after the holidays until we get to do it, though.

> How certain are you of your init sequence? Beyond 2/4 DSI data lanes, and 720x1280 vs 800x1280, I haven't analysed what is actually being changed in these tables for the different displays. Those [...]

We have a datasheet for the panel that gives exact byte-by-byte values for the init sequence, so we thought it would be save to assume these are correct. But we could probably also give it a try with some of the other tables. I guess we would need to adjust them to use 2 lanes instead of 4.

6by9, 2020-12-22:
I do have an init table for 2 lane ILI9881 at 720x1280 that i have had working on a display. I probably can't just release it, but I will do a quick diff to see what (if anything) is different between yours and mine.

6by9, 2020-12-23:
The init table for my screen exactly matches yours, except it's missing the 
```
	ILI9881C_SWITCH_PAGE_INSTR(0),
	ILI9881C_COMMAND_INSTR(0x11, 0x00),
```
at the end.

I have it working on 5.10. There does seem to be some race condition in my configuration, but blacklisting panel-ilitek-ili9881c and manually modprobing it has just loaded everything happily. I think that's down to my regulator configuration though. A magic delay somewhere is probably all that is needed.

6by9, 2020-12-23:
I am testing on a Pi4, but that shouldn't make a real difference with DSI configuration.

fschrempf, 2021-01-28:
> The init table for my screen exactly matches yours

That's interesting. I would have expected, that the resolution is also encoded somewhere in the init data. But on second thought and as sync pulse mode is used, I guess the display controller is able to decode the sync packages and therefore doesn't really need this information.

We are currently testing with latest rpi-5.10.y branch, but so far the results look pretty much as bad as above with the older kernel.

fschrempf, 2021-01-28:
> That's interesting. I would have expected, that the resolution is also encoded somewhere in the init data. But on second thought and as sync pulse mode is used, I guess the display controller is able to decode the sync packages and therefore doesn't really need this information.

Forget about this. Rereading your comment above I guess that the init data does contain the resolution, and it actually differs for your display, but everything else is the same.

fschrempf, 2021-01-28:
> The init table for my screen exactly matches yours, except it's missing the...

We now tested with 5.10 ([see this branch](https://github.com/wudo94/linux/tree/rpi-5.10.y_powertip_dsi)) and also removed the last two lines from the init table, but unfortunately the result is still similar as shown above. Do you have any ideas on how to do further debugging?

Also, what mode do you set for the DSI driver to use? I patched the panel driver to set the mode flags to `MIPI_DSI_MODE_VIDEO | MIPI_DSI_MODE_LPM | MIPI_DSI_MODE_VIDEO_SYNC_PULSE` ([see here](https://github.com/wudo94/linux/commit/48c97d253358104636fac198fe7a67610d4cd778#diff-f5eaf552a61705fbd1b7f8ea9160c9bd7381862ccd590e18216b1ec52f364f5eR883)).

wudo94, 2021-01-28:
On the Kernel 5.10.10 we have a limilar picture on our Powertip Display:
[image]

dmesg-Output of vc4:
[image]

The last lines of the config.txt:
```
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=vc4-kms-v3d
dtoverlay=rpi-ili9881
ignore_lcd=1
```

Here you can find the updated branch with the kernel 5.10.10: https://github.com/wudo94/linux/tree/rpi-5.10.y_powertip_dsi

Kefirr, 2021-03-03:
I try launch the same display but the display is still black. I tried with Kernel 5.10.10 and 4.19.97. Version of the Kernel 5.10.10 not loading DSI interface (VC4 driver). Older version of the Kernel 4.19.97 loading DSI interface but `dmesg` show message `4.705536] vc4-drm soc:gpu: failed to bind 3f700000.dsi (ops vc4_dsi_ops [vc4]): -517`

@wudo94 you connected display RESET line to GPIO4 on RPi? I use oscilloscope to watch RST line but the line is always on high state. The RST line never change state during loading custom driver `rpi-ili9881`. I base on your modifications.

wudo94, 2021-03-08:
You are correct. I connected the RESET to GPIO4.
On Startup I see two changes of the RESET-line:
[image]

 At the second change the display shows these lines at the first time, rest is black:
[image]

A few seconds later it seemst to be start the desktop and the display show the picture from my last comment.

There are some differences between init sequences of different displays with the ili9881.
The init sequence based on my repo is for the display Powertip PH720128T003-ZBC02.

john9969, 2022-06-12:
i'm using CM4 and 800x1280 LCD-TFT module with MIPI-DSI1 interface, capacitive touch panel and TP controller (I2C), i installed linux kernel (https://github.com/raspberrypi/linux) and i stucking in add driver ili9881C  in file config.txt 
If anyone can provide some help or information of similar setups this would be very much appreciated.