raspberrypi/linux #4017: Raspberry Pi 4 with Raspberry Pi OS (64-bit kernel : 5.10.1 : #1383) halt/poweroff/shutdown command issues

reraikes, 2020-12-21:
halt, reboot, poweroff, and shutdown commands don't work as defined in man pages

These commands power off when they should not:
halt
halt --halt
reboot --halt
poweroff --halt
shutdown -H now

These commands do not power off when they should:
halt -p
poweroff
reboot -p
poweroff -p
shuwdown now
shutdown -h now
shutdown -P now

pelwell, 2020-12-21:
Unless the 64-bit kernel performs differently to the 32-bit kernel in this regard, this is the wrong place for your issue.

pelwell, 2020-12-21:
Scratch that - I'll review the behaviour, but by this point it is so well established that changing it is likely to do more harm than good.

reraikes, 2020-12-21:
I don't believe poweroff operations have ever worked on 32-bit versions of Raspberry Pi OS (they all behave as halt).

On 64-bit versions of Raspberry Pi OS, halt and poweroff operations work but they are reversed:

halt -> poweroff / poweroff -> halt

mimiza, 2022-07-05:
I'm using Pi 4 8GB running Pi OS 64-bit. I can't neither shutdown nor reboot. THIS SUCKS!