raspberrypi/linux #4018: spi-bcm2835 driver fails in kernel 5.4.79 when cshigh is set as False in py-spidev

jfernandz, 2020-12-21:
**Describe the bug**
After talking with @cleverca22 at IRC we've concluded there have to be a problem with the `bcm2835` driver

**To reproduce**
Run this little script for testing the spi interface:

```
#!/usr/bin/env python3

import spidev
import time

spi = spidev.SpiDev()
spi.open(0, 0)

# Settings
frec = 20000
spi.cshigh = True
print(spi.cshigh)

i = 0
while i < 2000:

    spi.xfer2([i], frec)
    i += 1
    # print(i)
    time.sleep(1)

spi.close()
```

**Expected behaviour**
I expect to get some signal like 
[image]

**Actual behaviour**
I've got the following signal 
[image]

**System**
https://gist.github.com/jfernandz/d125063a7b49871af3d7aa8bcf61b149

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
We've been doing some researching and when I set `cshigh = False` I got the following error:

```
Traceback (most recent call last):
   File "./spi_test.py", line 11, in <module>
      spi.cshigh = False
SistemError: error return without exception set
```

This makes no sense because when `cshigh = True` I got the oposite what I expect and I cannot set it to `False`.

However, I've downloaded an older image for Raspbian OS with 4.19.x kernel and I have the expected behavior.

`cshigh = False`
[image]
[image]

`cshigh = True`
[image]
[image]

This issue has been treated in [py-spydev repo](https://github.com/doceme/py-spidev/issues/108) but it's not apparently a problem with the module itself but with the driver `spi-bcm2835`

Also I think people has been talking about the issue in here https://github.com/raspberrypi/linux/issues/3745 but I'm not sure this was solved.

Tomorrow I'll try the 5.10.x kernel (updated using `rpi-update`) but I'm not sure it will work, I'll keep you posted.

pelwell, 2020-12-22:
It's disappointing that you and @cleverca22 didn't read #3745 more carefully, because you would have found this:

> CS_HIGH handling has changed between 4.19 and 5.4 - the SPI framework forces CS_HIGH then leaves it up to the GPIO layer to handle the ACTIVE_LOW semantics. I don't like the change because it is (ab)using user-facing flags - CS_HIGH no longer means what you think it means. The attitude upstream (and I may be missing some nuance here) seems to be that CS_HIGH is a bus property rather than a device property - if you have to wait until an application starts to reverse the polarity of one of the CS pins then the device has possibly already been responding to and interfering with devices on other CS lines from the same
controller.
> 
> In short - dropping the device.cshigh = False line not only makes it work but is the right thing to do. Under other circumstances this kernel change might have broken other aspects of the py-spidev library, but I think the fact that it reads the initial value of the mode from spidev means that everything is fine unless you try to clear cshigh.

Since writing that my opinion of the upstream stance has softened a bit - CS polarity _is_ a bus property not a client/device property, and leaving it up to spi-dev is likely to bite you eventually. When I raised the problem with the upstream devs one of the responses was that they aren't interested in fixing spi-dev because it's an awful interface that needs to be replaced, but that doesn't help all the people stuck with it.

After a few more spi-dev problems I added a patch to our downstream kernel that causes any application change to the cshigh flag to be ignored - the value is restored after user settings have been applied. This means that `cshigh = True` still does nothing useful, but it allows the much more common `cshigh = False` to not break everything.

If you have a device that expects an active-high CS you need to change the gpio declaration in Device Tree:
```
		spi: spi0: spi@7e204000 {
			compatible = "brcm,bcm2835-spi";
			reg = <0x7e204000 0x200>;
			...
			cs-gpios = <&gpio 8 1>, <&gpio 7 1>;
```
The `1`s in the `cs-gpios` property indicate active-low - change either to a `0` to make it active-high. You could add `cs0_high` and `cs1_high` to the `spi0-2cs` overlay:
```
	__overrides__ {
		cs0_pin  = <&frag0>,"brcm,pins:0",
			   <&frag1>,"cs-gpios:4";
		cs1_pin  = <&frag0>,"brcm,pins:4",
			   <&frag1>,"cs-gpios:16";
		cs0_high  = <&frag1>,"cs-gpios:8=0";   // NEW
		cs1_high  = <&frag1>,"cs-gpios:20=0";  // NEW
		no_miso = <0>,"=2";
	};
```
And to use it, add something like the following to config.txt:
```
dtoverlay=spi0-2cs,cs1_high
```

cleverca22, 2020-12-22:
i had not seen https://github.com/raspberrypi/linux/issues/3745 and ive mentioned to @jfernandz a few times that i'm not familiar with the spi drivers under the pi, ive only delt with it a little bit in baremetal, and on AVR based chips

closed 2020-12-22 by jfernandz

jfernandz, 2020-12-22:
@pelwell I consider then this issue not a real issues so I've closed it. Anyway ... could not this be implemented in the device tree to just control the CS behavior in the `/boot/config.txt` file? (without further and manual modifications in the DT, I mean)

cleverca22, 2020-12-22:
yes, just create a new dtbo using https://www.raspberrypi.org/documentation/configuration/device-tree.md and what @pelwell gave above, then load it with dtoverlay=

jfernandz, 2020-12-22:
@cleverca22 but that modifications will produce a _selectable_ behavior? If so ... why is this not included by default in the DT?

pelwell, 2020-12-22:
If we put all the current optional behaviour in the base DTBs they would be quite large. Maintaining them as overlays keeps the size managaable and also allows us to avoid duplication. The only cases where we do build them in as `dtparam`s is for very small patches where the implementation has to vary between devices.

What I am happy to do is add the `cs<n>high` parameters to the standard `spi<n>-<n>cs` overlays.