raspberrypi/linux #4020: Full KMS does not work with pi4 and official touchscreen on kernel 5.10.1

maru-sama, 2020-12-22:
I recently moved from a Pi2 to a Pi4 and started using fake kms for output on the official touchscreen. (kivy console application) Looking at the changes for 5.10 I saw that now full KMS is supported for this combination so I updated the kernel to 5.10.1 via rpi-update and did a quick test.

config.txt
```
 ignore_lcd=1
 dtoverlay=vc4-kms-v3d  
 dtoverlay=vc4-kms-dsi-7inch
```
After a reboot the screen stayed blank (no rainbow) and during the boot-up flashed to white only to fade out afterwards.
In the dmesg I saw the following entries.

dmesg
```
[ 7.433705] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[ 7.433723] vc4_dsi fe700000.dsi: instat: 0x00000000
[ 7.433820] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[ 8.473696] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[ 8.473713] vc4_dsi fe700000.dsi: instat: 0x00000000
[ 8.473798] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[ 9.513694] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[ 9.513710] vc4_dsi fe700000.dsi: instat: 0x00000000
[ 9.513792] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
```
Please note fake KMS is working totally fine on the same hardware.

maru-sama, 2020-12-22:
Here a full DMESG output.
So apparently the screen is detected but does not reply?
```
    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.1-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1383 SMP Fri Dec 18 19:05:58 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000002fffffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  [mem 0x0000000030000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0 totalpages: 2061312
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 1864704 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s50572 r8192 d23156 u81920
[    0.000000] pcpu-alloc: s50572 r8192 d23156 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2059008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=DC:A6:32:D5:EA:2B vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait video=DSI-1:800x480@60,rotate=180
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000014c00000-0x0000000018c00000] (64MB)
[    0.000000] Memory: 7800092K/8245248K available (10240K kernel code, 1353K rwdata, 3152K rodata, 2048K init, 890K bss, 183012K reserved, 262144K cma-reserved, 7458816K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33808 entries in 67 pages
[    0.000000] ftrace: allocated 67 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3c8/0x59c with crng_init=0
[    0.000009] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000036] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000100] bcm2835: system timer (irq = 25)
[    0.000755] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000777] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000800] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000818] Switching to timer-based delay loop, resolution 18ns
[    0.001071] Console: colour dummy device 80x30
[    0.001839] printk: console [tty1] enabled
[    0.001905] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001960] pid_max: default: 32768 minimum: 301
[    0.002141] LSM: Security Framework initializing
[    0.002340] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002387] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.003977] Disabling memory control group subsystem
[    0.004115] CPU: Testing write buffer coherency: ok
[    0.004590] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.005564] Setting up static identity map for 0x200000 - 0x20003c
[    0.005799] rcu: Hierarchical SRCU implementation.
[    0.006733] smp: Bringing up secondary CPUs ...
[    0.007987] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 367 lines cut ...]
[   30.240294] Bluetooth: HCI socket layer initialized
[   30.240304] Bluetooth: L2CAP socket layer initialized
[   30.240319] Bluetooth: SCO socket layer initialized
[   30.250763] Bluetooth: HCI UART driver ver 2.3
[   30.250773] Bluetooth: HCI UART protocol H4 registered
[... cut ...]
[   47.574863] br-be0e9e6485fd: port 1(veth3b99087) entered disabled state
[   48.700241] cgroup: cgroup: disabling cgroup2 socket matching due to net_prio or net_cls activation
[   49.516350] eth0: renamed from veth83b355a
[   49.575831] IPv6: ADDRCONF(NETDEV_CHANGE): veth3b99087: link becomes ready
[   49.575942] br-be0e9e6485fd: port 1(veth3b99087) entered blocking state
[   49.575950] br-be0e9e6485fd: port 1(veth3b99087) entered forwarding state
[   49.576090] IPv6: ADDRCONF(NETDEV_CHANGE): br-be0e9e6485fd: link becomes ready
```

6by9, 2020-12-22:
Is your firmware up to date? ```vcgencmd version```
The main reason behind the timeouts was that the VPU (firmware) had been registering for the interrupts too, and was typically faster at responding to them than the ARM. The ARM therefore timed out. The transactions had actually been sent, but the kernel driver aborted at that point.

I've just had a niggle at the back of my mind as to whether that fix has indeed been merged. I think I may have thought I'd found an issue, but then not gone back to it once I'd convinced myself it was OK. I'll check later.

pelwell, 2020-12-22:
@6by9 This might jog your memory:
> just found issues getting DISP0 to work on CM4 with the DSI transfer side of this change.
WIP tag added for now.

maru-sama, 2020-12-22:
Hi,
```
vcgencmd version
Dec 15 2020 18:28:46
Copyright (c) 2012 Broadcom
version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start)
```
Looks recent

6by9, 2020-12-22:
@pelwell Not working today hence I wasn't firing up the VPN etc to check :-)
I thought adding ```ignore_lcd=1``` should have worked around that anyway, but maybe not. I'll be checking tomorrow.

@maru-sama Thanks - it was just one that I wanted to check to avoid a wild goose chase.

maru-sama, 2020-12-22:
It really looks like a race condition.

Using:
```
dtoverlay=rpi-ft5406
ignore_lcd=1
dtoverlay=vc4-kms-v3d,nocomposite,noaudio
dtoverlay=vc4-kms-dsi-7inch
```
Sometimes works from a cold boot. The brightness is not reduced though and the touch does not work. Rebooting always ends up in a non working setup.

6by9, 2020-12-22:
```dtoverlay=rpi-ft5406``` NO!

It shouldn't result in any harm, but that polls the firmware for the touchscreen data. The firmware is not longer involved in the process as everything is moved to the kernel.

Touch is via the edt-ft5406 overlay as per the README https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/README#L742
There is definitely a race condition there as the panel driver also includes power control to the display and touchscreen with no co-ordination with the touchscreen driver. During probe it tends not to have been enabled, and should the screen power down due to the screensaver etc then it kills the touchscreen power too.
The issue with probe can be overcome by blacklisting the edt-ft5x06 module and manually modprobing after the display is up, but you still need to avoid the screen powering down.

There are alternate drivers for the TC358762 as a bridge and the ATTiny regulator to solve these, but I haven't got the overlays fully working for those.

maru-sama, 2020-12-22:
Hmm,
You are right I forgot to change this when trying it out. Now apparently after a cold boot the display comes up BUT

- I still get the timeout messages in the logs <-- I still do not know if this really works
- The brightness is not set
- Screen blanking does not kick in on the console
- It looks like that the rgb channels are swapped. The AA fonts look jagged
- It does not survive a reboot, I get a pulsing white screen 
- The screen flickers a little bit

6by9, 2020-12-23:
Hmm, sounds very odd.
I've just tested with my 7" DSI panel on a Pi4 and it's fine. It is the earlier version of the panel, but I don't know if that will make a difference.

That driver doesn't register a backlight driver, so there is no backlight control - it should come on during boot and stay on. (The reworked drivers do register a backlight driver).

maru-sama, 2020-12-23:
Interesting that it works for you. In the meantime I had to revert back 5.4.79, since Bluetooth and wlan die after some uptime. The System was working fine otherwise. 
Could there be a general issue with 5.10 and my issue with the touchscreen just one manifestation of it?
Before I forgot on 5,10 also playing videos via the touch froze after around 30 seconds. Works perfectly fine with 5.4.79.

maru-sama, 2020-12-24:
Good Morning,

I updated to 5.10.2 and now it seems to work fine. I am still getting the timeouts in the logs, but the screen looks normal and the touchscreen works.

6by9, 2020-12-24:
The only thing I can think of is that mine is a first revision controller board which didn't support backlight dimming (just on/off). In theory the regulator/backlight controller is totally compatible, but it's just possible that there is something funny going on.

maru-sama, 2020-12-24:
I still get the interrupt errors though, is this something you see as well?

maru-sama, 2020-12-24:
Are you trying this on the console or X? I added a video= to the cmdline to be able to see the output.

mingzhangqun, 2021-01-06:
@maru-sama
Hi, Have you solved the problem yet?
I have the same problem as you with version 5.10.4-v7l+.

> [    8.362692] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    8.620286] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.620752] bcmgenet fd580000.ethernet eth0: Link is Down
[    9.994390] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[    9.994407] vc4_dsi fe700000.dsi: instat: 0x00000000
[    9.994506] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[   11.034356] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[   11.034373] vc4_dsi fe700000.dsi: instat: 0x00000000
[   11.034468] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[   12.234366] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[   12.234384] vc4_dsi fe700000.dsi: instat: 0x00000000
[   12.234480] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[   13.900398] broken atomic modeset userspace detected, disabling atomic

maru-sama, 2021-01-06:
No,
For the time being I moved back to fkms to be able to use the screen with my rpi4.

vianpl, 2021-01-13:
Hi running 5.10.6-v7l+ (32f9280932f948ecd43), RPi4b on a rather old 7" display, with:

```
ignore_lcd=1
dtoverlay=vc4-kms-v3d-pi4
dtoverlay=vc4-kms-dsi-7inch
```

I get the following output (only the relevant bits):

```
[    4.784819] i2c i2c-11: Added multiplexed i2c bus 0
[    4.785688] i2c i2c-11: Added multiplexed i2c bus 10
[    4.988954] [drm] Initialized v3d 1.0.0 20180419 for fec00000.v3d on minor 0
[    5.070882] debugfs: Directory 'fef00700.hdmi' with parent 'vc4-hdmi-0' already present!
[    5.072524] vc4-drm gpu: bound fef00700.hdmi (ops vc4_hdmi_ops [vc4])
[    5.087282] debugfs: Directory 'fef05700.hdmi' with parent 'vc4-hdmi-1' already present!
[    5.089385] vc4-drm gpu: bound fef05700.hdmi (ops vc4_hdmi_ops [vc4])
[    5.325124] debugfs: Directory 'fef00700.hdmi' with parent 'vc4-hdmi-0' already present!
[    5.333232] vc4-drm gpu: bound fef00700.hdmi (ops vc4_hdmi_ops [vc4])
[    5.347156] debugfs: Directory 'fef05700.hdmi' with parent 'vc4-hdmi-1' already present!
[    5.349425] vc4-drm gpu: bound fef05700.hdmi (ops vc4_hdmi_ops [vc4])
[    5.392014] debugfs: Directory 'fef00700.hdmi' with parent 'vc4-hdmi-0' already present!
[    5.393618] vc4-drm gpu: bound fef00700.hdmi (ops vc4_hdmi_ops [vc4])
[    5.399523] debugfs: Directory 'fef05700.hdmi' with parent 'vc4-hdmi-1' already present!
[    5.401735] vc4-drm gpu: bound fef05700.hdmi (ops vc4_hdmi_ops [vc4])
[    5.936155] rpi_touchscreen 10-0045: Atmel I2C read failed: -5
```
Works fine with fkms.

6by9, 2021-01-13:
Now I've completed a critical piece of work I'm investigating this.
It did work when I created the PR adding the overlays and driver tweaks, but it's not working for me now either on our rpi-5.9.y or rpi-5.10.y branches. I know I checked it on probably 5.10.3 or similar when I made my comments earlier in the thread.

Two issues:

```rpi_touchscreen 10-0045: Atmel I2C read failed: -5``` is the Atmel controller having hung with SDA pulled low. You'll see all addresses reporting if you use ```sudo i2cdetect -y 10``` in this case, and the display needs to be power cycled.
Using the kernel drivers seems to get it into this "upset" mode more frequently than the firmware.

```vc4_dsi fe700000.dsi: transfer interrupt wait timeout``` is the stranger one as there should be very little stopping the LP mode transmission of the commands, and the interrupt triggers when that transmission is complete. It's possible if the display isn't powered that it could be holding the DSI data lines in a particular state, but that should be reported as LP-11 or LP-00 contention.

...

Syncing back to afe38f9dbeef7ffffafc9db963602167b5bdbc33 which was what was pushed as PR #3954 and it has just worked fine.

My other realisation is that I have almost always had a camera module connected to the camera connector on the Pi. It shouldn't make a difference, but they do share an I2C bus.

maru-sama, 2021-07-14:
Hello,

Is there an update on this issue? I recently set-up a new rpi4 and totally forgot about this issue. Took my some time to figure out why it was not working.

JamesH65, 2021-07-14:
It's all in progress, there has been  a lot of churn on this part of the codebase, but its getting there.

tenderlove, 2021-09-03:
Hi, I've been running in to this issue too.  I'm seeing this in dmesg:

```
[    5.981293] vc4_dsi fe700000.dsi: instat: 0x00000000
[    5.981348] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[    7.005264] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[    7.005278] vc4_dsi fe700000.dsi: instat: 0x00000000
[    7.005323] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[    8.029262] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[    8.029275] vc4_dsi fe700000.dsi: instat: 0x00000000
[    8.029317] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[    9.053267] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[    9.053281] vc4_dsi fe700000.dsi: instat: 0x00000000
[    9.053324] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[   10.173267] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[   10.173280] vc4_dsi fe700000.dsi: instat: 0x00000000
[   10.173320] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
```

From `/proc/cpuinfo`:

```
Hardware	: BCM2835
Revision	: d03114
Serial		: 1000000030d2e3e5
Model		: Raspberry Pi 4 Model B Rev 1.4
```

Also one interesting datapoint is that I'm running with the 64bit OS:

```
pi@cyberdeck:~ $ uname -a
Linux cyberdeck 5.10.60-v8+ #1449 SMP PREEMPT Wed Aug 25 15:01:33 BST 2021 aarch64 GNU/Linux
pi@cyberdeck:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

If there's anything I can do to help with this issue, please let me know!

tenderlove, 2021-09-03:
I forgot to mention, this is in my `config.txt`:

```
ignore_lcd=1
dtoverlay=vc4-kms-v3d,nocomposite,noaudio
dtoverlay=vc4-kms-dsi-7inch
max_framebuffers=2
arm_64bit=1
```

maru-sama, 2021-11-11:
Hi, with bullseye now available I want to know if there has been progress on this since as far as I understood bullseye is switching to the real KMS and not fkms.

JamesH65, 2021-11-11:
This should be working. We have had some reports of issues on Pi3 and earlier, but AFAIK, Pi4 should work OK, although you will need to add the appropriate dtoverlay entry in config.txt to turn on the display. I think it is dtoverlay=vc4-kms-dsi-7inch

pelwell, 2021-11-11:
Note that the `config,txt` file supplied with Bullseye images includes `display_auto_detect=1`, causing the firmware to check for the official display and, if found, automatically load the `vc4-kms-dsi-7inch` overlay.

JamesH65, 2021-11-11:
Ahh, thought there would be something I had missed, thanks @pelwell

maru-sama, 2021-11-25:
Ok, i tried it after updating to bullseye and while I now have a display I still have the following errors with my setup here.

I cannot rotate the display. With fkms I can do

dtoverlay=rpi-ft5406,touchscreen-inverted-x=1,touchscreen-inverted-y=1

and the touchscreen gets rotated correctly. 
The second more grave issue for me is that I use the touchscreen with a python kivy app with uses SDL2 for the window WITHOUT X11.

With fkms I see

> [INFO   ] Window: Provider: sdl2
[INFO   ] GL: Using the "OpenGL" graphics system
[INFO   ] GL: Backend used <sdl2>
[INFO   ] GL: OpenGL version <b'OpenGL ES 3.1 Mesa 20.3.5'>
[INFO   ] GL: OpenGL vendor <b'Broadcom'>
[INFO   ] GL: OpenGL renderer <b'V3D 4.2'>
[INFO   ] GL: OpenGL parsed version: 3, 1
[INFO   ] GL: Shading version <b'OpenGL ES GLSL ES 3.10'>
[INFO   ] GL: Texture max size <4096>
[INFO   ] GL: Texture max units <16>

With the kms provider I get an error
> [INFO   ] Window: Provider: sdl2
[CRITICAL] Window: Unable to find any valuable Window provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes

Both issues stop me from using it in production (installed in the kitchen to control our sonos system :)

maru-sama, 2021-11-25:
Hi,

A short update on this one, adding the following entries to config.txt made it work

> dtoverlay=vc4-kms-v3d,nocomposite,noaudio
dtoverlay=vc4-kms-dsi-7inch,invx=1,invy=1

closed 2021-11-25 by maru-sama

StefanGubler, 2021-12-08:
I've tried this on a CM4 and it's not working. Can you please provide the full config.txt ?

Very strange, that Raspberry Pi does not support his own display

maru-sama, 2021-12-10:
Hi, the only settings that are enabled in my config are
```
dtparm=i2c_arm=on
start_x=0
enable_uart=0
[all]
dtoverlay=vc4-kms-v3d,nocomposite,noaudio
dtoverlay=vc4-kms-dsi-7inch,invx=1,invy=1
```