raspberrypi/linux #4023: vchiq is not working properly on 5.10 in 64-bit mode

pelwell, 2020-12-22:
PRs #4016 and #4022 are fixes for bulk transfer in 5.10, but even with one of them applied there's something wrong with vchiq bulk transfers in 64-bit builds. rpi-5.4.y had no such problems.

lategoodbye, 2020-12-29:
Could you please provide a more detailed scenario? Is v5.4 really the last working major version?

pelwell, 2020-12-29:
I don't know when the breakage occurred - creating this issue is the last thing I did before going on leave, and investigating the problem is likely to be the first thing I do on my return.

pelwell, 2021-01-04:
It should be fixed now:
```diff
Author: Phil Elwell <phil@raspberrypi.com>
Date:   Mon Jan 4 11:06:51 2021 +0000

    staging: vchiq: Fix bulk transfers on 64-bit builds
    
    The recent change to the bulk transfer compat function missed the fact
    the relevant ioctl command is VCHIQ_IOC_QUEUE_BULK_TRANSMIT32, not
    VCHIQ_IOC_QUEUE_BULK_TRANSMIT, as any attempt to send a bulk block
    to the VPU would have shown.
    
    Fixes: a4367cd2b231 ("staging: vchiq: convert compat bulk transfer")
    
    Signed-off-by: Phil Elwell <phil@raspberrypi.com>

diff --git a/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_arm.c b/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_arm.c
index e8e2e70ba314..a936102dbc34 100644
--- a/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_arm.c
+++ b/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_arm.c
@@ -1722,7 +1722,7 @@ vchiq_compat_ioctl_queue_bulk(struct file *file,
 {
        struct vchiq_queue_bulk_transfer32 args32;
        struct vchiq_queue_bulk_transfer args;
-       enum vchiq_bulk_dir dir = (cmd == VCHIQ_IOC_QUEUE_BULK_TRANSMIT) ?
+       enum vchiq_bulk_dir dir = (cmd == VCHIQ_IOC_QUEUE_BULK_TRANSMIT32) ?
                                  VCHIQ_BULK_TRANSMIT : VCHIQ_BULK_RECEIVE;
 
        if (copy_from_user(&args32, argp, sizeof(args32)))
```

pelwell, 2021-01-04:
Fixes have been sent upstream: http://lists.infradead.org/pipermail/linux-rpi-kernel/2021-January/007947.html