raspberrypi/linux #4024: firefox illegal instruction pi zero/zero w

racedowling, 2020-12-23:
raspberry pi zero (w) 
latest release  Linux GPS 5.4.79+ #1373 Mon Nov 23 13:18:15 GMT 2020 armv6l GNU/Linux\|

firefox-esr fails with illegal instruction when when accessing /usr/lib/firefox-esr/libxul.so Apparently this library is not compatible with arm6l

Starting program: /usr/bin/firefox-esr 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb69ff440 (LWP 3749)]
[Thread 0xb69ff440 (LWP 3749) exited]

Thread 1 "firefox-esr" received signal SIGILL, Illegal instruction.
0xae5fc094 in ?? () from /usr/lib/firefox-esr/libxul.so

ledlamp, 2021-04-11:
same with chromium :c

reukiodo, 2022-01-19:
I thought the whole purpose of keeping Raspbian 32bit was to continue supporting the original Raspberry Pi ?
How did this pass QA testing?

JamesH65, 2022-01-19:
You are commenting on a year old thread, are things still the same now? Note, Firefox is not our recommended browser, Chrome/Chromium is the one we test and optimise.

reukiodo, 2022-01-19:
I just installed firefox-esr on a fresh rasbian for raspberry pi 1b and got this error. After following a google trail finally found this bug and made this comment, so it's still very relevant.

JamesH65, 2022-01-20:
It seems unlikely we would do anything about this as Firefox is not a package we specifically support. We do have a version of Chrome that runs on ARMv6, but tbh, running any sort of modern browser on a core that old with that little memory is not going to give a good experience, if it works at all.

XECDesign, 2022-01-20:
I don't think that armv6 chromium builds will be going further for much longer. Epiphany seems to be the least (but still very) painful browser for armv6 right now.

reukiodo, 2022-02-28:
But why compile it expressly prohibiting those who want to run it on pi1/pi0 hardware? I believe that they are still supported hardware, so why not compile the software with the hardware supported?
Or at the very least, just leave the last working version as the latest to install.

JamesH65, 2022-02-28:
Because we get extra performance on the newer platforms by compiling code targeted specifically at them. The newer Arm cores have extra instructions, registers etc that means they will perform better with the appropriate compiler flags.

andrum99, 2022-02-28:
> But why compile it expressly prohibiting those who want to run it on pi1/pi0 hardware? I believe that they are still supported hardware, so why not compile the software with the hardware supported? Or at the very least, just leave the last working version as the latest to install.

For certain values of 'working'. Running Chromium on a Pi with only 512MB of RAM is not going to be a particularly productive experience. Keeping old builds of Chromium available for these boards via apt exposes users to security vulnerabilities, although note that Raspberry Pi have not said that they would stop shipping older builds for these boards.

XECDesign, 2022-02-28:
It's not like we're flipping a switch to say 'support armv7 only'. Support is dropping upstream and in Raspbian. Although we're still patching things to make them work on armv6 on Raspberry Pi OS, at some point that will start taking more time than we can justify.

reukiodo, 2022-03-01:
Oh, I thought these packages were compiled by the RPi team... are they compiled by upstream instead?

andrum99, 2022-03-01:
> Oh, I thought these packages were compiled by the RPi team... are they compiled by upstream instead?

On 32-bit Raspberry Pi OS most of the packages are built by the Raspbian project, which is a separate project not controlled by Raspberry Pi. (It did not help that Raspberry Pi used to call their distribution Raspbian, but that's another discussion). Debian is the upstream for the Raspbian project.

The reason Raspbian exists is that Debian does not compile anything for the combination of ARMv6 with the particular hardware floating point that the BCM2835 uses, so for Raspberry Pi a volunteer in the community stepped in and started the Raspbian project.

XECDesign, 2022-03-01:
It's all a bit too much to explain in full detail. But the issue isn't about compiling the software, it's about how that software is written, tested and packaged.