raspberrypi/linux #4028: compile failure with 5.10.2-patches ontop of Debian-5.10 and gcc-10

laroche, 2020-12-27:
**Describe the bug**
Compiling the raspberry pi patches ontop of a Debian 5.10 kernel (with Debian testing, so a gcc-10)
gives the following compiler error:

drivers/gpu/drm/ast/ast_mode.c:851:18: error: initialization of ‘void (*)(struct drm_crtc *, struct drm_atomic_state *)’ from incompatible pointer type ‘void (*)(struct drm_crtc *, struct drm_crtc_state *)’ [-Werror=incompatible-pointer-types]
  851 |  .atomic_flush = ast_crtc_helper_atomic_flush,
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/gpu/drm/ast/ast_mode.c:851:18: note: (near initialization for ‘ast_crtc_helper_funcs.atomic_flush’)

best regards,

Florian La Roche

popcornmix, 2020-12-27:
Are you building with our defconfig files? (e.g. bcm2711_defconfig)
That file isn't built for me (as AST isn't relevant for Pi and so CONFIG_DRM_AST is not set).

laroche, 2021-01-16:
I am building stock Debian sources with your patches applied. So no bcm2711_defconfig.
Instead of a fix I've also disabled DRM_AST now.

If this is not something you will follow up on, feel free to close this issue.

best regards,

Florian La Roche

closed 2021-01-17 by popcornmix

fvhovell, 2021-12-30:
Can be fixed by cherry-picking commit f6ebe9f9c9233, with the caveat to use merge-strategy `ours`, because of some rpi-specific changes unrelated to above compile error.

So: `git cherry-pick --strategy-option=ours f6ebe9f9c9233`