raspberrypi/linux #4030: V3D is disabled after apt-get upgrade on Pi 3B+

danielaregay, 2020-12-27:
Before the upgrade, Rtl_Airband was working like a charm. After I upgraded the software mid-December 2020, I started getting the message "**Unable to enable V3D. Please check your firmware is up to date.**"

I have tried updating several times and everything is updated. Any possible solutions?

popcornmix, 2020-12-28:
That message appears to come from the qpu accelerated hello_fft library (that I assume is used by rtl_airband).

What model of Pi are you using?
Can you post your config.txt settings? Running with cutdown firmware (gpu_mem=16) or vc4-kms-v3d driver would cause this error.

danielaregay, 2020-12-28:
I am using Pi 3B+. And yes, it uses hello_fft library. Here below is the config.txt settings. And thank you!!

_______________________________________________________________________________________________________________________________________
 # For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
use_gpu=False
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-kms-v3d
max_framebuffers=2

[all]
dtoverlay=vc4-kms-v3d
max_framebuffers=2
framebuffer_priority=2

popcornmix, 2020-12-28:
As I said, "dtoverlay=vc4-kms-v3d" uses the arm side kms driver (which includes the 3d shaders).
That is incompatible with firmware side 3d shader use (used by hello_fft).

You'll need to switch back to `dtoverlay=vc4-fkms-v3d` driver to use hello_fft based apps.

`use_gpu=False` isn't a setting for config.txt - you'll need to check the Rtl_Airband docs for where that should be used.

danielaregay, 2020-12-28:
> As I said, "dtoverlay=vc4-kms-v3d" uses the arm side kms driver (which includes the 3d shaders).
> That is incompatible with firmware side 3d shader use (used by hello_fft).
>
> You'll need to switch [...]

I did and changed it to the following. But still getting the same error.

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[all]
dtoverlay=vc4-fkms-v3d

6by9, 2020-12-28:
Minor correction to popcornmix.
vc4-fkms-v3d will also transfer the V3D block from the firmware to ARM, so hello_fft or similar will be unable to use it.

You will need to remove or comment out either ```dtoverlay=vc4-kms-v3d``` or ```dtoverlay=vc4-fkms-v3d``` to drop back to the legacy framebuffer driver and firmware GLES driver.

danielaregay, 2020-12-28:
> Minor correction to popcornmix.
> vc4-fkms-v3d will also transfer the V3D block from the firmware to ARM, so hello_fft or similar will be unable to use it.
>
> You will need to remove or comment out [...]

I tried removing those but still showing the message "Unable to enable V3D. Please update your firmware"

6by9, 2020-12-28:
Just to check (as you haven't provided the details), you are running a standard 32bit Raspberry Pi OS install? The GLES driver is not supported on 64bit OSes.
```uname -a``` and ```cat /etc/os-release```

danielaregay, 2020-12-28:
> Just to check (as you haven't provided the details), you are running a standard 32bit Raspberry Pi OS install? The GLES driver is not supported on 64bit OSes.
> `uname -a` and `cat /etc/os-release`

**uname -a**
Linux atc 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux

**cat /etc/os-release**
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

danielaregay, 2020-12-28:
Also the software Rtl_Airband worked smoothly before the apt-get upgrade. So I'm not sure if the problem has to do with it being 32 or 64 bit

6by9, 2020-12-28:
You're using a 32bit OS, so that's not the issue.

popcornmix, 2020-12-28:
From searching I believe `use_gpu=False` goes in overrides.cfg rather than config.txt.

danielaregay, 2020-12-28:
> From searching I believe `use_gpu=False` goes in overrides.cfg rather than config.txt.

I have already done that previously and is still on the overrides.cfg file

danielaregay, 2020-12-29:
> From searching I believe `use_gpu=False` goes in overrides.cfg rather than config.txt.

Anyways I formatted it and did a fresh install and everything works now. Thank you @popcornmix and @6by9 for your help. Much appreciated!!!

closed 2020-12-29 by danielaregay