raspberrypi/linux #4031: Cannot enable. Maybe the USB cable is bad?

luckyer, 2020-12-29:
**Describe the bug**
I am currently want to usb a FT230X on my raspberrypi, and i got the following errors 
[  237.460134] usb 1-1.3: new high-speed USB device number 23 using xhci_hcd
[  238.410762] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  239.320708] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  239.321211] usb 1-1-port3: attempt power cycle
[  240.580713] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  241.490802] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  241.491352] usb 1-1-port3: unable to enumerate USB device

This problem happeds everytime when i plug in the usb, and this is the kernel version i use now 
pi@asiair:~ $ uname -r
4.19.66-v7l+
What i am using is PI4 b 

And more logs is here
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.66-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP Thu Aug 15 12:02:08 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1011008
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:46:64:15 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=b52378d8-02 rootfstype=ext4 elevator=deadline fsck.repair=no rootwait fastboot noswap ro
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 3735592K/4050944K available (8192K kernel code, 661K rwdata, 2348K rodata, 2048K init, 850K bss, 53208K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 662 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28118 entries in 83 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000022] Switching to timer-based delay loop, resolution 18ns
[    0.000246] Console: colour dummy device 80x30
[    0.000709] console [tty1] enabled
[    0.000760] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.000803] pid_max: default: 32768 minimum: 301
[    0.001079] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001112] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001876] CPU: Testing write buffer coherency: ok
[    0.002298] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002948] Setting up static identity map for 0x200000 - 0x20003c
[    0.003121] rcu: Hierarchical SRCU implementation.
[    0.003999] smp: Bringing up secondary CPUs ...
[    0.004814] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... 261 lines cut ...]
[    5.382425] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.392972] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.393656] usbcore: registered new interface driver brcmfmac
[    5.659457] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.672728] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    6.087751] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    6.442412] EXT4-fs (mmcblk0p3): recovery complete
[    6.464300] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[    7.440432] random: crng init done
[    7.440447] random: 7 urandom warning(s) missed due to ratelimiting
[    7.554734] 8021q: 802.1Q VLAN Support v1.8
[    7.758281] bcmgenet: Skipping UMAC reset
[    7.851893] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[    7.880447] brcmfmac: power management disabled
[    8.871752] bcmgenet fd580000.genet eth0: Link is Down
[    9.051131] brcmfmac: power management disabled
[    9.103456] brcmfmac: power management disabled
[   13.031785] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   16.055838] Adding 2097148k swap on /dev/mmcblk0p4.  Priority:-2 extents:1 across:2097148k SSFS
[  155.380068] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  156.290171] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  156.290686] usb 1-1-port3: attempt power cycle
[  157.550381] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  158.480463] usb 1-1-port3: Cannot enable. Maybe the USB cable is bad?
[  158.481027] usb 1-1-port3: unable to enumerate USB device

luckyer, 2020-12-29:
Plus , i use the same usb cable on my ubuntu, it work fine . So , anyone have ideas? tks

popcornmix, 2020-12-31:
> pi@asiair:~ $ uname -r
4.19.66-v7l+

That's very old. Can you update and test again? You should be on a 5.4 kernel.