raspberrypi/linux #4032: Conventions on new raspberry pi overlays

menschel, 2020-12-29:
Hi, i'd like to start a discussion on this Issue.

Some of the recent device tree overlays for raspberry pi were implemented in a general / modular way, so they can be adapted by parameter to specific hardware. 

Therefore the question needs to be asked what the rules/conventions for new overlays are?

For me in particular the mcp251xfd overlay raised that question.

I recently bought a [CAN FD HAT](https://www.seeedstudio.com/CAN-BUS-FD-HAT-for-Raspberry-Pi-p-4742.html) that would technically need a hardware specific device tree overlay.

However it can be set up with just 3 lines in /boot/config.txt .

```
#the can chips
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi0-1,interrupt=24

#the rtc chip
dtoverlay=i2c-rtc,pcf85063
```

From developers point of view, it would be feasible to include the "general" overlays and just override parameters.

pelwell, 2020-12-29:
That's a good question. There isn't a way of packaging up a collection of other overlays by name. You could theoretically create a small .txt file that includes the lines above and get users to `include` it into config.txt, but then you have another a file to distribute.

Fortunately, overlays tend not to be large, so we don't mind extending the list of "official" overlays. Also, there is a precedent here - the `upstream` overlay, which is a combination of the `vc4-kms-v3d` and `dwc2` overlays. To avoid having to maintain a special combination of the two overlays by hand I wrote the `ovmerge` (https://github.com/raspberrypi/utils/blob/master/ovmerge/ovmerge) utility to apply overlays in source form, and so avoid losing labels etc.

To combine your overlays using ovmerge:
```
$ ovmerge mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi0-1,interrupt=25 i2c-rtc-overlay.dts,pcf85063 > can_fd_hat-overlay.dts
```
This can then be submitted here as a PR.

It would be even better if the origin of this overlay was made clear, in case any of the source overlays were to change in the future, and this can be done with the `-c` option to ovmerge, which will cause it to include a comment at the top of the overlay saying:
```
// redo: ovmerge -c mcp251xfd-overlay.dts,spi0-0,interrupt=25 mcp251xfd-overlay.dts,spi0-1,interrupt=25 i2c-rtc-overlay.dts,pcf85063
```
The matching `-r` option reads the comment from an overlay file and uses the same overlays and parameters to regenerate the output. This is something which gets run on our kernel source periodically, so changes should be picked up and merged back into the source tree.

menschel, 2020-12-29:
Thank you very much.

I was not aware that my usecase already has been thought of.
This actually makes things quite maintainable and less error prone.

So the contributors task is to 
1. combine the "general" overlays with `ovmerge -c` into the hardware specific overlay,
2. test the hardware specific overlay,
3. submit the hardware specific overlay as a PR.

pelwell, 2020-12-29:
That works for me.

menschel, 2020-12-30:
A paragraph describing this topic has been added to README, therefore closing this issue.
Thanks again, always a pleasure.

closed 2020-12-30 by menschel