raspberrypi/linux #4033: RPI4 - aarch64 kernel 5.10.3, gpu freeze when using latest popcornmix gbm branch when hw decoding non-hevc streams

nkichukov, 2020-12-29:
**Describe the bug**
When playing non-hevc stream with HW acceleration on kodi matrix from latest popcornmix gbm branch, the GPU freezes and raspberrypi requires cold reset.

**To reproduce**
Compile rpi 5.10.3 kernel for aarch64
Compile latest gbm branch from popcornmix xbmc repository
Have kodi configured to HW decode even non-hevc streams (default)
Play any stream different than hevc and observe how the gpu crashes instantly

**Expected behaviour**
No GPU crash/freeze is expected

**Actual behaviour**
The GPU freezes and there are no more updates in the video buffers. All other functionalities, except for video output continue to work fine. System load increases, so far up to 6, perhaps more if the system is not cold reset.

**System**

* Which model of Raspberry Pi? Pi4, 4GB
* Which OS and version : GNU/Gentoo aarch64
* Which firmware version (`vcgencmd version`)?
Dec 15 2020 18:28:46 
Copyright (c) 2012 Broadcom
version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start)
* Which kernel version (`uname -a`)?
kernel 5.10.3

**Logs**
2 dmesg log sections when the issue is reproduced are attached, see bottom.

**Additional context**
The issue does not occur when playing HEVC streams.
The issue does not occur if kodi is configured not to use HW decoding for non-hevc streams. So the same streams work fine and does not cause the freeze/crash if software decoding them.

bootcode firmware:
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Fri Dec 11 11:15:17 AM UTC 2020 (1607685317)
 LATEST: Fri Dec 11 11:15:17 AM UTC 2020 (1607685317)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

Kodi userspace is 32-bit and it runs from within privileged lxc container.

[crash1.txt](https://github.com/raspberrypi/linux/files/5751428/crash1.txt)
[chrash2.txt](https://github.com/raspberrypi/linux/files/5751431/chrash2.txt)

lukasMega, 2021-01-02:
I confirm. Same issue here on Raspbian.
`5.10.3-v7l+ #1386 SMP Sun Dec 27 14:05:40 GMT 2020 armv7l`
freezes after 5-10 seconds after starting video playback (VLC Player or Chromium - YouTube) on 32" LCD with resolution 2560x1440

popcornmix, 2021-01-04:
@lukasMega your issue is not related. VLC/Chromium use a very different interface to hw decode than kodi (which OP is using gbm/v4l2). Please open a new issue.

popcornmix, 2021-01-04:
@nkichukov for me on 32-bit RPiOS h264 decode (e.g. of [bbb](https://mirror.clarkson.edu/blender/demo/movies/BBB/bbb_sunflower_1080p_30fps_normal.mp4) works. It also works with arm_64bit=1. e.g.
```
pi@pi4:~ $ uname -a
Linux domnfs 5.10.4-v8+ #1387 SMP PREEMPT Mon Jan 4 11:45:26 GMT 2021 aarch64 GNU/Linux
```
I assume your aarch64 gentoo is 64-bit kernel and userland? HW decode with 64-bit userland is not something we have been directly testing (but there is work ongoing).

@6by9 I know mmal isn't supported on 64-bit userland but I assume using v4l2/mmal doesn't have that restriction, so would this be expected to work?

6by9, 2021-01-04:
> I know mmal isn't supported on 64-bit userland but I assume using v4l2/mmal doesn't have that restriction, so would this be expected to work?

Correct. V4L2 wrapping MMAL works on both 32 and 64bit systems.

nkichukov, 2021-01-04:
Thanks for the responses.

It does run aarch64 kernel, however xbmc runs within 32bit userspace so that I can make use of the proprietary libwidevinecdm library which is only available for arm at this moment.

I will try 5.10.4 kernel with the bbb_sunflower example and will report results (with HW decode turned on).

nkichukov, 2021-01-04:
It crashed again, so I was able to reproduce it with 5.10.4 kernel and the same video that you used from the link above.

Is it possible to be a combination of running 64-bit kernel and 32-bit userspace?

What are the next steps to debug this further?

nkichukov, 2021-01-04:
[crash3-5.10.4-kernel.dmesg.log](https://github.com/raspberrypi/linux/files/5766125/crash3-5.10.4-kernel.dmesg.log)

popcornmix, 2021-01-04:
I've tested with arm_64bit=1 on RPiOS (so 32-bit userland, 64-bit kernel) and it worked.
Does it fail repeatably or is it random?
The kernel log suggests it is taking out the firmware.

Are you building with internal or external ffmpeg?
You should end up with an ffmpeg tree matching [drm_prime_1](https://github.com/jc-kynesim/rpi-ffmpeg/tree/dev/4.3.1/drm_prime_1). I've just pushed to popcormix/gbm tree so internal ffmpeg matches that version.

nkichukov, 2021-01-04:
It fails every time, so no randomness. It does not take to seek to fail as it is almost instant, within the first 1 second of playback.

I am building with 'internal' ffmpeg from the gbm branch. Now I do see that I am some 62 commits behind, so I will catch-up and recompile kodi.

Below are the use flags for the ebuild:
USE="alsa gbm gles udev webserver xslt -X -airplay -bluetooth -bluray -caps -cec -css -dbus -dvd -lcms -libressl -libusb -lirc -mariadb -mysql -nfs -opengl -pulseaudio -raspberry-pi -samba -system-ffmpeg -systemd -test -udf -udisks -upnp -upower (-vaapi) (-vdpau) -wayland -zeroconf" PYTHON_SINGLE_TARGET="python3_8 -python3_9"

No matter what, even if that xbmc code change fixes the problem, the kernel should not hang like this, but be able to survive this when it comes from userland/firmware.

Will follow-up once I test the new xbmc code changes.

nkichukov, 2021-01-04:
The new code did change the behaviour a bit and the kernel message, playback lasted for about 20 seconds before it crashed. The kernel errorred:

```
[20555.441580] ------------[ cut here ]------------
[20555.441594] Firmware transaction timeout
[20555.441678] WARNING: CPU: 3 PID: 27585 at drivers/firmware/raspberrypi.c:64 rpi_firmware_transaction+0xdc/0x110
[20555.441685] Modules linked in: veth iptable_nat xt_nat nf_nat nf_conntrack nf_defrag_ipv4 ip_tables x_tables nfsd bridge stp llc binfmt_misc bfq brcmfmac sha256_generic libsha256 sha256_arm64 vc4 cfg80211 v3d rfkill raspberrypi_hwmon gpu_sched brcmutil hwmon snd_soc_core snd_bcm2835(C) snd_pcm_dmaengine snd_pcm bcm2835_isp(C) bcm2835_codec(C) bcm2835_v4l2(C) snd_timer bcm2835_mmal_vchiq(C) snd rpivid_hevc(C) videobuf2_dma_contig vc_sm_cma(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common drm_kms_helper evdev joydev drm drm_panel_orientation_quirks syscopyarea videodev sysfillrect sysimgblt fb_sys_fops mc uio_pdrv_genirq uio fixed
[20555.441994] CPU: 3 PID: 27585 Comm: kworker/3:2 Tainted: G         C        5.10.4-v8+ #1
[20555.442003] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[20555.442026] Workqueue: events get_values_poll [raspberrypi_hwmon]
[20555.442041] pstate: 00000005 (nzcv daif -PAN -UAO -TCO BTYPE=--)
[20555.442053] pc : rpi_firmware_transaction+0xdc/0x110
[20555.442063] lr : rpi_firmware_transaction+0xdc/0x110
[20555.442070] sp : ffffffc0222a3c60
[20555.442077] x29: ffffffc0222a3c60 x28: 0000000000000000 
[20555.442093] x27: ffffff8058389ec8 x26: ffffffc0114b5008 
[20555.442108] x25: 0000000000001000 x24: ffffff80648b7800 
[20555.442122] x23: ffffff80408920c0 x22: ffffffc010f034b0 
[20555.442136] x21: ffffff806f201d80 x20: ffffff8040892080 
[20555.442151] x19: 00000000ffffff92 x18: 0000000000000000 
[20555.442165] x17: 0000000000000000 x16: 0000000000000000 
[20555.442180] x15: 0000000000000000 x14: 0000000000000000 
[20555.442195] x13: 0000000000000297 x12: 0000000000000000 
[20555.442210] x11: 00000000000002a5 x10: ffffffc010e60e58 
[20555.442225] x9 : ffffffc0100c9800 x8 : ffffffc010e32ae8 
[20555.442240] x7 : ffffffc010e5eae8 x6 : ffffffc010e19000 
[20555.442254] x5 : 0000000000000001 x4 : ffffff80fb7e65f0 
[20555.442269] x3 : 0000000000000000 x2 : 0000000000000001 
[20555.442284] x1 : c0de68b158d1b000 x0 : 0000000000000000 
[20555.442300] Call trace:
[20555.442312]  rpi_firmware_transaction+0xdc/0x110
[20555.442323]  rpi_firmware_property_list+0xb8/0x170
[20555.442332]  rpi_firmware_property+0x74/0x104
[20555.442344]  get_values_poll+0x50/0x154 [raspberrypi_hwmon]
[20555.442357]  process_one_work+0x1d8/0x4b0
[20555.442367]  worker_thread+0x13c/0x460
[20555.442377]  kthread+0x144/0x150
[20555.442388]  ret_from_fork+0x10/0x20
[20555.442395] ---[ end trace 010275ab420256db ]---
[20555.442432] hwmon hwmon0: Failed to get throttled (-110)
[20564.209752] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:87:crtc-0] flip_done timed out
[20565.585643] raspberrypi-clk soc:firmware:clocks: Failed to change fw-clk-v3d frequency: -110
[20576.753723] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:87:crtc-0] flip_done timed out
[20586.993748] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [PLANE:31:plane-0] flip_done timed out
[20697.329990] bcm2835_audio bcm2835_audio: vchi message timeout, msg=5
```

kodi log, attached here:
[kodi.log](https://github.com/raspberrypi/linux/files/5767040/kodi.log)

popcornmix, 2021-01-05:
> Below are the use flags for the ebuild:
USE="alsa gbm gles udev webserver xslt -X -airplay -bluetooth -bluray -caps -cec -css -dbus -dvd -lcms -libressl -libusb -lirc -mariadb -mysql -nfs -opengl -pulseaudio -raspberry-pi -samba -system-ffmpeg -systemd -test -udf -udisks -upnp -upower (-vaapi) (-vdpau) -wayland -zeroconf" PYTHON_SINGLE_TARGET="python3_8 -python3_9"

I don't know how these map to kodi build options. I build with:
```
cmake ../kodi_gbm -DCMAKE_BUILD_TYPE=Debug -DCMAKE_INSTALL_PREFIX=/usr/local -DCORE_PLATFORM_NAME=gbm -DAPP_RENDER_SYSTEM=gles -DENABLE_VAAPI=OFF -DENABLE_VDPAU=OFF -DENABLE_DEBUGFISSION=OFF -DENABLE_TESTING=OFF -DENABLE_INTERNAL_SPDLOG=ON -DENABLE_INTERNAL_FMT=ON
```
I'd have thought you didn't want -raspberry-pi as gbm build is generic without specific raspberry pi platform code.
I also don't use pulse-audio with kodi.

Just in case it's clock/voltage related, can you try with "force_turbo=1" and (separately) with "over_voltage=2".

nkichukov, 2021-01-24:
Thanks, the firmware upgrade fixed the problem (from the related thread  -https://github.com/raspberrypi/linux/issues/4047#issuecomment-758806428)

closed 2021-01-24 by nkichukov