raspberrypi/linux #4035: Pi4B kernel 5.10.3 module aes-arm-bs hangs during insertion (IPsec)

hailfinger, 2020-12-30:
**Describe the bug**
The 5.10.3 kernel does boot on my Raspberry Pi 4 4GB, but IPsec doesn't work anymore. I tracked it down to dozens of processes waiting/hanging, all the same:
`/sbin/modprobe -q -- crypto-cbc(aes)`
Killing the modprobe processes is possible but loading the module will hang.
crypto-cbc(aes) is an alias of aes-arm-bs. 
The crypto-cbc(aes) module is pulled in by the kernel during IPsec setup.

**To reproduce**
To replicate the hang, just run
`modprobe aes-arm-bs`
That command will hang. I admit I didn't wait more than 20 minutes for it to complete, but I expect module loading to be done in milliseconds, not minutes.

**Expected behaviour**
`modprobe aes-arm-bs`
should not hang. It doesn't hang on kernel 5.4.79, where IPsec works just fine.

**Actual behaviour**
`modprobe aes-arm-bs`
hangs. The module initstate is "coming", but it never changes to "live". You can check by running
`cat /sys/module/aes_arm_bs/initstate`

**System**

* Which model of Raspberry Pi?
   * Pi 4 Model B with 4 GB RAM
* Which OS and version (`cat /etc/rpi-issue`)?
   * Raspberry Pi reference 2020-08-20
   * Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2
* Which firmware version (`vcgencmd version`)?
   * Dec 15 2020 18:29:44
   * Copyright (c) 2012 Broadcom
   * version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start_x)
* Which kernel version (`uname -a`)?
   *  Linux pi4 5.10.3-v7l+ #1386 SMP Sun Dec 27 14:05:40 GMT 2020 armv7l GNU/Linux

**Logs**
Unfortunately dmesg doesn't log any message.

**Additional context**
If you start strongSwan on a kernel with this problem, you will see the following message in the logs:
`charon too long to start... - kill kill`

hailfinger, 2020-12-30:
This might be fixed in 5.10.4:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v5.10.4&id=6aa4660cba6115ab1c6ad109b1101aefde45b7a5

popcornmix, 2021-01-04:
Looks like it's fixed on 5.10.4 (in latest rpi-update kernel)
```
pi@pi4:~ $ uname -a
Linux domnfs 5.10.4-v7l+ #1387 SMP Mon Jan 4 11:40:03 GMT 2021 armv7l GNU/Linux
pi@pi4:~ $ sudo modprobe aes-arm-bs
pi@pi4:~ $ 
```
Please test and close if happy.

hailfinger, 2021-03-07:
Thank you, this is fixed now.

closed 2021-03-07 by hailfinger