raspberrypi/linux #4036: Recursive dependency detected with bmc2711_defconfig on rpi-5.10.y branch

graysky2, 2020-12-31:
**Describe the bug**
The bmc2711_defconfig target causes a recursive dependency.

**To reproduce**
1. Checkout https://github.com/raspberrypi/linux/commit/7f8da0ec27634043873f470ef6c9986ef36924d4
2. `make bcm2711_defconfig`

**Expected behaviour**
The make target should create a `.config` without errors

**Actual behaviour**
The make target ends in errors:
```
% make bcm2711_defconfig
drivers/staging/vc04_services/vc-sm-cma/Kconfig:1:error: recursive dependency detected!
drivers/staging/vc04_services/vc-sm-cma/Kconfig:1:	symbol BCM_VC_SM_CMA is selected by BCM2835_VCHIQ_MMAL
drivers/staging/vc04_services/vchiq-mmal/Kconfig:1:	symbol BCM2835_VCHIQ_MMAL depends on BCM2835_VCHIQ
drivers/staging/vc04_services/Kconfig:14:	symbol BCM2835_VCHIQ is selected by BCM_VC_SM_CMA
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

make[1]: *** [scripts/kconfig/Makefile:88: bcm2711_defconfig] Error 1
make: *** [Makefile:602: bcm2711_defconfig] Error 2
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? Pi4B
* Which OS and version (`cat /etc/rpi-issue`)? Arch ARM aarch64
* Which firmware version (`vcgencmd version`)? `Dec 15 2020 18:28:46 
Copyright (c) 2012 Broadcom
version 4b4aff21f72c5b9ba39d83c7b0f8fa910a6ef99b (clean) (release) (start)`
* Which kernel version (`uname -a`)? `Linux workbench 5.10.3-1-ARCH #1 SMP PREEMPT Sun Dec 27 09:06:37 EST 2020 aarch64 GNU/Linux`

Dark-Sky, 2020-12-31:
I get the same also.

Exaga, 2020-12-31:
Ditto on Raspberry Pi (1) with posts above:

 Creating new .config from bcmrpi_defconfig ...
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
drivers/staging/vc04_services/vc-sm-cma/Kconfig:1:error: recursive dependency detected!
drivers/staging/vc04_services/vc-sm-cma/Kconfig:1:      symbol BCM_VC_SM_CMA is selected by BCM2835_VCHIQ_MMAL
drivers/staging/vc04_services/vchiq-mmal/Kconfig:1:     symbol BCM2835_VCHIQ_MMAL depends on BCM2835_VCHIQ
drivers/staging/vc04_services/Kconfig:14:       symbol BCM2835_VCHIQ is selected by BCM_VC_SM_CMA
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

scripts/kconfig/Makefile:88: recipe for target 'bcmrpi_defconfig' failed
make[1]: *** [bcmrpi_defconfig] Error 1
Makefile:602: recipe for target 'bcmrpi_defconfig' failed
make: *** [bcmrpi_defconfig] Error 2

ptTrR, 2020-12-31:
Same issue. There is something wrong.

geerlingguy, 2020-12-31:
Just adding a ditto—after pulling latest today my compiles started failing. Reverting to my previous branch checkout fixes it.

ncopa, 2021-01-01:
This is caused by f1ef95a09bd0c6c9a4a9c63abb650b4eccc28d61 from RPI conflicting with 
b4db71c2d7af0b4ce06d3036444e303bd4a501ad from upstream.

Reverting either of those fixes it.

pelwell, 2021-01-01:
Fixed by #4038.

Dark-Sky, 2021-01-01:
I confirm with the 7e7689a032a66685b6a045b75265579bae838a00 commit it will now compile.

graysky2, 2021-01-01:
Agreed, fixed, thanks!

closed 2021-01-01 by graysky2