raspberrypi/linux #4040: bcm2835-v4l2-codec.c has logic error identified by clang.

iucoen, 2021-01-02:
**Describe the bug**

When compiling the kernel with clang, clang warns:

`/home/mcui/workspace/rpi-kernel/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c:911:7: warning: logical not is only applied to the left hand side of this bitwise operator [-Wlogical-not-parentheses]
                if (!mmal_buf->mmal_flags & MMAL_BUFFER_HEADER_FLAG_EOS) {
                    ^                     ~
/home/mcui/workspace/rpi-kernel/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c:911:7: note: add parentheses after the '!' to evaluate the bitwise operator first
                if (!mmal_buf->mmal_flags & MMAL_BUFFER_HEADER_FLAG_EOS) {
                    ^
/home/mcui/workspace/rpi-kernel/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c:911:7: note: add parentheses around left hand side expression to silence this warning
                if (!mmal_buf->mmal_flags & MMAL_BUFFER_HEADER_FLAG_EOS) {
                    ^
/home/mcui/workspace/rpi-kernel/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c:1137:6: warning: logical not is only applied to the left hand side of this bitwise operator [-Wlogical-not-parentheses]
        if (!fmt->flags & V4L2_FMT_FLAG_COMPRESSED) {
            ^           ~
/home/mcui/workspace/rpi-kernel/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c:1137:6: note: add parentheses after the '!' to evaluate the bitwise operator first
        if (!fmt->flags & V4L2_FMT_FLAG_COMPRESSED) {
            ^
/home/mcui/workspace/rpi-kernel/drivers/staging/vc04_services/bcm2835-codec/bcm2835-v4l2-codec.c:1137:6: note: add parentheses around left hand side expression to silence this warning
        if (!fmt->flags & V4L2_FMT_FLAG_COMPRESSED) {
            ^
`

The warning identifies a legitimate bug. Parentheses are needed around operator&.

**To reproduce**
LLVM=1 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make bcm2709_defconfig
LLVM=1 ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make -j$(nproc) zImage modules dtbs

**System**
 Cross compiling from Ubuntu-20.04 x86_64

popcornmix, 2021-01-03:
I agree - looks like a genuine bug in the code. @6by9?

6by9, 2021-01-03:
Yes, patch in progress. It may explain some of the oddities that @jc-kynesim sometimes sees too.

closed 2021-01-05 by iucoen