raspberrypi/linux #4043: RPI 4 does not allow 2 screens and 3,5 mm audio jack

andrendahl, 2021-01-03:
Before the last Raspberry update/upgrade I had 2 sceeens running and using the audio jack at the same time. Bug reports on the Raspberry forum did not give anything else that I had to wait for a new upgrade and firmware. Some mentioned that 3 outputs would not be possible before kernel 5.10 arrived. Further it is mentioned that the current RPI software only support 2 outputs.

Data
Processor: BCM2711

Distribution: Raspbian GNU/Linux 10 (buster)

Kernel version: Linux 5.4.79-v7l+ armv7l

Firmware: #1373

Package(s): 0 upgradable(s)

When will it be possible to use 2 screens and audi jack at the same time again?

pelwell, 2021-01-03:
This problem is fixed in new firmwares. If you want to remain on a 5.4 kernel then you can run `sudo rpi-update 43022f5a` to get the last 5.4 release, which contains the fix.

andrendahl, 2021-01-03:
It was run, and all broke down including the sd card! 
It was installed in the terminal.
RPI shut down.
The 2nd terminal connected.
RPI did not start, I actually doubt that it started the boot sequence.

andrendahl, 2021-01-03:
This is what I wrote on the issue:

It was run, and all broke down including the sd card!
It was installed in the terminal.
RPI shut down.
The 2nd terminal connected.
RPI did not start, I actually doubt that it started the boot sequence.

andrendahl, 2021-01-04:
Further info:
The sd card went from 64Gb to 121Mb, and even gparted is not able to restore it.

pelwell, 2021-01-04:
rpi-update is a simple script that copies files around. It doesn't touch the partition table or the raw SD card. You have unfortunately suffered a card malfunction of some kind - perhaps because of a problem with the power supply.

pelwell, 2021-01-04:
I had no problems running the same rpi-update command here on a 4B.

andrendahl, 2021-01-07:
Thanks, I am in the process to install a new SD card and will test it again.

pelwell, 2021-01-07:
We did since find a problem on older 4Bs (those with 2711B0 SoCs) that could cause problems during audio playback. That is fixed in yesterday's rpi-update release.

andrendahl, 2021-01-07:
Unfortunately another new sd card were broken. Here is a copy of the commands:

pi@raspberrypi:~ $ sudo rpi-update 43022f5a
*** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
*** Performing self-update
*** Relaunching after update
*** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
*** We're running for the first time
*** Backing up files (this will take a few minutes)
*** Backing up firmware
*** Backing up modules 5.4.79-v7l+
#############################################################
WARNING: This update bumps to rpi-5.4.y linux tree
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769
'rpi-update' should only be used if there is a specific
reason to do so - for example, a request by a Raspberry Pi
engineer or if you want to help the testing effort
and are comfortable with restoring if there are regressions.
DO NOT use 'rpi-update' as part of a regular update process.
##############################################################
Would you like to proceed? (y/N)
*** Downloading specific firmware revision (this will take a few minutes)
% Total % Received % Xferd Average Speed Time Time Time Current
Dload Upload Total Spent Left Speed
100 168 100 168 0 0 908 0 --:--:-- --:--:-- --:--:-- 908
100 117M 0 117M 0 0 1163k 0 --:--:-- 0:01:43 --:--:-- 1408k
*** Updating firmware
*** Updating kernel modules
*** depmod 5.4.83+
*** depmod 5.4.83-v7+
*** depmod 5.4.83-v7l+
*** depmod 5.4.83-v8+
*** Updating VideoCore libraries
*** Using HardFP libraries
*** Updating SDK
*** Running ldconfig
*** Storing current firmware revision
*** Deleting downloaded files
*** Syncing changes to disk
*** If no errors appeared, your firmware was successfully updated to

43022f5adbbf5d6e05ea0e022a3090c2c9feff7c
*** A reboot is needed to activate the new firmware

Do you have ant ideas of what went wrong ?

pelwell, 2021-01-07:
So you've updated to the older release that still has the audio problem - I was thinking you would just `sudo rpi-update` - but I wouldn't expect that to prevent the Pi from booting. Exactly what happens when you (re)boot?

andrendahl, 2021-01-07:
Nothing, screen was black

pelwell, 2021-01-07:
Put the SD card in a card reader attached to another computer. Edit `cmdline.txt` and replace `quiet` with `xquiet` (or delete it altogether), then try booting the Pi - you might see some messages from the kernel. Also watch (or video) the green LED to see how long it takes for the activity to stop.

andrendahl, 2021-01-07:
I know it might sound strange, but this sd card is as well broken as the first one which I have tried almost (probably) all possibilities to rescue.
SD card here in Norway are expensive and I can not use any further for this project.  If it is correct that kernel 5.10 will solve this issue I do think that I will wait for that.

But thank you very much for your time and work for trying to help me. It is really appreciated.

andrendahl, 2021-01-08:
Is there any chance for that kernel 5.10 will solve the issue regarding several outputs like screens, audiojack mic etc ?

popcornmix, 2021-01-10:
The fix for two hdmi displays + analogue audio (which ran out of DMA channels) was a firmware side fix so is fixed with latest releases on 5.4 and 5.10.

andrendahl, 2021-01-11:
Maybe this will solve this issue. Using two monitors and the bluetooth for music at the same time. As far as I can see bluetooth is not dependant on the DMA ?