raspberrypi/linux #4044: [rpi 4 4gb] - kernel 5.10]  720 not working kms

jamieS95, 2021-01-03:
**Is this the right place for my bug report?**
not sure

**Describe the bug**
720 not working kms

**To reproduce**
added to config:
hdmi_group=1
hdmi_mode=3
dtoverlay=vc4-kms-v3d

changed resoultion to 720 in raspi-config

xrandr reports: 1080p 

**Expected behaviour**
720 resolution

**Actual behaviour**
1080 resolution

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? 4 4gb
* Which OS and version (`cat /etc/rpi-issue`)? newest raspi-update version 
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)? kernel 5.10

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

6by9, 2021-01-03:
With the switch to DRM/KMS many of the config.txt parameters are being replaced. ```hdmi_mode``` and ```hdmi_group``` fall into this category with regard use within X.  (They should still set the resolution of the console via the kernel command line, but X ignores this).

Use the Screen Configuration application within X and the setting should persist over reboots.

jamieS95, 2021-01-03:
Screen Configuration application worked to change resolution

closed 2021-01-03 by jamieS95