raspberrypi/linux #4048: Kernel Panic under Heavy Network Load

benpalm, 2021-01-05:
**Describe the bug**
Under heavy network load I get a kernel panic not syncing. Happens when torrenting a file via transmission-daemon to an connected usb-drive. Seems similar to issue #2538. Tried rpi-update, but the issue is reproducable 

**System**
Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2

Linux raspberrypi 5.10.4-v7l+ #1387 SMP Mon Jan 4 11:40:03 GMT 2021 armv7l GNU/Linux
Revision        : d03114
Serial          : 10000000b73914c1
Model           : Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

popcornmix, 2021-01-05:
Can you post the kernel panic?

benpalm, 2021-01-05:
Hi, sry but I am quite new to linux. How can I log/where do I find logs when the system has crashed?

popcornmix, 2021-01-05:
Depending on how it panics, you may still be able to issue commands (e.g. from ssh). Can you use ssh and run dmesg?
If you are locked out completely is the panic visible on screen? If so a photo of that may have some information in.

benpalm, 2021-01-05:
will do when back home in ~12h