raspberrypi/linux #4054: Network not working anymore though ethernet after heavy load RPi 4b

clougg, 2021-01-08:
**Describe the bug**
I cannot use ethernet anymore since I get IP but no connection at all while no problems on Wifi.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
I wanted to use my device as "server" attached to a external drive and with transmission, SMB and DLNA connected through ethernet port to get better speed than wireless.

**Actual behaviour**
Im using wifi while I find some fix for this but its way slower.

**System**
RASPINFO https://pastebin.pl/view/76187585

* Which model of Raspberry Pi? PI 4b
* Which OS and version (`cat /etc/rpi-issue`)? Linux raspberrypi 5.4.79-v7l+ #1373
* Which firmware version (`vcgencmd version`)? version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start)
* Which kernel version (`uname -a`)? Linux raspberrypi 5.4.79-v7l+

**Logs**
DMESG https://pastebin.pl/view/53644379

**Additional context**
Right now I havent it connected through ethernet since I use RPi trough ssh and a small touchscreen and If I plug ethernet I cannot use SSH, doesnt matter if im also connected to wifi, I cannot connect overall.

I dont know what else should I upload because Im new at this so thanks for the help and my apologies for the misspelled words!

pelwell, 2021-01-08:
```
[   45.075456] Bluetooth: RFCOMM TTY layer initialized
[   45.075473] Bluetooth: RFCOMM socket layer initialized
[   45.075490] Bluetooth: RFCOMM ver 1.11
[ 1172.658415] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[ 1175.753228] bcmgenet fd580000.ethernet eth0: Link is Down
[ 1177.951636] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```
What happened between the third and fourth lines in this log? The Pi had been booted for almost 20 minutes, but then something modified the network state - both for WiFi and Ethernet.

You mention heavy load in the title, but not elsewhere:
1. What is the Pi running?
2. What power supply is connect to the Pi?
3. I see you have a hub and an external disk - how are they powered?

clougg, 2021-01-08:
> ```
> [   45.075456] Bluetooth: RFCOMM TTY layer initialized
> [   45.075473] Bluetooth: RFCOMM socket layer initialized
> [   45.075490] Bluetooth: RFCOMM ver 1.11
> [ 1172.658415] brcmfmac: [...]

The change must be that i've been using it through wifi, but plugged the cable to try again, and unplugged it, to see if logs showed anything that could help searching whats happening.

1) DLNA, Samba server and Transmission. If I use VLC trough SMB to watch movies there is a point where this happens and I need to reinstall from scratch to fix it temporarily (maybe a day). 
2) Is a good quality power supply, came with the kit.
3) I have a small screen and an external hdd connected to its own power supply.

I already had the problem before having the touchscreen.

pelwell, 2021-01-08:
> Is a good quality power supply, came with the kit.

We see quite a few problems with "kit" power supplies.

Please repeat your test again without changing the WiFi or Ethernet configuration - don't enable or disable anything and don't touch the cable. When you get a failure, upload the dmesg output.

clougg, 2021-01-08:
The problem is that I cannot get the info while getting the failure. If I plug the ethernet cable i can't get the message since Im not able to connect through ssh, even if I'm already succesfully connected to wifi because I can't use internet.

pelwell, 2021-01-08:
The kernel messages should be copied to /var/log/kern.log, so you can find them after a reboot.

IComplainInComments, 2021-01-08:
> The problem is that I cannot get the info while getting the failure. If I plug the ethernet cable i can't get the message since Im not able to connect through ssh, even if I'm already succesfully connected to wifi because I can't use internet.

Plug the device into a monitor and see if the power warning appears on the monitor (just to help rule out a power supply issue, the firmware will display a graphic on the display if its detecting low power while in use).

clougg, 2021-01-08:
> Plug the device into a monitor and see if the power warning appears on the monitor (just to help rule out a power supply issue, the firmware will display a graphic on the display if its detecting [...]

Its not appearing, seen the touchscreen and nothing about low power.

IComplainInComments, 2021-01-08:
> Its not appearing, seen the touchscreen and nothing about low power.

Is it connected through HDMI? Or is it going through the display connector? Because I'm not sure if the firmware can display it through the Display connector (Im not sure what its called, its the connection on the board itself).

@pelwell Would know if it would or not.

clougg, 2021-01-08:
> Is it connected through HDMI? Or is it going through the display connector? Because I'm not sure if the firmware can display it through the Display connector (Im not sure what its called, its the [...]

It should since i get the full desktop and all, or im i wrong?

IComplainInComments, 2021-01-08:
> It should since i get the full desktop and all, or im i wrong?

Well I know the firmware can talk/control the GPU that would display this information. I just don't know if its rendered directly through the GPU or if its through the HDMI Controller. I Just wanna be safe we have the correct information.

clougg, 2021-01-08:
I can confirm I tried different power supply, cable and all and problem keeps btw

clougg, 2021-01-09:
Connected through hdmi no signal of low power on it

clougg, 2021-01-09:
> The kernel messages should be copied to /var/log/kern.log, so you can find them after a reboot.

https://pastebin.pl/view/2bbe05a6

Done, its big btw

clougg, 2021-01-09:
I think the failure happened Jan 6 and since then i cannot use ethernet.

pelwell, 2021-01-11:
1. Can you try a different ethernet cable? Or a different port on the switch?
2. Try enabling a small over-voltage by adding `over_voltage=2` to config.txt.