raspberrypi/linux #4056: Pi4 USB boot, unable to enumerate USB device

sblantipodi, 2021-01-08:
I am using my Raspberry Pi4 4GB along with my Cyberpower PR1500ELCD UPS.

UPS is connected to raspberry via USB and it worked flawlessly until I switched to a fresh install on Raspbian on an external USB SSD.

Raspbian works well on the SSD now but when I reboot, Raspbian does not see the UPS.
**I need to manually detach the UPS USB cable and reattach it to make it work.**
I think that this is a common problem for connected USB devices when using a USB SSD instead of a microSD.

Same config, same everything, works well if I install raspbian on microsd instead of the ssd.

Thank you guys!!!

This is the output of the dmesg.

```
[  143.668883] usb 1-1.4: new full-speed USB device number 11 using xhci_hcd
[  143.749144] usb 1-1.4: device descriptor read/64, error -32
[  143.937109] usb 1-1.4: device descriptor read/64, error -32
[  144.124924] usb 1-1.4: new full-speed USB device number 12 using xhci_hcd
[  144.205134] usb 1-1.4: device descriptor read/64, error -32
[  144.393074] usb 1-1.4: device descriptor read/64, error -32
[  144.501330] usb 1-1-port4: attempt power cycle
[  145.104931] usb 1-1.4: new full-speed USB device number 13 using xhci_hcd
[  145.105150] usb 1-1.4: Device not responding to setup address.
[  145.313091] usb 1-1.4: Device not responding to setup address.
[  145.520910] usb 1-1.4: device not accepting address 13, error -71
[  145.600942] usb 1-1.4: new full-speed USB device number 14 using xhci_hcd
[  145.601141] usb 1-1.4: Device not responding to setup address.
[  145.809115] usb 1-1.4: Device not responding to setup address.
[  146.016957] usb 1-1.4: device not accepting address 14, error -71
[  146.017395] usb 1-1-port4: unable to enumerate USB device
```

timg236, 2021-01-08:
Does adding over_voltage=1 to config.txt make any difference?

sblantipodi, 2021-01-08:
@timg236 thanks for the answer, no, no difference.

timg236, 2021-01-09:
"Raspbian works well on the SSD now but when I reboot, Raspbian does not see the UPS."

Just to confirm that it boots ok from the SSD but then fails when Linux tries to use USB to mount the root file-system? 

If you have a USB hub it would be useful to know whether that changes the behaviour.

sblantipodi, 2021-01-10:
@timg236 
Raspbian boots without errors but it can't enumerate the USB device (an uninterruptible power supply) until I unplug it and plug it again.
I'm starting from a fresh clean install of raspbian.

This problem happens only if I boot raspbian from an USB SSD,
if I boot from microSD I have no problem 

I don't have an USB hub unfortunately.

sblantipodi, 2021-01-18:
@timg236 
I "workarounded/fixed" it by changing a bootloader config:

```
sudo -E rpi-eeprom-config --edit
USB_MSD_PWR_OFF_TIME=0
```

when raspberry reboots, it power off the USB for the time specified in this option: "USB_MSD_PWR_OFF_TIME",  
the default value is 1000.

changing it to 0 solved my problem with my UPS.

having a default value USB_MSD_PWR_OFF_TIME=1000 seems to create more problem than benefits.  
there are hundred of users with the same problem with all kind of devices.

may I ask you why that value is 1000 instead of 0?

timg236, 2021-01-18:
@sblantipodi You haven't stated which version of the bootloader you are using, presumably you upgraded to 2021-01-16? 

USB devices need to handle disconnect/power off properly. This configuration supports the maximum set of devices on Pi 4B R1.1 and I can't see it being changed because that will break support for existing devices. However, the EEPROM configuration is automatically migrated to future releases so quirks for particular devices are preserved.

sblantipodi, 2021-01-18:
Thanks for the answer @timg236 
Using 2020-12-14.

Is there some change in this regards in the latest eeprom?

Thank you

timg236, 2021-01-18:
Yes, the commit on 2020-12-14 commit which promoted 2020-12-11 exposed race condition when resetting the XHCI controller and controlling the USB port power. This was a regression compared to the default production release - Sep 2020. USB_MSD_PWR_OFF_TIME=0 would have bypassed this code at a higher level.

The stable/latest release 2021-01-16 which is the default APT release and there's been some positive test reports that the USB_MSD_PWR_OFF_TIME=0 workaround was not required i.e. vanilla config work. Although, if if USB_MSD_PWR_OFF_TIME=0 works for you then there's no harm leaving that enabled. A non-zero time is more significant for HDD enclosures and some wireless keyboards.

sblantipodi, 2021-01-18:
ok thanks for the answer, I appreciate it.
I confirm that the latest firmware fixed the problem without the needs of the quirck.

closing the issue.

closed 2021-01-18 by sblantipodi

timg236, 2021-01-18:
Excellent many thanks for the test report