raspberrypi/linux #4059: Ethernet connection randomly drops into NO-CARRIER

and3rson, 2021-01-09:
**System information**

- Board: RPi4
- Linux pi 5.4.83-5-ARCH #1 SMP PREEMPT Sun Jan 3 16:11:27 UTC 2021 armv7l GNU/Linux
- ArchLinuxARM, raspberrypi/linux@abdf918004f2a4c203891e94328b8cf76fbae45f - https://archlinuxarm.org/packages/armv7h/linux-raspberrypi4/log?id=099b14c198c20091322de28e7a07f51289c5947d
- ETH connection: full duplex, autoneg on, speed 1000 (`autoneg off` with manual duplex/speed setting did not work for me at all)

**Description**

I'm using Raspberry Pi 4 with ArchLinuxARM distro. Network is managed by `netctl`. `networkd`/`resolved` are disabled.

After some random time (5 minutes to 6 hours) an ethernet connection goes down. Sometimes it brings down a wireless connection as well. A trace appears in `journalctl` (see logs section below).

In cases when wireless connection was still up and I was able to SSH into the machine, `ip addr` was showing `eth0` as "NO-CARRIER,BROADCAST,MULTICAST,UP" with a configured IPv4/IPv6 address. I would then run `ip link set eth0 down` and then `systemctl restart netctl@wired`. This brought the ethernet connection up again.

- I've tried moving RPi4 away from the router to avoid electromagnetic interference and used the original 5.1V/3A power supply.
- I've updated entire system.
- I've tried `networkd` and `netctl` with and without `dhcpcd`.

**Logs**
```
Jan 09 17:03:35 pi kernel: ------------[ cut here ]------------                                                    
Jan 09 17:03:35 pi kernel: WARNING: CPU: 3 PID: 195 at drivers/net/phy/phy.c:716 phy_state_machine+0xd8/0x190       
Jan 09 17:03:35 pi kernel: Modules linked in: btsdio bluetooth ecdh_generic ecc brcmfmac brcmutil cfg80211 bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_
memops videobuf2_v4l2 videobuf2_common videodev mc raspberrypi_hwmon i2c_bcm2835 iproc_rng200 rng_core rpivid_mem uio_pdrv_genirq uio sch_fq_codel i2c_dev snd_bcm2835(C) snd_pcm snd_timer snd drm drm_panel_orientation_quirks fuse ip_tables
 x_tables ipv6                                                                                                                                                                                                                                 
Jan 09 17:03:35 pi kernel: CPU: 3 PID: 195 Comm: kworker/3:3 Tainted: G         C        5.4.83-5-ARCH #1                                                                                                                                      
Jan 09 17:03:35 pi kernel: Hardware name: BCM2711                                                                                                                                                                                              
Jan 09 17:03:35 pi kernel: Workqueue: events_power_efficient phy_state_machine                                                                                                                                                                 
Jan 09 17:03:35 pi kernel: [<c0210cdc>] (unwind_backtrace) from [<c020c594>] (show_stack+0x10/0x14)
Jan 09 17:03:35 pi kernel: [<c020c594>] (show_stack) from [<c0d0780c>] (dump_stack+0x90/0xb0)
Jan 09 17:03:35 pi kernel: [<c0d0780c>] (dump_stack) from [<c022c844>] (__warn+0xd0/0xf8)
Jan 09 17:03:35 pi kernel: [<c022c844>] (__warn) from [<c022cc0c>] (warn_slowpath_fmt+0x7c/0xa8)
Jan 09 17:03:35 pi kernel: [<c022cc0c>] (warn_slowpath_fmt) from [<c09fbddc>] (phy_state_machine+0xd8/0x190)
Jan 09 17:03:35 pi kernel: [<c09fbddc>] (phy_state_machine) from [<c0248c04>] (process_one_work+0x1e8/0x584)
Jan 09 17:03:35 pi kernel: [<c0248c04>] (process_one_work) from [<c0248ffc>] (worker_thread+0x5c/0x568)
Jan 09 17:03:35 pi kernel: [<c0248ffc>] (worker_thread) from [<c024eb14>] (kthread+0x170/0x188)                                                                                                                                                
Jan 09 17:03:35 pi kernel: [<c024eb14>] (kthread) from [<c02010d8>] (ret_from_fork+0x14/0x3c)                                                                                                                                                  
Jan 09 17:03:35 pi kernel: Exception stack(0xe2e9ffb0 to 0xe2e9fff8)
Jan 09 17:03:35 pi kernel: ffa0:                                     00000000 00000000 00000000 00000000
Jan 09 17:03:35 pi kernel: ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jan 09 17:03:35 pi kernel: ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jan 09 17:03:35 pi kernel: ---[ end trace 560aac2fd1e428b6 ]---
Jan 09 17:03:35 pi kernel: bcmgenet fd580000.ethernet eth0: Link is Down
```

Thanks in advance!

and3rson, 2021-01-10:
Another traceback:

```
Jan 10 18:22:01 pi kernel: ------------[ cut here ]------------
Jan 10 18:22:01 pi kernel: WARNING: CPU: 1 PID: 4557 at drivers/net/phy/phy.c:838 phy_stop+0xa0/0xa8
Jan 10 18:22:01 pi kernel: called from state HALTED
Jan 10 18:22:01 pi kernel: Modules linked in: btsdio bluetooth ecdh_generic ecc brcmfmac brcmutil cfg80211 bcm2835_v4l2(C) bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev raspberrypi_hwmon i2c_bcm2835 mc iproc_rng200 rng_core rpivid_mem uio_pdrv_genirq uio sch_fq_codel i2c_dev snd_bcm2835(C) snd_pcm snd_timer snd crypto_user drm fuse drm_panel_orientation_quirks ip_tables x_tables ipv6
Jan 10 18:22:01 pi kernel: CPU: 1 PID: 4557 Comm: ip Tainted: G        WC        5.4.83-5-ARCH #1
Jan 10 18:22:01 pi kernel: Hardware name: BCM2711
Jan 10 18:22:01 pi kernel: [<c0210cdc>] (unwind_backtrace) from [<c020c594>] (show_stack+0x10/0x14)
Jan 10 18:22:01 pi kernel: [<c020c594>] (show_stack) from [<c0d0780c>] (dump_stack+0x90/0xb0)
Jan 10 18:22:01 pi kernel: [<c0d0780c>] (dump_stack) from [<c022c844>] (__warn+0xd0/0xf8)
Jan 10 18:22:01 pi kernel: [<c022c844>] (__warn) from [<c022cc0c>] (warn_slowpath_fmt+0x7c/0xa8)
Jan 10 18:22:01 pi kernel: [<c022cc0c>] (warn_slowpath_fmt) from [<c09fbf34>] (phy_stop+0xa0/0xa8)
Jan 10 18:22:01 pi kernel: [<c09fbf34>] (phy_stop) from [<c0a06acc>] (bcmgenet_close+0x118/0x248)
Jan 10 18:22:01 pi kernel: [<c0a06acc>] (bcmgenet_close) from [<c0bbf7d0>] (__dev_close_many+0xac/0x134)
Jan 10 18:22:01 pi kernel: [<c0bbf7d0>] (__dev_close_many) from [<c0bcc580>] (__dev_change_flags+0xcc/0x1f0)
Jan 10 18:22:01 pi kernel: [<c0bcc580>] (__dev_change_flags) from [<c0bcc6bc>] (dev_change_flags+0x18/0x48)
Jan 10 18:22:01 pi kernel: [<c0bcc6bc>] (dev_change_flags) from [<c0be18ac>] (do_setlink+0x21c/0x8f8)
Jan 10 18:22:01 pi kernel: [<c0be18ac>] (do_setlink) from [<c0be2688>] (__rtnl_newlink+0x59c/0x81c)
Jan 10 18:22:01 pi kernel: [<c0be2688>] (__rtnl_newlink) from [<c0be294c>] (rtnl_newlink+0x44/0x64)
Jan 10 18:22:01 pi kernel: [<c0be294c>] (rtnl_newlink) from [<c0be089c>] (rtnetlink_rcv_msg+0x124/0x2ec)
Jan 10 18:22:01 pi kernel: [<c0be089c>] (rtnetlink_rcv_msg) from [<c0c25f2c>] (netlink_rcv_skb+0xc0/0x120)
Jan 10 18:22:01 pi kernel: [<c0c25f2c>] (netlink_rcv_skb) from [<c0c25754>] (netlink_unicast+0x178/0x21c)
Jan 10 18:22:01 pi kernel: [<c0c25754>] (netlink_unicast) from [<c0c259ac>] (netlink_sendmsg+0x1b4/0x378)
Jan 10 18:22:01 pi kernel: [<c0c259ac>] (netlink_sendmsg) from [<c0ba3124>] (____sys_sendmsg+0x1bc/0x234)
Jan 10 18:22:01 pi kernel: [<c0ba3124>] (____sys_sendmsg) from [<c0ba3a7c>] (___sys_sendmsg+0x70/0xa4)
Jan 10 18:22:01 pi kernel: [<c0ba3a7c>] (___sys_sendmsg) from [<c0ba4e44>] (sys_sendmsg+0x54/0x98)
Jan 10 18:22:01 pi kernel: [<c0ba4e44>] (sys_sendmsg) from [<c02011d4>] (__sys_trace_return+0x0/0x2c)
Jan 10 18:22:01 pi kernel: Exception stack(0xeed7dfa8 to 0xeed7dff0)
Jan 10 18:22:01 pi kernel: dfa0:                   0056ead8 00000002 00000003 beaa47d0 00000000 00000000
Jan 10 18:22:01 pi kernel: dfc0: 0056ead8 00000002 beaa4854 00000128 b6fa8550 00000000 00000000 0056ecb8
Jan 10 18:22:01 pi kernel: dfe0: 0056ed50 beaa4780 00542ab4 b6e8c9f8
Jan 10 18:22:01 pi kernel: ---[ end trace 92309d6ad851410b ]---
```

and3rson, 2021-01-12:
Fixed by ditching Pi and replacing it with ODROID-XU4. /s

JamesH65, 2021-01-12:
That's a workaround, not a fix.

closed 2021-01-12 by pelwell