raspberrypi/linux #4060: Extra chips select cycle with Buster on Pi 3 Model B

Fatsie, 2021-01-11:
**Is this the right place for my bug report?**
As far as I can tell the problem is with the SPI driver that is part of the kernel package.

**Describe the bug**
I am using the SPI interface on a Raspberry Pi 3 Model B Rev 1.2 to drive SPI interfaces on the [AD5522 development board](https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/EVAL-AD5522.html)
The bug is that after an xfer on the spi interface is complete a second assertion and de-assertion of the chips select signal is done.
I am using spidev module from python3, but verified the same behaviour by using a C program that accesses /dev/spidev0.0 directly using an IOCTL. This makes me conclude problem is in kernel driver and not in python code.

**To reproduce**
Send data over SPI bus and extra chip select signal can be observed.

**Expected behaviour**
Extra chip select cycle to not be there.

**Actual behaviour**
When sending data over SPI bus most of the time an extra chips select cycle is seen. This is oscilloscope capture:
[image]
The SPI clock is given in yellow, the purple is chips select (active low).
This extra cycle is problematic as it initiates a register write on the AD5522 which causes it to go in a disabled state. It thus makes AD5522 unusable from a Raspberry Pi.
Behaviour when seen from a stretch based raspbian:
[image]
Here everything is as it should be.
I would like to switch to buster version of OS to use newer python version.

**System**
```
System Information
------------------

Raspberry Pi 3 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990
b5662f092c106beed40c9f, stage5

Linux raspberrypi 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/L
inux
Revision	: a02082
Serial		: 000000000ca59ac6
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Nov 30 2020 22:13:46 
Copyright (c) 2012 Broadcom
version ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Bestandssysteem 1K-blokken Gebruikt Beschikbaar Geb% Aangekoppeld op
/dev/root         14966820  7661004     6631696  54% /
devtmpfs            440240        0      440240   0% /dev
tmpfs               473520        0      473520   0% /dev/shm
tmpfs               473520    12588      460932   3% /run
tmpfs                 5120        4        5116   1% /run/lock
tmpfs               473520        0      473520   0% /sys/fs/cgroup
/dev/mmcblk0p1       44220    24052       20169  55% /boot
tmpfs                94704        0       94704   0% /run/user/109
tmpfs                94704        0       94704   0% /run/user/1000
[... 210 lines cut ...]
[    0.000000] Linux version 5.4.79-v7+ (dom@buildbot) (gcc version 8.4.0 (Ubunt
u/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:22:33 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instructio
n cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 M
iB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id share
d-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49740 r8192 d23988 u81920
[    0.000000] pcpu-alloc: s49740 r8192 d23988 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835
```

**Additional context**
[My blog](https://chips4makers.io/blog/poor-mens-smu-part-2.html)

pelwell, 2021-01-11:
Are you saying that on the AD5522 asserting CS with no SPI clock cycles is enough to trigger an operation on the device? Is that even allowed by the SPI protocol?

Fatsie, 2021-01-11:
Yes, that is what I see. For the ADC on the dev board the chip select signal also has extra functionality: the de-assertion of the chip select signal does trigger a capture and conversion of the ADC input. It does seem common in devices to attach functionality to chips select, regardless if it is according to the standard.

Fatsie, 2021-01-11:
I upgraded kernel to 20210108 and problem seem to be solved. Let me do some more testing before closing this bug report.

pelwell, 2021-01-11:
That would be convenient.

( Your avatar is a blast from the past. (-8 )

Fatsie, 2021-01-11:
(I'm retro)
Unfortunately the problem is still there with upgraded kernel package, so my system is now:
```console
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
```