raspberrypi/linux #4063: Option to set max ALSA volume for HDMi to 0dB.

JWahle, 2021-01-12:
The possible max volume for HDMI and headphone outputs is +4dB.
Loud audio files played over HDMI at this volume are audibly distorted.
This is a problem for Raspi audio players like [moOde](http://www.moodeaudio.org), that set the ALSA volume to 100% by default.
Of course, moOde could set the volume to 0dB, but then users cannot use the more intuitive %-scale to set their volume.
Would it be possible to set the HDMI max volume to 0dB or provide a setting in `config.txt` to achieve this?

**To reproduce**
I have a Pi 3B with moOde 7.0.0 (2020-12-02-raspios-buster-armhf-lite.zip), but I think this applies to all Raspberrys with current Raspbian.
1. run `amixer -c 0 sset HDMI 100%`
This should output:
```
Simple mixer control 'HDMI',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]
```
2. Play loud audio.
=> Audio is distorted in very loud passages.

JWahle, 2021-01-12:
Related issues in other projects:
https://github.com/Fornoth/spotify-connect-web/issues/103
https://github.com/moode-player/moode/issues/275

pelwell, 2021-01-12:
See also https://github.com/raspberrypi/linux/issues/2982.

JWahle, 2021-01-17:
Thanks for the link.

I assume, it is not possible to set the max volume to 0dB only for HDMI and keep the 4dB max volume for the headphone output, right?

closed 2021-02-04 by JWahle