raspberrypi/linux #4064: Network down after reboot + USB reset/poweroff handling

ElVasquito, 2021-01-12:
I previously used a PI3 mainly for retro gaming, so I have some little experience tinkering with it.
Now I just aquired a PI4 to use it as a NAS, with a 2.5" 2TB USB3 HDD powered directly from the PI, fully updated/upgraded/dist-upgraded.
I am powering the PI through the GPIO header, using pins 2+4 for +5V, 6+9 for ground, with thick and short wires (not the regular colored ones which are very thin).

There are two main issues/complaints for me as of now:

1) The dreaded #3108 after reset. In my case, both LEDs in the LAN jack remain off after a sudo reboot. Sometimes unplugging and replugging the LAN cable makes the LEDs turn on and have the LAN working properly, sometimes resetting the router fixes the issue as well, and sometimes not and I need to power cycle to get the LEDs to turn on and have the LAN working again.
Just seen a related commit made some hours ago? Maybe this is about to be fixed once and for all. How can I check it?

2) Which is somewhat related I think, but to the USB ports and not the LAN. And it happens on the PI3 as well:
On reset or poweroff, the hard drive makes a loud clicking/rattling of the head. 
The "Power-Off_Retract_Count" item in the hard disk's SMART info is increasing as I poweroff/reset the PI. If I reboot or poweroff my PC or laptop with the drive connected, there's definitely no clicking, so there seems to be an issue on how the USB ports are handled on reset/poweroff of the PI3+PI4 (and maybe the older ones as well?).
Originally, I had two clicks from the HDD, one right on reset (just when the screen goes blank) and another one while POSTing, I  prevented this last click by adding USB_MSD_PWR_OFF_TIME=0 to the EEPROM, but for the one on reset/poweroff I haven't found anything that would prevent it.
What I noticed is that when issuing the reboot/poweroff command, the OS unloads its stuff, and suddenly (while the HDD LED is still blinking rapidly) the reset occurs, like it is not giving the drive some extra time to park its heads gracefully before cutting the power to the ports. This clicking definitely shorts the lifespan of the drive's head. Is this power-reset really needed?
I've seen this mentioned already, here: https://www.raspberrypi.org/forums/viewtopic.php?t=247093 but the solution was of no help to me because I am not booting from SD, so the proposed script can't be run as the HDD can't be umounted till the very end.
But still, I think it is a good idea: send a sleep command to the drive after unloading the OS and just before the reset/poweroff. The sleep command will park the head normally, without the abrupt clicking.
Can this be implemented somehow?

Thanks in advance for any clues!

pelwell, 2021-01-12:
One problem per "Issue", please.

closed 2021-01-12 by pelwell

ElVasquito, 2021-01-12:
Ok, will copy item 2 in a new issue and use #3108 for item 1, thank you.