raspberrypi/linux #4066: 3B+ stops responding after a few minutes when booted with FullKMS enabled

jose1711, 2021-01-12:
**Describe the bug**
With newly installed Raspbian, updated after I enable FullKMS via `raspi-config` and reboot I am only able to get uptime of a few minutes before Pi stops responding and a manual reboot must be applied. If I disable KMS all goes back to normal.

**To reproduce**
- use NOOBS to install Raspbian
- use `pi-wizard` to finish configuration - no wifi, only ethernet, update the system
- use `raspi-config` to enable fullKMS
- reboot
- wait a few minutes (typically not more than 10) while the system is idle (booted in graphical environment)

**Expected behaviour**
System continues running.

**Actual behaviour**
System stops responding, screen goes blank, ping is not possible.

**System**
* Pi3B+ but I've observed similar behaviour on PiZeroW
* latest Raspbian
* official PSU, only peripherals connected during test are HDMI monitor + ethernet cable

**Logs**
n/a

popcornmix, 2021-01-12:
Are you able to ssh in and report output of "dmesg" when the pi appears hung?
I suspect you'll see something like:
```[   19.679228] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out```

kms on Pi2/Pi3 is not really ready for active use. It may work better on 5.10 kernel but there is still a "flip_done timed out" issue being investigated.

6by9, 2021-01-12:
And a Pi0/0W is going to REALLY struggle with Full KMS enabled. With only 512MB of RAM you'll get heavy memory pressure, and the low end ARM core won't be really up to the 3D setup.

jose1711, 2021-01-12:
> Are you able to ssh in and report output of "dmesg" when the pi appears hung?
> I suspect you'll see something like:
> `[ 19.679228] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out`

Impossible. As soon as it dies, it's dead for good. I may hook up serial console and monitor kernel messages there if you think that makes any sense.

jose1711, 2021-01-12:
> And a Pi0/0W is going to REALLY struggle with Full KMS enabled. With only 512MB of RAM you'll get heavy memory pressure, and the low end ARM core won't be really up to the 3D setup.

Point taken. This was only a quick test after I suspected that hw may be faulted.

popcornmix, 2021-01-12:
> Impossible. As soon as it dies, it's dead for good. I may hook up serial console and monitor kernel messages there if you think that makes any sense.

There may be a panic reported that is visible to serial, but doesn't make it to display, so that could be helpful.
But there's not much point testing on 5.4 kernel as development is continuing on 5.10 and the code is far more advanced.
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=288234

mripard, 2021-07-06:
We fixed a lot of similar bugs recently, so it might be worth testing with the latest kernel and let us know if the issue is still there