raspberrypi/linux #4067: Just two flaws in the first install process

rin67630, 2021-01-12:
I just re-did a fresh reinstallation of Raspbian Buster using Raspberry imager.
That worked well.
At the initial configuration, i have entered, as required, a new password. 
Then I went through the localization and I have selected Germany, English language, German keyboard.
Then I rebooted.
A usual for me performing Preferences/RaspberryPi configuration, I selected SSH and then got a warning that SSH together _with the default password_ were at risk.
The _keyboard selected was US_.

So the two of initial configuration options I have chosen were not respected.

Not a big issue for me, but worth mentioning.
At least the not executed password change should be addressed.

pelwell, 2021-01-12:
I have seen something similar once, and put it down to a sequencing issue. This isn't the ideal location to report the problem because it is unlikely to be down to a bug in the Linux kernel, but as I can't think of a better one at the moment it can stay.

Do you have any ideas, @spl237?

spl237, 2021-01-12:
This is a known bug in the wizard in the December release - if the language was changed, further settings were ignored due to a permissions issue. It's already been fixed.

closed 2021-01-12 by pelwell