raspberrypi/linux #4068: Bug-blank screen after rotate to right/left by using vc4-kms-v3d-pi4

jamwu, 2021-01-12:
**Describe the bug**
Blank screen after rotate to right/left by using vc4-kms-v3d-pi4

**To reproduce**
All these steps base on a clean os-2020-12-02-raspios-buster-armhf.
1. comment dtoverlay=vc4-fkms-v3d and add dtoverlay=vc4-kms-v3d-pi4 to /boot/config.txt and save.
2. reboot. 
3. after this boot, the screen shows ok(because without rotate yet). Rotate your screen(what i used is an HDMI screen) to right/left and save, now the screen was rotated to the right and shows ok.
4. reboot again. You will see the problem.

**Expected behaviour**
The os will save the screen rotation. When boot, automatically rotate to the last rotation.

**Actual behaviour**
Can see the rainbow screen and the splash screen (no rotated, stay normal), when booting. But after that, just a black screen, nothing!! And the screen itself shows the "check cable connection" message.
The pi boot ok can connect with ssh.

**System**
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage4
pi@raspberrypi:~ $ vcgencmd version
Nov 25 2020 17:10:27
Copyright (c) 2012 Broadcom
version 2e6292a5304b95ecfd700fc5ab2b793402056332 (clean) (release) (start)
```

**Logs**
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.79-v7l+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:27:40 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49804 r8192 d23924 u81920
[    0.000000] pcpu-alloc: s49804 r8192 d23924 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1010432
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:9C:62:BC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=39204adc-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x17cc0000-0x1bcc0000] (64MB)
[    0.000000] Memory: 3655604K/4050944K available (10240K kernel code, 739K rwdata, 2816K rodata, 2048K init, 854K bss, 133196K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31111 entries in 61 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x344/0x518 with crng_init=0
[    0.000009] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000024] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000103] bcm2835: system timer (irq = 17)
[    0.000755] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000771] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000789] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000802] Switching to timer-based delay loop, resolution 18ns
[... 438 lines cut ...]
Screen 0: minimum 320 x 200, current 1080 x 1920, maximum 7680 x 7680
HDMI-1 connected primary 1080x1920+0+0 right (normal left inverted right x axis y axis) 435mm x 239mm
   1920x1080     60.00*+
   1680x1050     59.88
   1280x1024     75.02    60.02
   1440x900      74.98    59.90
   1280x960      60.00
   1280x720      60.00
   1024x768      75.03    60.00
   800x600       75.00    60.32
   640x480       75.00    72.81    66.67    59.94
   720x400       70.08
HDMI-2 disconnected (normal left inverted right x axis y axis)

```

**confige.txt**
```
pi@raspberrypi:~ $ cat /boot/config.txt |grep -Ev '^$|#'
dtparam=audio=on
[pi4]
max_framebuffers=2
[all]
dtoverlay=vc4-kms-v3d-pi4
```

6by9, 2021-01-12:
You have 
```
[  182.394515] Under-voltage detected! (0x00050005)
[  186.554158] Voltage normalised (0x00000000)
```
multiple times warnings in your logs. You need a better power supply.

On undervoltage or overheat conditions the firmware tries to add warning icons to the screen. This is incompatible with vc4-kms-v3d (you don't need the -pi4 extension any more). 
You need to add
```avoid_warnings=1```
to config.txt to disable this, but better to fix the underlying issue of a dodgy power supply.

@popcornmix @pelwell Thoughts on automatically setting ```avoid_warnings=1``` if vc4-kms-v3d is loaded?
I haven't checked to ensure that it only disables the warning icons and not the other monitoring.

pelwell, 2021-01-12:
@6by9 are you saying that the warnings are actively harmful in this case or just unnecessary?

popcornmix, 2021-01-12:
The code does:
```
   int on = number && !C(avoid_warnings) && !intctrl_is_interrupt_masked(INTERRUPT_VIDEOSCALER);
```
which certainly intends to disable visible warnings with kms.

pelwell, 2021-01-12:
For the record, `avoid_warnings=1` will also disable the flashing of the PWR LED, but it doesn't stop the measurement or the CPU throttling.

6by9, 2021-01-12:
> are you saying that the warnings are actively harmful in this case or just unnecessary?

Memory had said harmful. The firmware goes and updates the dlists to try and add the overlay, but it's not in charge of the HVS.

@popcornmix Thanks - I'd totally missed the ```INTERRUPT_VIDEOSCALER``` masking. That should stop it.
Ah, a commit by you in 2016 (d2b73a2e). I think it's still in some of the docs.

jamwu, 2021-01-12:
> On undervoltage or overheat conditions the firmware tries to add warning icons to the screen. This is incompatible with vc4-kms-v3d (you don't need the -pi4 extension any more).
> You need to add [...]

@6by9 Yes,my power supply maybe not good enough.I will try a better one. But...It doesn't seem logical.If it is the power supply problem, why only rotate to right/left(90°/270°) cause this issue？normal/inverted(0°/180°) are ok, don't have this problem, with the same power supply.

6by9, 2021-01-12:
H&V flips can be done for near zero cost whilst composing in the HVS.
Transpose is done as a second pass through GL

jamwu, 2021-01-13:
@6by9 
After i change to another power supply. Without under-voltage warning. But the result is the same. Even i add
```
avoid_warnings=1
```
and using vc4-kms-v3d instead of vc4-kms-v3d-pi4.
Still black screen after the splash screen.

Here is the log.
**dmesg**
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.79-v7l+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:27:40 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49804 r8192 d23924 u81920
[    0.000000] pcpu-alloc: s49804 r8192 d23924 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1010432
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:9C:62:BC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=39204adc-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x17cc0000-0x1bcc0000] (64MB)
[    0.000000] Memory: 3655604K/4050944K available (10240K kernel code, 739K rwdata, 2816K rodata, 2048K init, 854K bss, 133196K reserved, 262144K cma-reserved, 3264512K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 31111 entries in 61 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x344/0x518 with crng_init=0
[    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000023] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000104] bcm2835: system timer (irq = 17)
[    0.000754] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000770] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000786] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000799] Switching to timer-based delay loop, resolution 18ns
[    0.001047] Console: colour dummy device 80x30
[    0.001084] printk: console [tty1] enabled
[    0.001138] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=540000)
[    0.001160] pid_max: default: 32768 minimum: 301
[    0.001311] LSM: Security Framework initializing
[    0.001510] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001533] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002943] Disabling memory control group subsystem
[    0.003068] CPU: Testing write buffer coherency: ok
[    0.003599] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004530] Setting up static identity map for 0x200000 - 0x20003c
[    0.004739] rcu: Hierarchical SRCU implementation.
[    0.005434] smp: Bringing up secondary CPUs ...
[    0.006680] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008055] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009361] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009522] smp: Brought up 1 node, 4 CPUs
[    0.009538] SMP: Total of 4 processors activated (432.00 BogoMIPS).
[    0.009552] CPU: All CPU(s) started in HYP mode.
[    0.009565] CPU: Virtualization extensions available.
[    0.010425] devtmpfs: initialized
[... 288 lines cut ...]
[   15.566551] Bluetooth: BNEP socket layer initialized
[   22.231762] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   32.471796] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   42.711721] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] flip_done timed out
[   52.951766] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:65:crtc-2] flip_done timed out
[   54.099889] fuse: init (API version 7.31)
[... cut ...]
[   55.821992] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   55.822888] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   55.825006] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   55.825296] vc4_hdmi fef05700.hdmi: ASoC: can't open interface fef05700.hdmi: -19
[   55.886815] Bluetooth: RFCOMM TTY layer initialized
[   55.886837] Bluetooth: RFCOMM socket layer initialized
[   55.886858] Bluetooth: RFCOMM ver 1.11

```

6by9, 2021-01-13:
I've just noticed that you're trying vc4-kms-v3d for Pi4 on the 5.4 kernel branch. That branch has an early version of the driver which is known to have issues, which is why the option is not enabled via raspi-config. 
The main development for vc4-kms-v3d is on the 5.10 branch, and is available via ```rpi-update``` (normal warnings of not doing this on a critical Pi and backing up first).

jamwu, 2021-01-13:
> The main development for vc4-kms-v3d is on the 5.10 branch, and is available via `rpi-update` (normal warnings of not doing this on a critical Pi and backing up first).

Yes, you are right! After update to 5.10.6. This problem solved!

pelwell, 2021-01-13:
This is when you close the issue.

closed 2021-01-13 by pelwell

colesnicov, 2024-05-05:
Hi @jamwu. Please, which model RPI have you used here? 

I try to turn the drm layer of 90 degrees on my **RPI 2B without 'X'** and I do not give that. I wrote about it here https://github.com/raysan5/raylib/issues/3958 If you would like to look