raspberrypi/linux #4074: Raspberry Pi 3 Model B hangs after some time of network and disk activity

handlerug, 2021-01-14:
**Describe the bug**
Raspberry Pi 3 Model B Rev 1.2 hangs after 2–3 days of uptime. Probably USB controller is dying though I can't verify if everything else is still running.

**To reproduce**
1. Connect an external SATA to USB controller.
2. Do some network and disk activity. (serve content using nginx, run samba, seed some torrents using transmission-daemon, have an RSS reader running and retrieving new content every hour)
3. Wait.
4. Profit

**Expected behaviour**
I've expected Raspberry Pi to remain accessible from network and (if it's the one to blame) USB controller to work.

**Actual behaviour**
Raspberry Pi "hangs" after a while. It stops reading from disk and it's not accessible from network (I think it's because Ethernet is connected via USB). The strange thing is that SD card LED is still blinking (though not as often as before hanging) and the power LED is on. Sometimes HDD randomly reconnects and gets a new... pseudo-file name? (/dev/sda -> /dev/sdb)

**System**
(I have no `raspinfo` on my system)

* Which model of Raspberry Pi? — Raspberry Pi 3 Model B Rev 1.2
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Jan  8 2021 14:33:35
Copyright (c) 2012 Broadcom
version 194a85abd768c7334bbadc3f1911c10a7d18ed14 (clean) (release) (start)
```
* Which kernel version (`uname -a`)? `Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux`

**Logs**
Sometimes it's like this:
```
Jan 14 17:34:29 raspberrypi kernel: [ 8547.616970] usb usb1-port1: disabled by hub (EMI?), re-enabling...
Jan 14 17:34:29 raspberrypi kernel: [ 8547.617002] usb 1-1: USB disconnect, device number 2
Jan 14 17:34:29 raspberrypi kernel: [ 8547.617021] usb 1-1.1: USB disconnect, device number 3
Jan 14 17:34:29 raspberrypi kernel: [ 8547.617956] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 7
Jan 14 17:34:29 raspberrypi kernel: [ 8547.617988] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Jan 14 17:34:29 raspberrypi kernel: [ 8547.618399] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Jan 14 17:34:29 raspberrypi kernel: [ 8547.687762] usb 1-1.2: USB disconnect, device number 4
Jan 14 17:34:29 raspberrypi kernel: [ 8548.036835] Indeed it is in host mode hprt0 = 00001901
Jan 14 17:34:29 raspberrypi kernel: [ 8548.176798] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:29 raspberrypi kernel: [ 8548.456848] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:30 raspberrypi kernel: [ 8548.736798] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:30 raspberrypi kernel: [ 8549.016797] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:30 raspberrypi kernel: [ 8549.296834] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Jan 14 17:34:30 raspberrypi kernel: [ 8549.296936] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:30 raspberrypi kernel: [ 8549.436803] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:31 raspberrypi kernel: [ 8549.716792] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:31 raspberrypi kernel: [ 8549.996795] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:31 raspberrypi kernel: [ 8550.276858] Indeed it is in host mode hprt0 = 00001501
Jan 14 17:34:31 raspberrypi kernel: [ 8550.556858] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Jan 14 17:34:31 raspberrypi kernel: [ 8550.556935] usb usb1-port1: attempt power cycle
```

It can also go like this:
```
Jan 13 20:37:28 raspberrypi kernel: [167181.793442] EXT4-fs error: 46 callbacks suppressed
Jan 13 20:37:28 raspberrypi kernel: [167181.793461] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235336: comm transmission-da: reading directory lblock 0
...
Jan 13 20:37:33 raspberrypi kernel: [167186.708864] EXT4-fs (sdb1): recovery complete
Jan 13 20:37:33 raspberrypi kernel: [167186.709487] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
Jan 14 12:09:50 raspberrypi kernel: [223123.755291] EXT4-fs (sda1): error count since last fsck: 729174
Jan 14 12:09:50 raspberrypi kernel: [223123.755337] EXT4-fs (sda1): initial error at time 1610427184: __ext4_find_entry:1532: inode 21438466
Jan 14 12:09:50 raspberrypi kernel: [223123.755367] EXT4-fs (sda1): last error at time 1610545050: __ext4_find_entry:1532: inode 21233665
Jan 14 15:00:53 raspberrypi kernel: [233386.793976] NYET/NAK/ACK/other in non-error case, 0x00000002
Jan 14 15:00:53 raspberrypi kernel: [233386.794027] NYET/NAK/ACK/other in non-error case, 0x00000002
Jan 14 15:00:53 raspberrypi kernel: [233386.794049] NYET/NAK/ACK/other in non-error case, 0x00000002
Jan 14 15:00:53 raspberrypi kernel: [233386.934135] usb usb1-port1: disabled by hub (EMI?), re-enabling...
Jan 14 15:00:53 raspberrypi kernel: [233386.934168] usb 1-1: USB disconnect, device number 2
Jan 14 15:00:53 raspberrypi kernel: [233386.934188] usb 1-1.1: USB disconnect, device number 3
Jan 14 15:00:53 raspberrypi kernel: [233386.935192] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 3
Jan 14 15:00:53 raspberrypi kernel: [233386.935229] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Jan 14 15:00:53 raspberrypi kernel: [233386.935653] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Jan 14 15:00:53 raspberrypi kernel: [233386.995191] usb 1-1.2: USB disconnect, device number 10
Jan 14 15:00:53 raspberrypi kernel: [233387.694012] Indeed it is in host mode hprt0 = 00001501
...
Jan 14 15:00:55 raspberrypi kernel: [233388.953982] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Jan 14 15:00:55 raspberrypi kernel: [233388.954051] Indeed it is in host mode hprt0 = 00001501
...
Jan 14 15:00:56 raspberrypi kernel: [233390.213993] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Jan 14 15:00:56 raspberrypi kernel: [233390.214040] usb usb1-port1: attempt power cycle
Jan 14 15:03:16 raspberrypi kernel: [233529.776355] EXT4-fs error (device sdb1): __ext4_find_entry:1532: inode #21238067: comm transmission-da: reading directory lblock 0
...
Jan 14 15:03:16 raspberrypi kernel: [233529.977342] EXT4-fs warning (device sdb1): dx_probe:761: inode #21233786: lblock 0: comm transmission-da: error -5 reading directory block
(then it's spamming the message above a lot)
```

**Additional context**
* External SATA to USB controller with an HDD connected. Powered with a separate adapter.
* Official Raspbian/Raspberry Pi OS, apt-upgraded once. No dist-upgrades were done.
* Kernel from apt repos. No rpi-update done.
* Official Raspberry Pi microUSB power source. No thunderbolt icon when external monitor is connected.
* Running headless (I don't have USB peripherals and monitor connected). I've connected an UART to USB converter because I hope kernel panics are going to be dumped to UART too (earlycon enabled). 

How can I check kernel panics over serial TTY (via UART)? That's the preferable option for me. I can connect a monitor as well if that's not the case. If you need more info please ask.

pelwell, 2021-01-14:
Adding `enable_uart=1` to config.txt will enable serial TX on GPIO 14 (GPIO 15 is the RX), and by default on Raspberry Pi OS that hosts a Linux console.

A recurring theme between those traces is `(EMI?)`, which suggests an electrical problem - try swapping the cable.

handlerug, 2021-01-14:
> Adding `enable_uart=1` to config.txt will enable serial TX on GPIO 14 (GPIO 15 is the RX), and by default on Raspberry Pi OS that hosts a Linux console.

I already have `enable_uart=1`, so I guess I'm good. The kernel is gonna dump stacktraces there too, right? (I'm not even sure it crashes). I'll also setup some basic monitoring to get notified of the issue and try to login via serial TTY.

> A recurring theme between those traces is `(EMI?)`, which suggests an electrical problem - try swapping the cable.

Hm, that's interesting. I couldn't decipher that :D Are you saying that it's enough to make the USB controller go down so that network access cuts off? It's not like I have a lot of USB-A to USB-A cables here (the one I use is packaged with the SATA controller) but I'll try swapping it. Might be also a problem with the controller itself, though it seems that it's of a fine quality.

Thanks for the quick reply!

P33M, 2021-01-14:
Swapping the cable might be a bit difficult because the port that's signalling disconnect is the Pi's root port (i.e connection between SMSC chip and bcm2837).

That said, electrical issues including external noise can filter into the board and disrupt things, so changing power supply or adding a hub between the SATA adapter and Pi may have an effect.

handlerug, 2021-01-14:
Power supply is unlikely to be a problem here IMO, but the SATA adapter? I don't really trust that thing. So getting a hub might be a good idea though it's rather expensive here.

I just put some load on it. I'll also connect a monitor for a good measure to check for kernel panics.

handlerug, 2021-01-15:
While I was sleeping, the USB controller has disconnected (log entries in the middle):
```
Jan 14 23:59:42 raspberrypi kernel: [   14.558349] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Jan 14 23:59:43 raspberrypi kernel: [   16.097798] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
Jan 15 00:00:57 raspberrypi kernel: [   76.738127] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
Jan 15 10:10:16 raspberrypi kernel: [36636.046056] usb 1-1.2: USB disconnect, device number 4
Jan 15 10:10:17 raspberrypi kernel: [36637.214683] usb 1-1.2: new high-speed USB device number 5 using dwc_otg
Jan 15 10:10:17 raspberrypi kernel: [36637.345707] usb 1-1.2: New USB device found, idVendor=152d, idProduct=2329, bcdDevice= 1.00
Jan 15 10:10:17 raspberrypi kernel: [36637.345720] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=5
Jan 15 10:10:17 raspberrypi kernel: [36637.345727] usb 1-1.2: Product: USB to ATA/ATAPI bridge
Jan 15 10:10:17 raspberrypi kernel: [36637.345734] usb 1-1.2: Manufacturer: JMicron
Jan 15 10:10:17 raspberrypi kernel: [36637.345741] usb 1-1.2: SerialNumber: 7D577A903183
Jan 15 10:10:17 raspberrypi kernel: [36637.349626] usb-storage 1-1.2:1.0: USB Mass Storage device detected
Jan 15 10:10:17 raspberrypi kernel: [36637.350107] usb-storage 1-1.2:1.0: Quirks match for vid 152d pid 2329: 8020
Jan 15 10:10:17 raspberrypi kernel: [36637.350201] scsi host1: usb-storage 1-1.2:1.0
Jan 15 10:10:18 raspberrypi kernel: [36638.438632] scsi 1:0:0:0: Direct-Access     WDC WD50 03ABYX-01WERA1   01.0 PQ: 0 ANSI: 2 CCS
Jan 15 10:10:18 raspberrypi kernel: [36638.439646] sd 1:0:0:0: Attached scsi generic sg0 type 0
Jan 15 10:10:18 raspberrypi kernel: [36638.444870] sd 1:0:0:0: [sdb] 976773168 512-byte logical blocks: (500 GB/466 GiB)
Jan 15 10:10:18 raspberrypi kernel: [36638.445785] sd 1:0:0:0: [sdb] Write Protect is off
Jan 15 10:10:18 raspberrypi kernel: [36638.445806] sd 1:0:0:0: [sdb] Mode Sense: 28 00 00 00
Jan 15 10:10:18 raspberrypi kernel: [36638.446563] sd 1:0:0:0: [sdb] No Caching mode page found
Jan 15 10:10:18 raspberrypi kernel: [36638.446581] sd 1:0:0:0: [sdb] Assuming drive cache: write through
Jan 15 10:10:18 raspberrypi kernel: [36638.553944]  sdb: sdb1
Jan 15 10:10:18 raspberrypi kernel: [36638.557187] sd 1:0:0:0: [sdb] Attached SCSI disk
Jan 15 10:10:42 raspberrypi kernel: [36662.651606] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21233665: comm transmission-da: reading directory lblock 0
Jan 15 10:10:42 raspberrypi kernel: [36662.672687] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21233665: comm transmission-da: reading directory lblock 0
Jan 15 10:10:43 raspberrypi kernel: [36663.086175] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21233665: comm transmission-da: reading directory lblock 0
Jan 15 10:10:43 raspberrypi kernel: [36663.106740] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21233665: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36707.965389] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235409: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36707.985832] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235409: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.004972] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235409: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.024018] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235409: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.042981] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235400: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.061767] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235400: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.080347] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235400: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.097951] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235400: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.113923] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235409: comm transmission-da: reading directory lblock 0
Jan 15 10:11:28 raspberrypi kernel: [36708.129488] EXT4-fs error (device sda1): __ext4_find_entry:1532: inode #21235409: comm transmission-da: reading directory lblock 0
Jan 15 10:11:58 raspberrypi kernel: [36738.587701] EXT4-fs error: 30 callbacks suppressed
```
Then it started to disconnect and immediately connect again and again, approximately every 1-2 hours. Notice that there's no more `(SMI?)`. EXT4 errors are probably caused by sda getting renamed to sdb after reconnection but mountpoint still pointing to sda. Are electrical issues still the reason?

handlerug, 2021-01-21:
Okay, it has been working fine without any interrupts for 7 days with external USB drive disconnected. Probably an issue with the adapter, therefore closing this issue. Thanks for advice!

closed 2021-01-21 by handlerug