raspberrypi/linux #4076: Kernel 5.10.7 Compile Warnings

Dark-Sky, 2021-01-14:
When compiling kernel 5.10.7 I get these warnings. Is this normal:

```
arch/arm64/boot/dts/overlays/i2c-gpio-overlay.dts:15.5-24: Warning (reg_format): /fragment@0/__overlay__/i2c@0:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/overlays/i2c-gpio.dtbo: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/overlays/i2c-gpio.dtbo: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/overlays/i2c-gpio.dtbo: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/overlays/i2c-gpio-overlay.dts:14.20-23.6: Warning (avoid_default_addr_size): /fragment@0/__overlay__/i2c@0: Relying on default #address-cells value
arch/arm64/boot/dts/overlays/i2c-gpio-overlay.dts:14.20-23.6: Warning (avoid_default_addr_size): /fragment@0/__overlay__/i2c@0: Relying on default #size-cells value
```

pelwell, 2021-01-14:
It is - I tried getting rid of them once, with dire consequences.

I'll buy a case of beer (or equivalent) if someone can get it to compile cleanly.

Dark-Sky, 2021-01-14:
Lol. Close this when you want to.

pelwell, 2021-01-14:
Thanks, but I'm going to leave it here as a challenge.

pelwell, 2022-05-19:
You shouldn't see these warnings in one of our downstream branches because we have patches to suppress them:
* rpi-5.10.y: f4590cfb672e86de142af37c11776d1510c48018
* rpi-5.15.y: d88be102ffc9c66abde1fba7d7bf69ff1e19d4d4
etc.
Aside from that, I've given up.

closed 2024-12-12 by Dark-Sky