raspberrypi/linux #4077: [5.10.5/arm64] - RPi3 /sys/class/thermal/thermal_zone0/temp not working

Joulinar, 2021-01-15:
**Describe the bug**
Currently I'm not able to read CPU temperature on RPi3B+ while the exakt same image/sd card is working well on a RPi4B

```
cat /sys/class/thermal/thermal_zone0/temp
cat: /sys/class/thermal/thermal_zone0/temp: Invalid argument
```

**To reproduce**
run `cat` on `sys/class/thermal/thermal_zone0/temp`

**Expected behaviour**
CPU temp should be displayed as it is on RPi4B

**Actual behaviour**
temperature file could not be read

**System**
* Which model of Raspberry Pi?
RPi3B+

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage4

* Which firmware version (`vcgencmd version`)?
Jan  7 2021 18:33:15
Copyright (c) 2012 Broadcom
version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start_cd)

* Which kernel version (`uname -a`)?
Linux DietPi3 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.
<details>
  <summary>dmesg logfile attached. Click to expand!</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.5-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000] Zeroed struct page in unavailable ranges: 512 pages
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000] On node 0 totalpages: 255488
[    0.000000]   DMA zone: 3992 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 255488 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s91224 r8192 d31656 u131072
[    0.000000] pcpu-alloc: s91224 r8192 d31656 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 251496
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=tty1 root=PARTUUID=ad09722e-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait net.ifnames=0
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[... 246 lines cut ...]
[    4.294132] libphy: lan78xx-mdiobus: probed
[    4.308848] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): int urb period 64
[    4.452770] scsi 0:0:0:0: Direct-Access      USB      SanDisk 3.2Gen1 1.00 PQ: 0 ANSI: 6
[    4.453971] sd 0:0:0:0: [sda] 30031872 512-byte logical blocks: (15.4 GB/14.3 GiB)
[    4.454984] sd 0:0:0:0: [sda] Write Protect is off
[    4.455008] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[    4.455416] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[    4.484511]  sda: sda1
[    4.488375] sd 0:0:0:0: [sda] Attached SCSI removable disk
[    4.581140] mc: Linux media interface: v0.10
[    4.634774] videodev: Linux video capture interface: v2.00
[    4.727637] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    4.728978] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[    4.728998] [vc_sm_connected_init]: start
[    4.731541] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[    4.731614] [vc_sm_connected_init]: failed to initialize shared memory service
[    4.740625] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    4.751957] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[    4.762959] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    4.767173] sd 0:0:0:0: Attached scsi generic sg0 type 0
```
</details>

**Additional context**
Issue happen on a fresh installation of RPi OS 64bit

MichaIng, 2021-01-15:
Just for completeness: `/sys/class/hwmon/hwmon0/temp1_input` is affected as well, which is functional on 5.10.5 32-bit kernel. `vcgencmd measure_temp` works it would be good if at least one of the generic kernel tunables worked as well.

DougieLawson, 2021-01-16:
Same problem exists on 3B, 3B+ and 3A running 5.10.7-v8+

A 3B running 5.10.7-v7+ works OK, so it's a 32 vs 64 bit error.

pelwell, 2021-01-16:
This will get some attention next week.

ricktendo, 2021-01-18:
I wonder if this is why the fan on my PoE hat does not work?

I re-installed the original kernel from the repo and fan is working again.

pelwell, 2021-01-18:
I think the problem here is simply that the 3 and the 4 are sharing a kernel build, and that bcm2711_defconfig does not include `CONFIG_BCM2835_THERMAL=y`. I can test the obvious fix tomorrow, but I thought I'd mention it now in case anyone else is in a position to try themselves.

pelwell, 2021-01-19:
The obvious fix has been pushed to rpi-5.10.y and rpi-5.11.y, and will be in future kernel releases.

MichaIng, 2021-01-19:
And I thought the RPi 3 is using https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm64/configs/bcmrpi3_defconfig, but of course both use the same as there is only a single 64-bit kernel shipped, so all makes sense then 😄.

Just to remember where this has been changed initially:
- https://github.com/raspberrypi/linux/issues/3071
- https://github.com/raspberrypi/linux/pull/3073

I hope the two builtins do not interfere with each other. If there is more SoC-specific, it could make sense to build separate kernels by times. But to not ship 5 or more kernels for every model with the official DEB package, 32-bit and 64-bit should probably be separated first 😉.

pelwell, 2021-01-19:
There's no possibility of interference - you've heard of Device Tree, haven't you 😉?

MichaIng, 2021-01-19:
Okay, thanks for clarifying, yes of course I know about device trees, but I lack enough insights to be such about such things 😅. Thanks for the quick fix 👍.

pelwell, 2021-01-24:
A kernel with this fix is available via rpi-update.

closed 2021-01-24 by pelwell

Joulinar, 2021-01-24:
ok thx for the fix. Looks good.

```
root@DietPi3:~# uname -a
Linux DietPi3 5.10.9-v8+ #1396 SMP PREEMPT Thu Jan 21 16:01:58 GMT 2021 aarch64 GNU/Linux
root@DietPi3:~# cat /sys/class/thermal/thermal_zone0/temp
46160
root@DietPi3:~#
```