raspberrypi/linux #4080: vc4-kms-v3d != vc4-kms-v3d-pi4 on ubuntu 21.04

fossfreedom, 2021-01-18:
> Unless you are running old firmware or an old set of overlays, `vc4-kms-v3d` and `vc4-kms-v3d-pi4` should be identical on a 2711-based device (Pi 4B, Pi 400 and CM4). Run `sudo vcdbg log msg |& grep File` to see what is actually being loaded.

_Originally posted by @pelwell in https://github.com/raspberrypi/linux/issues/3618#issuecomment-762161558_

I'm using pi4B 8GB - obs from #3618 using vc4-kms-v3d-p4 as an overlay gives 3d acceleration (help - about states the v3d driver in use).  Using the vc4-kms-v3d overlay gives llvmpipe software rendering in help - about.  

The linked obs says both should be identical if using the latest firmware.  On Ubuntu the firmware installed is  v1.20210108 (https://launchpad.net/ubuntu/+source/linux-firmware-raspi2 changelog)

Rather than spamming the linked issue I've raised this separately.

Any idea why on Ubuntu there is a difference?

---

vc4-kms-v3d-pi4 specified in /boot/firmware/config.txt

```

dad@dad-desktop:/opt/vc/bin$ sudo ./vcdbg log msg |& grep File
005522.632: brfs: File read: /mfs/sd/config.txt
005523.230: brfs: File read: 910 bytes
005622.392: brfs: File read: /mfs/sd/config.txt
006712.144: brfs: File read: 910 bytes
011415.922: brfs: File read: /mfs/sd/initrd.img
011455.236: brfs: File read: 23582665 bytes
011466.660: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
011497.238: brfs: File read: 47624 bytes
011682.215: brfs: File read: /mfs/sd/config.txt
011744.853: brfs: File read: 910 bytes
011757.778: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
012283.810: brfs: File read: 3710 bytes
012288.756: brfs: File read: /mfs/sd/cmdline.txt
013644.408: brfs: File read: 117 bytes
018385.224: brfs: File read: /mfs/sd/vmlinuz
```

vc4-kms-v3d specified in /boot/firmware/config.txt

```
dad@dad-desktop:/opt/vc/bin$ sudo ./vcdbg log msg |& grep File
[sudo] password for dad: 
005783.255: brfs: File read: /mfs/sd/config.txt
005783.850: brfs: File read: 906 bytes
005882.596: brfs: File read: /mfs/sd/config.txt
007005.713: brfs: File read: 906 bytes
011729.858: brfs: File read: /mfs/sd/initrd.img
011769.189: brfs: File read: 23582665 bytes
011781.046: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
011811.754: brfs: File read: 47624 bytes
011996.583: brfs: File read: /mfs/sd/config.txt
012059.203: brfs: File read: 906 bytes
012072.236: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d.dtbo
012098.727: brfs: File read: 2689 bytes
012100.767: brfs: File read: /mfs/sd/cmdline.txt
013409.971: brfs: File read: 117 bytes
018266.038: brfs: File read: /mfs/sd/vmlinuz
```

pelwell, 2021-01-18:
Ubuntu is obviously not distributing `/mfs/sd/overlays/overlay_map.dtb`. I'll nudge one their devs.

fossfreedom, 2021-01-18:
hmm ... There definitely is an overlay_map.dtb file in the /boot/firmware folder i.e. the same folder that has the config.txt file that I edited in `dtoverlay=vc4-kms-v3d`

pelwell, 2021-01-19:
But the firmware is expecting it to be inside the overlays directory. Try moving or copying it there.

pelwell, 2021-01-19:
I can see how the problem might have arisen - a build rule that says `.dtbo` files go into `overlays` and `.dtb` files don't, for example - but Canonical are now aware of the problem and it should be fixed in future releases. For now you have the easy workaround of copying `/boot/firmware/overlay_map.dtb` into `/boot/firmware/overlays/` after installation.

fossfreedom, 2021-01-19:
Excellent.  This does work quite nicely now to make kms and kms-pi4 equivalent.

Cheers for coordinating with Canonical - look forward to seeing this in Ubuntu proper in the future.

closed 2021-01-19 by fossfreedom