raspberrypi/linux #4082: 5.4.y no rt version

funny1023, 2021-01-20:
5.4.y no rt version for cm4

funny1023, 2021-01-20:
cm4 only use 5.4.y 
can not be used 4.19.rt

funny1023, 2021-01-20:
5.4.83

patches-5.4.78-rt44.tar.xz                         23-Nov-2020 16:20    271K
patches-5.4.82-rt45.tar.gz                         11-Dec-2020 19:18    384K
patches-5.4.82-rt45.tar.sign                       11-Dec-2020 19:18     228
patches-5.4.82-rt45.tar.xz                         11-Dec-2020 19:18    271K
patches-5.4.82-rt46.tar.gz                         21-Dec-2020 20:00    386K
patches-5.4.82-rt46.tar.sign                       21-Dec-2020 20:00     228
patches-5.4.82-rt46.tar.xz                         21-Dec-2020 20:00    273K
patches-5.4.84-rt47.tar.gz                         22-Dec-2020 17:23    386K
patches-5.4.84-rt47.tar.sign                       22-Dec-2020 17:23     228
patches-5.4.84-rt47.tar.xz                         22-Dec-2020 17:23    273K
patches-5.4.87-rt48.tar.gz                         08-Jan-2021 16:55    386K
patches-5.4.87-rt48.tar.sign                       08-Jan-2021 16:55     228
patches-5.4.87-rt48.tar.xz

pelwell, 2021-01-20:
The -rt branches are hosted here, but not supported. If somebody is willing to maintain a 5.4-rt branch here in the same way as has been done previously (using a combination of merge commits from rpi-5.4.y and the -rt patches) then they should speak up.

funny1023, 2021-01-21:
rt path has no 5.4.83 ,only 5.4.84 or 5.4.82,how to use rt in cm4

pelwell, 2021-01-21:
You have a few options:
a. Try the nearest available patch set and see if it applies.
b. Revert the changes between 5.4.82 and 5.4.83 to downgrade rpi-5.4.y.
c. Wind back to an earlier rpi-5.4.y version that does have a corresponding patch set.
d. Wait until rpi-5.4.y is updated and hope that it corresponds with an RT patch set.

andypugh, 2021-01-23:
This isn't a solution, but there is a set of rt kernel debs hosted at LinuxCNC. 
www.linuxcnc.org/dists/buster/base/binary-armhf/
32-bit only, and not currently working on all variants.

funny1023, 2021-01-26:
this is not kernel code  for cm4，how to use it in cm4 and update it？

thx

andypugh, 2021-01-26:
The (rt-patched) kernel source is at https://github.com/LinuxCNC/raspberrypi-linux
But there is little difference between using that and using the main repo and patching it.

funny1023, 2021-01-26:
ok，i try it in cm4，thx

funny1023, 2021-01-26:
What are the differences that affect what？

kdoren, 2021-02-20:
I made a 5.10.16 kernel with PREEMPT_RT.  Looks good so far on Pi4.  Boots on Pi3 but jack2 audio driver will make it crash fairly quickly.  Also 5.10 kernel, (both standard and rt, have SD card activity on Pi3 every 1 sec that is not present on Pi4.
I install from .tgz file but I also built & posted debs

[https://github.com/kdoren/linux/releases/tag/5.10.16-rt30](https://github.com/kdoren/linux/releases/tag/5.10.16-rt)