raspberrypi/linux #4084: Data transfer over USB becomes unreliable after upgrading to v5.4.83-v71+

TupelUK, 2021-01-20:
I think there is a bug in the latest pi4 Linux release, because data transfer over USB becomes unreliable after upgrading from Linux v4.19-v71+ to v5.4.83-v71+ on raspberry pi4.

We are using the USB C port as a RNDIS gadget to get TCP data to a PC.  I found the minimum python code required to reproduce the problem, it sets up a socketserver on the pi4, and the client on the PC repeatedly asks for 10MBs data.  In Linux v4.19-v71+ it is robust, and I can do the transfer > 5000 times (expected behaviour).  In v5.4.83-v71+ exactly the same code works 50 to 500 times then the link starts to break down, I get time out errors (even with time out set to 10 seconds), and frequently after that the link freezes entirely and requires the USB to be disconnected and reconnected for it to return.  It even takes down an SSH connection to the pi which is separate software (but connecting over the same USB cable).

I'm unable to use v4.19-v71+ from now because it is not compatible with the latest pi4 hardware, so I hope someone can help - either with a bug fix or workaround suggestions.

To reproduce:
Test code I am using is here:
[USBdataTest.zip](https://github.com/raspberrypi/linux/files/5843894/USBdataTest.zip)

Set up the pi4 as a usb gadget:
To /boot/config.txt add:
dtoverlay=dwc2
To /boot/cmdline.txt add:
modules-load=dwc2,g_ether
and use the /etc/network/interfaces file provided.
On the pi4 run:
python3 lanTestPi2.py 
On a windows PC run:
python3 lanClientTest.py 
It can be reproduced with or without an external PSU for the pi.

Other information:
PC OS: Windows 10
Pi4 boot loader: Sep 3 2020
Python version: 3.7

* Which model of Raspberry Pi? Pi4
* Which OS and version? 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2
* Which firmware version?Jan  8 2021 14:32:15 version 194a85abd768c7334bbadc3f1911c10a7d18ed14 (clean) (release) (start_x)
* Which kernel version?  Linux TupelPi1 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
I upgraded by using:
sudo apt update
sudo apt full-upgrade

Logs:
Time out error (from PC recv command).
No error from pi4, but PC can no longer connect to it.

Please let me know if there is any other information I can provide, or if I can do any tests for you to narrow down the problem.

TupelUK, 2021-01-21:
I have managed to narrow down the issue to the particular github commit that causes it.

If I do:
sudo rpi-update ab2197b00eb5ba600556c37ac79b9e05a41e767d
(thats the Oct 8, 2020 commit of the Hexxeh/rpi-firmware repo)
I get expected network behaviour.

If I do:
sudo rpi-update 98d907955a74ba1833f9c31b1fe11b237f8ba599
(thats the Oct 15, 2020 commit)
The network freezing happens.

The (for me) troublesome commit say something about "Fix USB init for 2835 bootcode.bin".

This gives a solution to our issue for now, because the functional Oct 8, 2020 commit is new enough to work on the new pi4 hardware, and old enough not to have the network freeze issue...but I would be surprised if it is not affecting others.

TupelUK, 2021-01-21:
FYI: the issue is also present in the latest commit I tried:  Jan 18, 2021: commit number: 246f5ead8d2d017fd2734d79e1f9a606fb65b72c

popcornmix, 2021-01-21:
@timg236 does "Fix USB init for 2835 bootcode.bin" affect start.elf and could it have an impact on later USB use?

timg236, 2021-01-21:
Not as far as I can tell. This could would only be used if rpiboot (usb device boot) is used. The change was to the ensure that soft disconnect was still called between bootcode.bin and start.elf on a Pi3 in USb mode.

timg236, 2021-01-21:
[fw-github-linux-4084.zip](https://github.com/raspberrypi/linux/files/5852034/fw-github-linux-4084.zip)

There's a separate commit around then which might be squashed. Please could you try 'sudo rpi-update 98d907955a74ba1833f9c31b1fe11b237f8ba599' and apply the attached experimental (untested fix) which disables some USB PHY reset code.

TupelUK, 2021-01-22:
I tested the files, but unfortunately it did not fix the issue.

timg236, 2021-01-22:
I've run the attached python scripts plus some other SSH and HTTP download soak tests with the latest APT FW release with a CM4 + Pi4. No errors observed. There's another theoretical change attached

[github-linux-4084-v2.zip](https://github.com/raspberrypi/linux/files/5857496/github-linux-4084-v2.zip)

timg236, 2021-01-22:
I managed to get a failure after 'pursuading' RNDIS to work on my Windows 10 machine.  It's possible that this won't fail on CM4 + IO board. 

Attached is a possible fix candidate which passed 1000 iterations a couple of times so I'll leave it running over the weekend. I think v2.zip won't fail either but this more targetted

[github-linux-4084-v5.zip](https://github.com/raspberrypi/linux/files/5858471/github-linux-4084-v5.zip)

TupelUK, 2021-01-25:
Thanks so much for the rapid response.  I've run some more tests:

On clean install:
 - linux version #1379, orig files 2 out of 400 tests timed out, then connection froze.
 - linux version #1379, 4084-v5.zip files pass 2x 1000 tests - no fails.

So the 4084-v5.zip files fix the problem on a fresh install, which is great.

I realised that the install I was testing with was quite modified and it seems this enhances the problem.  The 4084-v5.zip does not fix it, even though it does work on the clean install.

On modified install:
 - linux commit 98d..., 4084-v5.zip files connection frozen completely after 250 tests.
 - linux old commit 50b..., 4084-v5.zip files pass 2x 1000 tests - no fails.

I listed the changes that seem most relevant to the usb:
 - read only file system like this:  http://hallard.me/raspberry-pi-read-only/
 - bridge-utils installed
 - arduino piInstaller installed.

So for now (since I need my modifications), I'm using the old 50b... commit as a solution.  

I realise that this could be becoming an ever more challenging issue to debug, but if you think it could be important I'm happy to help if I can.

timg236, 2021-01-25:
Thanks for the test results. V5 restores a single PHY register setup operation which shouldn't be relevant on BCM2711, unfortunately, that code is pretty ancient so getting a definitive answer may take a while.
Since v5 improves things I'll try to get it merged as a partial fix until the Linux side is understood. It's also unusual that it takes so long to fail when this sort of problem normally causes enumeration or device configuration to fail.
I'll post an update when I know more.

timg236, 2021-01-28:
The firmware fix is now available in the latest rpi-update kernel. I've been testing with the 5.10 kernel which is now the default rpi-update kernel. The APT packages are switching to 5.10 very soon.

SpaceRat42, 2021-01-28:
After installing the new kernel via rpi-update on a rpi4 nfs stopped working. Other network services seemed nof affected (ssh, samba). After reverting back to 8c2ba64 nfs worked again.

timg236, 2021-01-28:
Deleted off topic comment