raspberrypi/linux #4087: Default branch does not match kernel version used in raspberrypi/firmware

pbrkr, 2021-01-21:
The kernel used in the raspberrypi/firmware repository is now 5.10.7 whereas the default branch in this repository is still rpi-5.4.y. I also see that the latest upstream version merged into this branch is 5.4.83, the latest 5.4 kernel is 5.4.91 so that suggests that maintenance has already started to move over to the rpi-5.10.y branch.

However on the other hand, I see that the rpi-5.10.y branch is still being rebased onto each new upstream kernel release instead of using merge commits which makes tracking this branch impossible.

So what's the currently supported branch? I'd like to update things in the meta-raspberrypi Yocto Project layer but need a bit more info before I can do that.

Cc: @agherzan, @kraj

pelwell, 2021-01-21:
As with previous kernel jumps, the default branch will be changed when the `stable` branch (which is used for Raspberry Pi OS releases) is switched to 5.10.

pbrkr, 2021-01-21:
@pelwell In that case, I think upstream kernel releases should still be merged into the rpi-5.4.y branch. I'm happy to submit a PR to merge in 5.4.91 if it will be considered.

pelwell, 2021-01-21:
When were you planning on bumping rpi-5.4.y, @popcornmix?

popcornmix, 2021-01-21:
At the moment rpi-update master branch testing is 5.10 kernel. There aren't regular 5.4 kernels being built.

pbrkr, 2021-01-22:
@popcornmix So I assume there's no plan to merge further upstream kernel releases into the rpi-5.4.y branch. Which would mean there is no branch which is stable (i.e. not being rebased) and currently getting security updates.

As I said, I'm happy to submit a PR to merge in 5.4.91, would that at least be considered for merging?

popcornmix, 2021-01-22:
Bumping the source tree is straightforward. I can do that if it's useful.
It's how kernel builds get tested and released that is the difficulty.

Moving apt kernel to 5.10 is imminent, and that is the kernel you get from rpi-update. We need to have a discussion exactly when we mark 5.10 as the default branch here and switch from rebasing to merging. Normally that coincides with kernel reaching apt.

Currently (virtually) no one is testing 5.4.84 - 5.4.91 so there is a small chance of a regression that hasn't been spotted.
If you are happy building that tree and testing it (minor kernel bumps rarely cause issues), then I can bump 5.4 source tree.
But I'm hoping the move to 5.10 will be very soon.