raspberrypi/linux #4088: Kernel 5.10.5-v8+ is missing module xt_physdev

treibholz, 2021-01-21:
**Describe the bug**
Kernel 5.10.5-v8+ is missing module xt_physdev.ko

Kernel 5.4.x (1.20201201-1) had this module and it is needed for my usecase (k3s-node).

**To reproduce**
* install latest kernel-package on Raspbian: raspberrypi-kernel 1.20210111.master-1
* reboot
* try `modprobe xt_physdev`

```
root@hostname:~# modprobe xt_physdev
modprobe: FATAL: Module xt_physdev not found in directory /lib/modules/5.10.5-v8+
root@hostname:~#
```

**Expected behaviour**
```
root@hostname:~# modprobe xt_physdev
root@hostname:~#
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Model: Raspberry Pi 4 Model B Rev 1.1
* OS: Raspberry Pi reference 2020-08-20
  Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7252c154838ec5b4576f29c996ac8fe3750cae12, stage4
* Firmware: Jan  7 2021 18:27:46 
  Copyright (c) 2012 Broadcom
  version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start_cd)
* uname -a: Linux k3s-p02 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux

**Logs**
not relevant

**Additional context**
To verify, I cloned this repo (branch rpi-5.10.y), after `make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig`, I checked `.config` for `CONFIG_NETFILTER_XT_MATCH_PHYSDEV` and found:

```
# CONFIG_NETFILTER_XT_MATCH_PHYSDEV is not set
```

After setting it to `CONFIG_NETFILTER_XT_MATCH_PHYSDEV=m`, I compiled a kernel, installed it and now it works for me.

***Crosscheck in branch rpi-5.4.y***

after running `make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig`, I checked `.config` and yes, it was set to build as module before:

```
CONFIG_NETFILTER_XT_MATCH_PHYSDEV=m
```

I also saw, that `arch/arm64/configs/bcm2711_defconfig` in rpi-5.10.y is quite "young" (only 10 commits), while it in branch rpi-5.4.y, there are 65 commits. Maybe something was missed in the merge.

I also expect this module to be missing in the kernels for other models and also armhf.

popcornmix, 2021-01-21:
Fair point.  Added here: https://github.com/raspberrypi/linux/commit/3f9ab74a54cf02e4810027f71e897ce48303ebc7
will be in future rpi-update builds.

treibholz, 2021-01-21:
Thank you for 3f9ab74, I'm not used to such quick response!

So with #4089 I missed my chance to get my name in the commit-log without doing lot of work! :-)

closed 2021-01-21 by treibholz

pelwell, 2021-01-21:
I can break it for you again if you like?