raspberrypi/linux #4090: Pi4 stuck on reboot

manolomalele, 2021-01-23:
As in the subject i have an issue and I'm going crazy on this.

Everything started few days ago. Before that no problems at all.

This is what happens.

Same sd cards of the last months
Same pi4
Same sd to uSD extender
Moodeaudio os.

Booting from the extender or from the pi is fine in every case.

Rebooting works only if I put the uSD in the pi slot or if I use a specific SD (Sandisk Ultra II 8gb - sdhc) clearly via the extender (this means the extender works).

I also flashed a card with latest raspbian os and after update & upgrade boots and reboots with no problems at all.

Other cards (SD or uSD) via the exender give the error you can see in the linked videos.

Why? What happened?

The bootloader is the latest one.

This is a video of the issue:

Same SD and just cold start and reboot

Pi3
https://www.youtube.com/watch?v=nObf2W4TJ_E

Pi4
https://www.youtube.com/watch?v=VK3PBLTnWeQ

The pi4 worked smoothly till last week with alle the sd and ųSD i have and the extender.
After upgrading (brand new installation) to moode 7 it doesn't perform reboot (ONLY reboot).

I hope someone can help me.

Thanks

pelwell, 2021-01-23:
A photograph of the boot screen messages would be much easier to read.

manolomalele, 2021-01-23:
[image]
The videos show the behaviour. 
Anyway here is it the photo