raspberrypi/linux #4096: IRQ trigger rising edge doesnt work with 5.10.4

Makurisan, 2021-01-25:
I called this function in the probe and the handler immeadiately is called and the value on the GPIO is 0. It seems the irq handler is configured with POLLing enabled but I want only rising edge.

	rc = devm_request_irq(&vhub->spi->dev, vhub->irq, ast_vhub_irq, IRQF_TRIGGER_RISING | IRQF_SHARED,
						  KBUILD_MODNAME, vhub);

Is this a bug or what leads to this.

pelwell, 2021-01-25:
How are you declaring the GPIO interrupt? It is normal (and perhaps necessary) to declare the trigger type using the IRQ flags in Device Tree, e.g. (from https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/sc16is750-i2c-overlay.dts):
```
			sc16is750: sc16is750@48 {
				compatible = "nxp,sc16is750";
				reg = <0x48>; /* i2c address */
				clocks = <&sc16is750_clk>;
				interrupt-parent = <&gpio>;
				interrupts = <24 2>; /* IRQ_TYPE_EDGE_FALLING */
				gpio-controller;
				#gpio-cells = <2>;
				i2c-max-frequency = <400000>;
			};
```
`IRQ_TYPE_EDGE_RISING` has the value 1, so your interrupts declaration would be:
```
				interrupt-parent = <&gpio>;
				interrupts = <0 1>; /* IRQ_TYPE_EDGE_RISING */
```

Makurisan, 2021-01-25:
Ok, at the moment we use the config.txt with
dtoverlay=anyspi:spi0-0,dev="hubshield,v-hub",speed=1000000
In the linux documentation it is said that request_irq with parameter flag equal zero use the device tree otherwise the flags which are specified.

Is it possible to set the parameter in the config.txt?

pelwell, 2021-01-25:
The `anyspi` overlay doesn't support interrupts. You will have to make a custom overlay to go along with your out-of-tree driver.

Makurisan, 2021-01-26:
Ok, but I don't understand why I can't define it in code... Can you explain the reason for that?

pelwell, 2021-01-26:
Post the code you are using to configure the GPIO IRQ.

Makurisan, 2021-01-26:
I have changed it to the above overlay and it is the same, means polling.

`	vhub->irq = gpio_to_irq(vhub->gpio_irq);
	if (vhub->irq < 0) {
		dev_info(&vhub->spi->dev, "Failed to request spi irq:%d from device tree\n",
					vhub->irq);
	}
	else {
		rc = of_irq_get(vhub->spi->dev.of_node, 0);
		dev_info(&vhub->spi->dev, "Succeeded to request gpio:%d with irq:%d\n", vhub->gpio_irq, vhub->irq);
	}
	// irq_set_irq_type(vhub->irq, IRQ_TYPE_LEVEL_HIGH);
	// irq_set_irq_type(vhub->irq, IRQ_TYPE_EDGE_FALLING );
	rc = devm_request_threaded_irq(&vhub->spi->dev, vhub->irq, NULL, ast_vhub_irq,
					IRQF_ONESHOT | IRQF_TRIGGER_RISING | IRQF_TRIGGER_FALLING, KBUILD_MODNAME, vhub);
	if (rc)
	{
		dev_err(&vhub->spi->dev, "Failed to request interrupt\n");
		goto err;
	}
`

pelwell, 2021-01-26:
How can you tell that it is polling?

Makurisan, 2021-01-26:
If I load the kernel driver the irq handler is called immeadiately. 

Inside the handler I print a debug message and with it the value from the gpio "gpio_get_value(g_gpio_ip_irq)". The function gets the  value "1" if I set the gpio to "1". Do I set the gpio to low the gpio in the handler has the value "0". 

You see it is polling regardless what I set in the request_irq function.

pelwell, 2021-01-26:
There's a typo in your code (`IIRQF_TRIGGER_FALLING`) - are you sure you are building and running it?

Also, are you sure there is no glitching on the IRQ line?

Have a look at my gpio-fsm driver - drivers/gpio/gpio-fsm.c and arch/arm/boot/dts/overlays/fsm-demo-overlay.dts - for an example of code that uses edge triggering. In your situation I would add some debug code to that driver to confirm that the edge triggering is working as expected, then see how your code differs.

Makurisan, 2021-01-26:
Yes, this was an copy,paste,rewrite problem.
I now changed to 

`	rc = devm_request_irq(&vhub->spi->dev, vhub->irq, ast_vhub_irq, 0, KBUILD_MODNAME, vhub);
`
and in the overlay 

`interrupts = <17 2>; /* IRQ_TYPE_EDGE_FALLING */`

The gpio is now from the overlay. The same result. always polling.

pelwell, 2021-01-26:
As I said above, start with the known-good gpio-fsm driver, perhaps changing the configuration to use GPIO 0, and add enough debug to prove to yourself that edge triggering works. If it doesn't, I can test that configuration here.

Makurisan, 2021-01-27:
No, I dont do that. I have compiled our kernel with 5.4.83-v7+ on an rpi 3 and it works as expected.

It is a problem in your changes. Do it like I and you will see the problem.

pelwell, 2021-01-27:
You are asking me to investigate a problem that so far only you are seeing, with a non-standard driver and overlay. I've requested that you test with an (RPi-)standard driver and overlay - is that too much to ask?

Makurisan, 2021-01-27:
Ok, write exactly what I should do and I test it.

Makurisan, 2021-02-04:
Is it possible that someone work on this?

pelwell, 2021-02-04:
It's still on the list to look at. If you can provide a minimal test-case driver to demonstrate the problem that simply watches a GPIO line, where I can run a patch cable between it and another GPIO and trigger it with raspi-gpio (to avoid bounces) then it will get higher priority.

Makurisan, 2021-02-18:
The problem still exist but I made further investigation. The problem comes with the gpio irq function "devm_request_irq". For other reasons I switched to "devm_request_threaded_irq". This version works fine. The trigger comes only once.

pelwell, 2021-02-18:
If you are calling a function that sleeps in the IRQ handler then all sorts of things might go wrong - the threaded IRQ handler would avoid that.

Makurisan, 2021-02-18:
Yes, I read about this. An other point regarding the above DTs. How can I set pull down to the gpio 25 in the dts?

	fragment@1 {
		target = <&spi0>;
        __overlay__ {
			/* needed to avoid dtc warning */
    		#address-cells = <1>;
			#size-cells = <0>;
 			status = "okay";
            vhub: vhub@48 {
				compatible = "hubshield,v-hub";
				reg = <0x0>;
				interrupt-parent = <&gpio>;
				interrupts = <0 2>; /* IRQ_TYPE_EDGE_FALLING */
                reset-gpios = <**&gpio 25 1**>; // GPIO_ACTIVE_HIGH
				gpio-controller;
				#gpio-cells = <2>;
				spi-max-frequency = <4000000>;
			};
		};
	};

Thanks for your help.

pelwell, 2021-02-18:
Do you mean pull? Pull is normally redundant on an output GPIO - the drive of the pin overrides the pull. What do you think it will achieve?

That reset-gpios declaration looks wrong - the data doesn't match the comment; from the dt-bindings:
```
#define GPIO_ACTIVE_HIGH 0
#define GPIO_ACTIVE_LOW 1
```
If you do really want to set the pull for some reason then it will need a separate pinctrl declaration under the gpio node - I'll show you how once I know that you need it.

Makurisan, 2021-02-18:
This is my actual DTS.

I want GPIO 5 as INPUT PullDown  // this is per default OUT

Thanks

I	fragment@1 {
		target = <&spi0>;
        __overlay__ {
			/* needed to avoid dtc warning */
    		#address-cells = <1>;
			#size-cells = <0>;
 			status = "okay";
            vhub: vhub@48 {
				compatible = "hubshield,v-hub";
				reg = <0x0>;
				interrupt-parent = <&gpio>;
				interrupts = <5 2>; /* IRQ_TYPE_EDGE_FALLING */
        reset-gpios = <&gpio 25 1>; // GPIO_ACTIVE_HIGH
				max-ports = <5>; // count of virtual hub ports
				endpoints = <15>; // count of hub generic endpoints
				gpio-controller;
				#gpio-cells = <2>;
				spi-max-frequency = <1000000>;
			};
		};
	};

	__overrides__ {
		ports = <&vhub>,"max-ports:0";
		spi_irq = <&vhub>,"interrupts:0";
        spi_cs =  <&vhub>,"reg:0";
		speed =  <&vhub>,"spi-max-frequency:0";
    };

pelwell, 2021-02-18:
You want a pull _down_ on a _falling_ edge IRQ on GPIO _5_ (not 25, as you first said)? You'd normally use a pull _up_ with a _falling_ edge IRQ.

Do you have nothing to say about the reset GPIO?

Makurisan, 2021-02-19:
The reset-pin is named for our devboard which is connected to the pi.  If I know who it must be configured for GPIO 5, I know it for GPIO 25.
Btw the GPIO5 is working. I set the pull down at the moment with raspi-gpio. 

Please show how this can be configured with DTS.

pelwell, 2021-02-19:
This should work - we add a new fragment to add the pins definition under the gpio controller, then add pinctrl properties to the main driver node so that the pins are activated when the driver is probed:
```
fragment@1 {
	target = <&spi0>;
    __overlay__ {
		/* needed to avoid dtc warning */
		#address-cells = <1>;
		#size-cells = <0>;
		status = "okay";
        vhub: vhub@48 {
			compatible = "hubshield,v-hub";
			reg = <0x0>;
			interrupt-parent = <&gpio>;
			interrupts = <5 2>; /* IRQ_TYPE_EDGE_FALLING */
    reset-gpios = <&gpio 25 1>; // GPIO_ACTIVE_HIGH
			max-ports = <5>; // count of virtual hub ports
			endpoints = <15>; // count of hub generic endpoints
			gpio-controller;
			#gpio-cells = <2>;
			spi-max-frequency = <1000000>;
			pinctrl-names = "default";
			pinctrl-0 = <&vhub_pins>;
		};
	};
};

fragment@2 {
	target = <&gpio>;
	__overlay__ {
		vhub_pins: vhub_pins {
			brcm,pins = <5>;
			brcm,function = <0>; // input
			brcm,pull = <1>;     // down
		};
	};
};
```
You could have worked this out for yourself by reading the other overlays - all the source code is available to browse: https://github.com/raspberrypi/linux/tree/rpi-5.10.y/arch/arm/boot/dts/overlays

Makurisan, 2021-02-19:
Great! Sorry, can you show me how the pin 25 output, pullup must be referenced in fragment@2?

pelwell, 2021-02-19:
Just add a second pin, function and pull to the vhub_pins node. I'll do the first one for you:
```
brcm,pins = <5 25>;
```
An output is function 1, and a pull 2 is up.

pelwell, 2021-02-19:
But I still think setting a pull on an output is strange - I think you will find that making it an output early is enough so you can set the pull to 0 (none).

Makurisan, 2021-02-19:
Is the value for output correct?
	
	__overlay__ {
			vhub_pins: vhub_pins {
				brcm,pins = <5 25>;	   		
				brcm,function = <0 1>; // input, output
				brcm,pull = <1 0>;     // down, none
			};
		};

pelwell, 2021-02-19:
Yes - that looks good.

Makurisan, 2021-02-19:
Oh, you answered. How can I check if something goes wrong? dtc  compile is correct.

pelwell, 2021-02-19:
Oh, you asked another question. Use `sudo vcdbg log msg` to look for overlay errors, and `raspi-gpio get 0-27` to check the pin state.

Makurisan, 2021-02-19:
GPIO 25: level=0 fsel=0 func=INPUT

pi@pi3:~/vusb » sudo vcdbg log msg
001408.866: brfs: File read: /mfs/sd/config.txt
001409.898: brfs: File read: 1784 bytes
001479.408: brfs: File read: /mfs/sd/config.txt
001480.342: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
001499.999: brfs: File read: 1784 bytes
001691.071: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
001692.399: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
001692.462: *** Restart logging
001711.771: HDMI0: hdmi_pixel_encoding: 162000000
001717.115: dtb_file 'bcm2710-rpi-3-b.dtb'
001723.196: brfs: File read: /mfs/sd/bcm2710-rpi-3-b.dtb
001723.219: Loading 'bcm2710-rpi-3-b.dtb' to 0x100 size 0x6e14
001736.617: brfs: File read: 28180 bytes
001749.257: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
001814.796: brfs: File read: 1523 bytes
001819.589: brfs: File read: /mfs/sd/config.txt
001820.209: dtparam: audio=on
001836.414: brfs: File read: 1784 bytes
001838.813: brfs: File read: /mfs/sd/cmdline.txt
001838.874: Read command line from file 'cmdline.txt':
001838.898: 'console=serial0,115200 console=tty1 root=PARTUUID=4e17d452-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles'
003548.976: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
003603.744: brfs: File read: 166 bytes
004035.373: brfs: File read: /mfs/sd/kernel7.img
004035.398: Loading 'kernel7.img' to 0x8000 size 0x6034a8
004035.430: Device tree loaded to 0x2eff8d00 (size 0x7288)
004040.062: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
007662.740: vchiq_core: vchiq_init_state: slot_zero = 0xf7580000, is_master = 1
007667.391: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
007673.228: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x

Makurisan, 2021-02-19:
Modinfo has with some version this plus sign in the name:
vermagic:       5.10.11-v7+
What does this mean?

pelwell, 2021-02-19:
The overlay doesn't appear in the firmware log, so you must be loading it with the `dtoverlay` command under Linux? You might find some messages in the kernel log - `dmesg` - so if you're loading the overlay from the command line then:
```
$ sudo dmesg -C                # to clear out the old messages
$ sudo dtoverlay your-overlay
$ dmesg                        # to show any new messages
```

The `+` just means there are downstream changes. As long as it matches `uname -r` it's all good.

Makurisan, 2021-02-19:
I know, because I print some messages:
- hub port, spi clock 

But now I get this overlay: WARNING. This is new. I seems there is a problem with fragment@2?

Is the order in the dts correct? this second overrides?

[Feb19 16:03] v_hub: loading out-of-tree module taints kernel.
[Feb19 16:04] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/spidev@0/status
[  +0.000047] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
[  +0.038799] v_hub spi0.0: Hub device has initiated 5 hub ports.
[  +0.000055] v_hub spi0.0: Spi clock set at 1000 KHz.
[

/*
 * Universal device tree overlay for SPI devices
 */

/dts-v1/;
/plugin/;

/*
#include <include/dt-bindings/interrupt-controller/irq.h>
 */

/ {
	compatible = "brcm,bcm2711";

	fragment@0 {
		target = <&spidev0>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@1 {
		target = <&spi0>;
        __overlay__ {
			/* needed to avoid dtc warning */
    		#address-cells = <1>;
			#size-cells = <0>;
 			status = "okay";
      vhub: vhub@48 {
				compatible = "hubshield,v-hub";
				reg = <0x0>;
				interrupt-parent = <&gpio>;
				interrupts = <5 2>; /* IRQ_TYPE_EDGE_FALLING */
        reset-gpios = <&gpio 25 1>; // GPIO_ACTIVE_HIGH
				max-ports = <5>; // count of virtual hub ports
				endpoints = <15>; // count of hub generic endpoints
				gpio-controller;
				#gpio-cells = <2>;
				spi-max-frequency = <1000000>;
			};
		};
	};
	fragment@2 {
		target = <&gpio>;
		__overlay__ {
			vhub_pins: vhub_pins {
				brcm,pins = <5 25>;	   		
				brcm,function = <0 1>; // input, output
				brcm,pull = <1 0>;     // down, none
			};
		};
	};
	__overrides__ {
		ports = <&vhub>,"max-ports:0";
		spi_irq = <&vhub>,"interrupts:0";
    spi_cs =  <&vhub>,"reg:0";
		speed =  <&vhub>,"spi-max-frequency:0";
    };

};

Makurisan, 2021-02-19:
I see something. The chip is bcm2711 but the pi is Pi3 B. Can this be the cause? 
If I want the dts for 3, 4 pi is  BCM2837 correct?

pelwell, 2021-02-19:
For a generic overlay (not BCM2711-specific) use `brcm,bcm2835`.

The warnings about leaks are harmless and unavoidable.

Makurisan, 2021-02-19:
I checked some dts. Is a line like this 
pinctrl-0 = <&vhub_pins>;
necessary because this is not in the dts?

pelwell, 2021-02-19:
Fragment 2 puts vhub_pins in the dts. The pinctrl-0 line is necessary if you want the GPIOs configured automatically with pulls etc.

Makurisan, 2021-02-19:
Thanks, this it works now.

Makurisan, 2021-02-23:
I dont know what happens but not it isn't working and it is not clear why.
If I list the active spi devices with "ls -l /dev/spi*": 
crw-rw---- 1 root spi 153, 0 Feb 23 08:18 /dev/spidev0.1
The device with which I will working is "spidev0.0".
What must appear in the dev list, if I overload the spidev0 with the above DTS?

pelwell, 2021-02-23:
The `/dev/spidev*` devices are created by the spidev driver - with spidev on 0.0 disabled there will be no `/dev/spidev0.0`. You might find an entry for your device in `/sys/class/spi_master` or `/sys/bus/spi`, otherwise you need to look for device-specific evidence; for example, an enc28j60 will appear as a network interface in `ifconfig` (probably `eth1`).

Makurisan, 2021-02-23:
Ah, yes my device is under:
ls /sys/class/spi_master/spi0/spi0.0

pi@pi3:~/spidev-test(master○) » ls /sys/class/spi_master/spi0/spi0.0
driver           modalias  power      spi0.0:p2  spi0.0:p4  statistics  uevent
driver_override  of_node   spi0.0:p1  spi0.0:p3  spi0.0:p5  subsystem

I create a udc and I have 4 spi0.0:p1 ... p4
Do you know what here is now the device because I want to loopback miso and mosi but the spidev-test needs the device?

pelwell, 2021-02-23:
> Do you know what here is now the device because I want to loopback miso and mosi but the spidev-test needs the device?

I'm not sure that I understand your question - can you reword it? - but it isn't possible to use spidev and a dedicated driver with the same SPI device/CS. If you want to use spidev you'll have to disable your device node and create an spidev node instead.

N.B. This "Issues" section is for reporting bugs in our kernels. I think you've got past that stage now.

closed 2021-02-23 by pelwell

Makurisan, 2021-02-23:
thanks for your help.